beagleboard/linux #19: BB-VIEW-LCD7

jadonk, 2015-01-15:
Latest kernel didn't seem to have the overlay in it for me. I thought we had support, but I'm guessing not. Let me know if I missed it somewhere.

I'm trying out a hack: https://github.com/beagleboard/linux/commit/a4b6affd637157158bf610230f727d27a86a2385

jadonk, 2015-01-15:
I confirmed this mostly worked using the build at:
http://builds.beagleboard.org/linux/3.8.13-bone69-bb-view-7/a4b6affd637157158bf610230f727d27a86a2385/

I edited uEnv.txt to include the line:
optargs=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN capemgr.enable_partno=BB-VIEW-LCD7-01

However, the color is off:
[image]

Also, like most resistive touch displays, it isn't the most accurate around the edges.

RobertCNelson, 2015-01-15:
There's two patches disabled that will fix that:

https://github.com/RobertCNelson/bb-kernel/blob/am33x-v3.8/patches/capes/0037-capes-element14_bb_view_lcd_capes.patch

&

https://github.com/RobertCNelson/bb-kernel/blob/am33x-v3.8/patches/fixes/0009-sitara_red_blue_swap_workaround.patch

It'll swap the red/blue channels, however it breaks the lcd4/7/etc..

Regards,

jadonk, 2015-01-15:
Can't we put that setting under the control of the device-tree configurable parameter?

RobertCNelson, 2015-01-15:
The patch tries to do that, "bgrx_16bpp" it just doesn't actually work...

jadonk, 2015-01-15:
@RobertCNelson Any idea why the MstaXXXX 148f:7601 Ralink driver doesn't seem to be there?

RobertCNelson, 2015-01-15:
Out of tree drivers..

https://github.com/rcn-ee/mt7601u

You might be able to get away with:

sudo apt-get install mt7601u-modules-3.8.13-bone69

Then copy:
/lib/modules/3.8.13-bone69/kernel/drivers/net/wireless/mt7601Usta.ko

to your custom build module directory, followed by:

sudo depmod -a `uname -r`
sudo update-initramfs -uk `uname -r`

Regards,

jadonk, 2015-01-15:
I did that except forgot to do the update-initramfs. Thanks! It does work. Do we need this one to stay out-of-tree? Do we need to add building it to the autobuilder?

jadonk, 2015-01-16:
Helpful note on running the calibrator: https://gist.github.com/anonymous/0d64683994a0d78afdce

fagle, 2017-04-27:
the branch “3.8.13 bone69 bb view 7 ” by @jadonk  works fine for element14_bb_view_lcd_capes, will(/did) you merge that to "3.8" or newer branches. will that cause any conflict to other types of similar lcds?

RobertCNelson, 2017-04-27:
@fagle or you could just use v4.4.x with the overlay manually forced:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-VIEW-LCD7-01-00A0.dts

remember to switch xorg to 24bit mode

```
sudo sed -i -e 's:16:24:g' /etc/X11/xorg.conf
```

Regards,

RobertCNelson, 2017-04-27:
@fagle just tested with u-boot overlays:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

in /boot/uEnv.txt set:

enable_uboot_overlays=1
uboot_overlay_addr0=/lib/firmware/BB-VIEW-LCD7-01-00A0.dtbo
disable_uboot_overlay_video=1
disable_uboot_overlay_adc=1
#enable_uboot_cape_universal=1

then on bootup you should see: (note i have the lcd4 plugged in, and
booting off the eMMC)

```
U-Boot SPL 2017.05-rc2-00002-gbcecad9214 (Apr 17 2017 - 21:20:43)
Trying to boot from MMC2

U-Boot 2017.05-rc2-00002-gbcecad9214 (Apr 17 2017 - 21:20:43 -0500),
Build: jenkins-github_Bootloader-Builder-542

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
Reset Source: Global external warm reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
Card did not respond to voltage select!
mmc_init: -95, time 12
Card did not respond to voltage select!
mmc_init: -95, time 13
Card did not respond to voltage select!
mmc_init: -95, time 13
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
Card did not respond to voltage select!
mmc_init: -95, time 12
Card did not respond to voltage select!
mmc_init: -95, time 12
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
1863 bytes read in 14 ms (129.9 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.4.62-ti-r99 ...
8649792 bytes read in 564 ms (14.6 MiB/s)
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
loading /boot/dtbs/4.4.62-ti-r99/am335x-boneblack-uboot.dtb ...
53128 bytes read in 44 ms (1.2 MiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/AM335X-20-00A0.dtbo ...
378 bytes read in 41 ms (8.8 KiB/s)
uboot_overlays: loading /lib/firmware/BB-VIEW-LCD4-01-00A0.dtbo ...
5735 bytes read in 67 ms (83 KiB/s)
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1105 bytes read in 56 ms (18.6 KiB/s)
uboot_overlays: uboot loading of
[/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt
[disable_uboot_overlay_video=1]...
uboot_overlays: uboot loading of [/lib/firmware/BB-ADC-00A0.dtbo]
disabled by /boot/uEnv.txt [disable_uboot_overlay_adc=1]...
uboot_overlays: add [enable_uboot_cape_universal=1] to /boot/uEnv.txt
to enable...
loading /boot/initrd.img-4.4.62-ti-r99 ...
5311919 bytes read in 350 ms (14.5 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1
root=UUID=8fa273de-f279-49a3-ad9e-3c1065f98b9a ro rootfstype=ext4
rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
...
debug: [bootz 0x82000000 0x88080000:510daf 88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8faef000, end 8ffffdaf ... OK
   reserving fdt memory region: addr=88000000 size=6e000
   Loading Device Tree to 8fa7e000, end 8faeefff ... OK

Starting kernel ...
```

fagle, 2017-05-01:
@RobertCNelson  I tried as you suggest, the lcd display is ok. But the screen touch is still not working. The cursor follows my pen when i click down, but it keeps on going down to buttom edge when i release and i couldn't push any icon, it has some kind of jitter. This problem has been solved by commit 
 bc8f3d73f6c21ff4ef48e42336080c4f804d3646 and  a4b6affd637157158bf610230f727d27a86a2385 on branch “3.8.13-bone69-bb-view-7”  but i can't see relevant changes in 4.4 branch. I think the main difference is the file ti_am335x_tsc.c, but i don't quite understande the difference.

closed 2018-10-22 by RobertCNelson