beagleboard/linux #20: Power button status on event0

ruisebastiao, 2015-01-24:
Hello,

i want to monitor the status of the power button in the beaglbone black, i found in google some info and i tested the following script:

#!/bin/sh

BTN=/dev/input/event0

echo "Waiting for events..."

while true; do
	BTNVAL=`hexdump -e '8/2 "%x " "\n"' -n 16 $BTN | grep ' 74 ' | awk '{print $7}'`
	if [ "$BTNVAL" = "1" ]
	then
		echo "Power button detected"
		break
	fi
	echo "(other event detected)"
	sleep 1
done

echo "Sending shutdown..."
/sbin/shutdown -h now
echo "Finished!"

but i didn't work,

When i dumped the /dev/input/event0 nothing appears, why i'm not receiving power button events ?

I'm using a yocto generated linux distro with kernel 3.14. 

Thanks

ruisebastiao, 2015-01-25:
i tried another solution and installed acpid but when i try to start it i get the following error:

RTNETLINK1 answers: No such file or directory
acpid: error talking to the kernel via netlink

RobertCNelson, 2015-01-26:
Two patches need to be ported to v3.14.x:

https://github.com/beagleboard/linux/commit/aaa639f981b1675ebc04c0a5d3aff9933a6863b8
https://github.com/beagleboard/linux/commit/846897d56ddd71618847bf565b03080903e543fc

ruisebastiao, 2015-01-26:
Ok,

Thanks!

RobertCNelson, 2015-01-26:
btw, if you get something to work, feel free to try and push it upstream, I haven't seen anyone working on that issue.
Regards,

ruisebastiao, 2015-01-26:
Ok,
I will apply the patches today

RobertCNelson, 2015-01-26:
Okay, this work for me..
https://gist.github.com/RobertCNelson/8d0bb47015ed2b0ec242

I'm going to do a little more testing before i push it to the bb.org 3.14. repo...

ruisebastiao, 2015-01-27:
it work for me too. . on a Beaglebone Black + BBView 4.3 LCD running yocto distro 1.7

Thanks

closed 2015-01-27 by ruisebastiao

ruisebastiao, 2015-01-27:
Hi Robert,

Since i never did it before, is there any special procedure for pushing my revision to the upstream?

Best Regards,
Rui