beagleboard/linux #25: chipsee beaglebone expansion. the touchscreen not work

luchofuentes, 2015-03-27:
Hi , i have http://www.chipsee.com/product/evm/beagle/beaglebone-black-expansion-capacitive.html , compiled the kernel With am335x-bone-bbb-exp-c.dtsi , the touchscreen worked , but  Invested , the axis Y goes to 1024 to 0  , and axis X goes 0 to 768 , how to fix it ? driver is the problem ?

Thank !

RobertCNelson, 2015-03-27:
calibrate it with "xinput_calibrate"...

luchofuentes, 2015-03-27:
i tried calibrate with xinput_calibrate but did not work , and tested  with evtest and here look that the axis this invested , RobertCNelson , RobertCNelson could you work this cape ?

RobertCNelson, 2015-03-27:
Well, using xinput_calibrate worked for me, the last time i tested it, thou my startup script didn't always detect it:

http://elinux.org/Beagleboard:Capes_3.8_to_3.14#Chipsee_bbb-exp-c
https://github.com/RobertCNelson/boot-scripts/blob/master/3rdparty/xinput_calibrator_pointercal.sh

Regards,

luchofuentes, 2015-03-27:
I use archlinux arm , where I put this configuration ?

*dtb=<device>-bbb-exp-c.dtb
screen [x]
backlight (gpio) [x]
backlight (pwm) [ ]
touchscreen (capactive touch) [x] (the startup calibrator script doesn't always pick it up on startup)
buttons [x]
led [x]*

RobertCNelson, 2015-03-27:
Well, "lightdm" has a "display-setup-script" option that i use to link to that script.

luchofuentes, 2015-03-27:
this is my props of my touchscreen
*xinput --list-props "EP0790M09"
Device 'EP0790M09':
	Device Enabled (116):	1
	Coordinate Transformation Matrix (117):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (242):	0
	Device Accel Constant Deceleration (243):	1.000000
	Device Accel Adaptive Deceleration (244):	1.000000
	Device Accel Velocity Scaling (245):	10.000000
	Device Product ID (235):	0, 0
	Device Node (236):	"/dev/input/event1"
	Evdev Axis Inversion (246):	0, 0
	Evdev Axis Calibration (247):	1024, 0, 0, 600
	Evdev Axes Swap (248):	0
	Axis Labels (249):	"Abs MT Position X" (240), "Abs MT Position Y" (241), "None" (0), "None" (0)
	Button Labels (250):	"Button Unknown" (237), "Button Unknown" (237), "Button Unknown" (237), "Button Wheel Up" (122), "Button Wheel Down" (123)
	Evdev Scrolling Distance (251):	0, 0, 0
	Evdev Middle Button Emulation (252):	0
	Evdev Middle Button Timeout (253):	50
	Evdev Third Button Emulation (254):	0
	Evdev Third Button Emulation Timeout (255):	1000
	Evdev Third Button Emulation Button (256):	3
	Evdev Third Button Emulation Threshold (257):	20
	Evdev Wheel Emulation (258):	0
	Evdev Wheel Emulation Axes (259):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (260):	10
	Evdev Wheel Emulation Timeout (261):	200
	Evdev Wheel Emulation Button (262):*

closed 2015-03-27 by luchofuentes