beagleboard/linux #27: Capes support for kernel 4.0?

jiapei100, 2015-05-05:
Hi, it seems those capes under firmware haven't been updated ever since kernel 3.8?

More specifically,
1) Here is the folder "capes" for kernel 3.8
https://github.com/beagleboard/linux/tree/3.8/firmware/capes

2) However, there is no subfolder named "capes" ever since 3.13
https://github.com/beagleboard/linux/tree/3.13/firmware

3) Even if for kernel 3.12, the "capes" folder under "firmware" is imported from 3.8, with this commit info: "capes: import from 3.8"
https://github.com/beagleboard/linux/tree/3.12/firmware/capes

Now, here comes the question:
I'm using kernel 4.0, and I would like to know the capes support under this kernel....

Suggestions please?

Best Regards
Pei

RobertCNelson, 2015-05-05:
Some capes are supported under "4.0.x" via dtb-rebuilder:

https://github.com/RobertCNelson/dtb-rebuilder/tree/4.0.x

Most of the focus is still on v3.14.x: 

https://github.com/beagleboard/linux/tree/3.14

Regards,

jiapei100, 2015-05-06:
Hi, Robert:

As I've already asked. The thing is with kernel 4.0, there is NO file /sys/devices/bone_capemgr.*, so, the way to build a cape compatible firmware is different from the traditional way.

How to build my own cape firmware for a particular cape with your dtb-rebuilder?

1) To use traditional way to build my own cape firmware is understandable, but only before kernel 3.8;
2) Without /sys/devices/bone_capemgr.*, there is NO way to build my own cape firmware after kernel 3.13, but now, I'm playing with kernel 4.0.x
3) Here comes the question, how to build something new/non-existing with dtb-rebuilder with kernel 4.0.x ?

The thing is not "SOME capes have already been supported by dtb-rebuilder". The thing is: is there a simple manual for me to catch up how to write my own cape firmware specifically for my cape, and how to modify dtb-rebuilder?

Can you please help?

Cheers
Pei

RobertCNelson, 2015-05-07:
https://github.com/beagleboard/linux/tree/4.1

jiapei100, 2015-05-12:
This is great that it seems 4.1 comes with the cape support again.
However, the current linux-stable
<code> git clone git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git </code>
only supports v4.0 so far
I can easly
<code> git checkout v4.0 -b v4.0 </code>
but there is no v4.1 yet.
Does that mean I still have to wait?

Cheers
Pei

RobertCNelson, 2015-05-12:
v4.1-rc3 came out on Sunday, so pull linus's git over master..

hussainali0935, 2015-07-06:
Hi,
I downloaded kernel 4.1.1 from beagleboard repo.
I want to use cape manager mechanism to add audio cape to the kernel.
But there is no bone_capemgr folder in /sys/devices. I found bone_capemgr in a different path ( /sys/devices/platform) but when i try to add dtbo  to "slots" ( with "echo": command )  an error occurs and the task is not completed.
Where am i in wrong?
Regards

Output of the "dmesg | tail" command is here:

[    4.704129] Freeing unused kernel memory: 448K (c0950000 - c09c0000)
[    4.836344] EXT4-fs (mmcblk0p2): re-mounted. Opts: data=ordered
[    4.942415] random: dd urandom read with 16 bits of entropy available
[  163.203207] bone_capemgr bone_capemgr: part_number 'BB-BONE-AUDI-02', version 'N/A'
[  163.211262] bone_capemgr bone_capemgr: slot #4: override
[  163.216896] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[  163.224217] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-BONE-AUDI-02'
[  163.237112] __of_adjust_tree_phandle_references: Illegal property (size) 'fixup' @/__local_fixups__
[  163.246663] bone_capemgr bone_capemgr: slot #4: Failed to resolve tree

RobertCNelson, 2015-07-06:
@hussainali0935 

interesting, i hadn't tried echo with 4.1.x in a while, it was not working early in development.  anywho, double check what version of dtc you are running?

dtc --version

Then install dtc from here:

https://raw.githubusercontent.com/beagleboard/bb.org-overlays/master/dtc-overlay.sh

Regards,

hussainali0935, 2015-07-07:
Thanks for quick reply,
dtc version on my machine is:
Version: DTC 1.4.0-gf345d9e4.
and running "dtc-overlay.sh" results "dtc is up to date."
My question is:
I am not familiar with DTS structure and i can not change DTS to  support Audio Cape.
On the other hand, I want to use TLV320AIC23 CODEC to playing audio, and hence I must migrate to kernel 4.1.1 from 3.8, because kernel 3.8 dos not support this CODEC.
As the result, i am forced to use cape manager mechanism to change pin muxing.
It is my solution, Since i am newcomer in linux!!
So, Does cape manager mechanism exist in kernel 4.1.1??
and how Could i add Audio Cape dtbo to OS??

Regards,

RobertCNelson, 2015-07-07:
@hussainali0935 
```
Version: DTC 1.4.0-gf345d9e4
```
So you didn't follow the readme.md here:
https://github.com/beagleboard/bb.org-overlays

```
./dtc-overlay.sh
./install.sh
```

One version of the Audio cape is here:
https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BONE-AUDI-02-00A0.dts

Regards

hussainali0935, 2015-07-11:
Thanks a lot for quick answering.
I was in mistake. I thought that i should update DTC on DESKTOP LINUX!!
But i can not update dtc on BBB. because i can not connect BBB to internet.
Furthermore, i've made Root File System with the BUILDROOT and it is not include apt-get package.
with this situation, What do i do??

Please help me...

Regards

RobertCNelson, 2015-07-13:
@hussainali0935 well, patch buildroot, to include the dtc changes...

hussainali0935, 2015-07-21:
Thanks Robert,
At the moment, I'd left patch buildroot ( because I do not know how do this) and updated dtc with connecting BBB to internet. It was done succesfully.
The "Audio Cape" was added to  "slots", but there is still an error and "alsamixer" does not run.
Output of "demesg" is here:

[  137.061224] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-BONE-AUDI-02'
[  137.075747] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-BONE-AUDI-02-00A0.dtbo' loaded; overlay id #0
[  137.473312] 2-0018 supply IOVDD not found, using dummy regulator
[  137.473367] 2-0018 supply DVDD not found, using dummy regulator
[  137.473398] 2-0018 supply AVDD not found, using dummy regulator
[  137.473437] 2-0018 supply DRVDD not found, using dummy regulator
[  137.477092] davinci_evm sound: ASoC: CPU DAI (null) not registered
[  137.483687] davinci_evm sound: snd_soc_register_card failed (-517)
[  137.506609] davinci_evm sound: ASoC: CODEC DAI tlv320aic3x-hifi not registered
[  137.514289] davinci_evm sound: snd_soc_register_card failed (-517)

Regards,

hussainali0935, 2015-07-21:
Hi,
I also did other work.
Because I want to use "AIC23 CODEC" I changed "BB-BONE-AUDI-02.dts"  with replacing all "tlv320aic3x" with "tlv320aic23". I do not know this work is right or not. ( If not, please tell me what I should do) .
After this change, there was same error:

[   98.179219] bone_capemgr bone_capemgr: slot #4: override
[   98.179257] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[   98.179299] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-BONE-AUDI-02'
[   98.198630] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-BONE-AUDI-02-00A0.dtbo' loaded; overlay id #0
[   98.556095] davinci_evm ocp:sound: ASoC: CPU DAI (null) not registered
[   98.563052] davinci_evm ocp:sound: snd_soc_register_card failed (-517)
[   98.597954] davinci_evm ocp:sound: ASoC: CPU DAI (null) not registered
[   98.604884] davinci_evm ocp:sound: snd_soc_register_card failed (-517)
[   98.621659] davinci_evm ocp:sound: ASoC: CODEC DAI tlv320aic3x-hifi not registered
[   98.629653] davinci_evm ocp:sound: snd_soc_register_card failed (-517)

Regards,

hussainali0935, 2015-08-04:
Hi Robert,
My problem has not solved yet,
Please help me if possible,
I  urgently need to set up BBB AUDIO interface with the TLV320AIC23 codec,

Thanks and Regards,

RobertCNelson, 2015-08-04:
@hussainali0935 
I cleaned up:
https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BONE-AUDI-02-00A0.dts

last week, so it now loads correctly, i wasn't able to get audio output working yet, but maybe you'll figure out that last piece..
Regards,

mickae1, 2015-12-04:
Any news on the audio part Robert ?

RobertCNelson, 2015-12-04:
the BB-BONE-AUDI-02 isn't working at this point..

cswank, 2015-12-28:
@RobertCNelson you said earlier in this thread:

"i hadn't tried echo with 4.1.x in a while, it was not working early in development."

I'm having trouble with echo right now in 4.1 and I'm wondering what the symptoms of it "not working early in development" are?  My symptoms are that when trying to load a overlay with "echo BB-W1 > /sys/devices/platform/slots" shows no error messages, however, the w1-gpio module does not get loaded.

RobertCNelson, 2015-12-28:
@cswank "/sys/devices/platform/slots" that's 3.8, if your using 4.1.x+ read the "readme"

https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md

Regards,

cswank, 2015-12-28:
I'm sorry, I typed it incorrectly, the path is "/sys/devices/platform/bone_capemgr/slots" and the kernel is 4.1:  

# ls /sys/devices/platform/bone_capemgr/slots
/sys/devices/platform/bone_capemgr/slots
# uname -a
Linux bbb 4.1.13 #1 SMP PREEMPT Sun Dec 27 16:04:48 MST 2015 armv7l GNU/Linux

The kernel was built using bb.org_defconfig from this beagleboard/linux repo.

RobertCNelson, 2015-12-28:
@cswank from the readme:
```
sudo sh -c "echo 'BB-UART1' > /sys/devices/platform/bone_capemgr/slots"

[  255.828371] bone_capemgr bone_capemgr: part_number 'BB-UART1', version 'N/A'
[  255.828436] bone_capemgr bone_capemgr: slot #4: override
[  255.828468] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[  255.828501] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-UART1'
[  255.855687] 48022000.serial: ttyS1 at MMIO 0x48022000 (irq = 181, base_baud = 3000000) is a 8250
[  255.856691] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-UART1-00A0.dtbo' loaded; overlay id #0

debian@beaglebone:~$ cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-UART1
```

cswank, 2015-12-28:
@RobertCNelson 

    # sh -c "echo 'BB-W1' > /sys/devices/platform/bone_capemgr/slots"
    # dmesg | grep Wq
    # dmesg | grep W1
    [   20.863027] bone_capemgr bone_capemgr: part_number 'BB-W1', version 'N/A'
    [   20.883187] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-W1'
    [   20.896095] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-W1-00A0.dtbo' loaded; overlay id #0
    # cat /sys/devices/platform/bone_capemgr/slots
     0: PF----  -1
     1: PF----  -1
     2: PF----  -1
     3: PF----  -1
     4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-W1

It looks like it works, but: 

    # lsmod
    Module                  Size  Used by    Not tainted
    # lsmod
    Module                  Size  Used by    Not tainted
    # ls /sys/bus/w1
    ls: /sys/bus/w1: No such file or directory

I'm curious if echoing to slots is working for anyone.  Here is my dts file just in case it is somehow the issue.

/*compile:

    $ dtc -O dtb -o BB-W1-00B0.dtbo -b o -@ BB-W1-00B0.dts

NOTE: on arch linux be sure to install dtc-overlay:

    # pacman -S dtc-overlay
*/

/dts-v1/;
/plugin/;
/ {
        compatible = "ti,beaglebone", "ti,beaglebone-black";
        part-number = "BB-W1";
        version = "00A0";
        
        exclusive-use = "P9.22";
        
        fragment@0 {
                target = <&am33xx_pinmux>;
                __overlay__ {
                        dallas_w1_pins: pinmux_dallas_w1_pins {
                                pinctrl-single,pins = < 0x150 0x37 >;
                        };
                };
        };
        
        fragment@1 {
                target = <&ocp>;
                __overlay__ {
                        onewire {
                                compatible      = "w1-gpio";
                                pinctrl-names   = "default";
                                pinctrl-0       = <&dallas_w1_pins>;
                                status          = "okay";
                                gpios           = <&gpio0 2 0>;
                        };
                };
        };
};

RobertCNelson, 2015-12-28:
@cswank i have no idea if arch has the correct dtc version yet..

build/use this branch for "dtc"
https://github.com/RobertCNelson/dtc/commits/bb.org-4.1-dt-overlays5

i know one user had this work last week:

https://gist.github.com/RobertCNelson/d6c4d2015c6a8f7de0c8

instead of:

```
target = <&ocp>;
```
Use:

```
target-path="/";
```

Regards,

cswank, 2015-12-28:
@RobertCNelson thanks for pointing out that dtc repo.  I cloned it, checked out out the bb.org-4.1-dt-overlays5 branch, and built it.  I used the dtc binary to build this overlay (you said to replace target = <&ocp>; with target-path="/"; so applied that to the first fragment too, and also tried it with only replacing what you explicitly suggested).

    /dts-v1/;
    /plugin/;
    / {
            compatible = "ti,beaglebone", "ti,beaglebone-black";
            part-number = "BB-W1";
            version = "00A0";

            exclusive-use = "P9.22";

            fragment@0 {
                    target-path ="/";
                    __overlay__ {
                            dallas_w1_pins: pinmux_dallas_w1_pins {
                                    pinctrl-single,pins = < 0x150 0x37 >;
                            };
                    };
            };

            fragment@1 {
                    target-path ="/";
                    __overlay__ {
                            onewire {
                                    compatible      = "w1-gpio";
                                    pinctrl-names   = "default";
                                    pinctrl-0       = <&dallas_w1_pins>;
                                    status          = "okay";
                                    gpios           = <&gpio0 2 0>;
                            };
                    };
            };
    };  

With this command:

    dtc -O dtb -o BB-W1-00A0.dtbo -b 0 -@ BB-W1-00A0.dts

And I get the same result on the beagle bone (no w1-wire module gets loaded).  And if you are wondering, yes the w1-module is available and can be loaded with modprobe.

I am getting the feeling that there is something wrong with echoing to the slots file.

As far as the gist you pasted, it looks like that isn't an overlay but instead would be put into something like am335x-bone.dts in the kernel source, right?  I'm I'm trying to figure out device trees out so I'm no expert.

cswank, 2015-12-28:
And I just thought of this possibility.  The linux build of beagleboard/linux was done via buildroot.  Perhaps the dtbo file I built was correct, but the dtb file that gets loaded by u-boot was built using a dtc that isn't doing it correctly.  Then the overlay is getting applied to a device tree that was generated by a  bunk dtb file.  

That is a total guess, but is it possible?

RobertCNelson, 2015-12-28:
During a kernel build, the built-in version of dtc is used..

btw, here is the non-overlay version:
http://kvurd.com/blog/beaglebone-black-1-wire-temperature-debian-8-kernel-4.1-no-capemgr/

(there's a possiblty, onewire doesn't work as an overlay, in our current 4.1.x build, some config's have to be modules since when built-in they fail..)

Regards,

cswank, 2015-12-29:
Thanks for that link.  That was a great learning experience.  I built the dtb file and applied it to my beaglebone/linux kernel in buildroot and ... nothing.  No w1-gpio module loaded. 

For a sanity check I applied the same dtb file to the uEnv.txt of a system built with https://rcn-ee.com/rootfs/bb.org/release/2015-11-03/console/BBB-eMMC-flasher-debian-7.9-console-armhf-2015-11-03-2gb.img.xz and it worked, I was able to read temperature.

I'm not sure where the kernel of the image I installed came from (mainline or beagleboard/linux) but it is 4.1 (which was installed via "apt-get install linux-image-4.1.9-bone16").  I'd like to figure out next why this isn't working in buildroot and beagleboard/linux.

RobertCNelson, 2015-12-30:
@cswank 
Just got a few dallas one wire devices, we have to make sure a few modules are loaded.. I've setup an example dts and read script here:
https://github.com/RobertCNelson/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts
https://github.com/RobertCNelson/bb.org-overlays/blob/master/examples/BB-W1-P9.12/example.sh

Regards,

cswank, 2015-12-30:
@RobertCNelson cool, thanks.  I'll try those out too. 

Like I said, the dtb I used from that blog worked on your debian image but not on the beagleboard/linux buildroot install.  If you were me, where would be the first place you look in order to debug the buildroot problem?  

Also, could you tell me what the source of the linux kernel is when you do apt-get install linux-image-4.1.9-bone16 on your image?  Is it mainline?  The reason I ask is that if your image installs beagleboard/linux then I will know the problem isn't there.

RobertCNelson, 2015-12-30:
@cswank the 4.1.9-bone16 tag is here:

https://github.com/RobertCNelson/linux-stable-rcn-ee/commits/4.1.9-bone16

Regards,

cswank, 2015-12-30:
Ok, then I guess there is a chance the the problem is in beaglebone/linux I suppose.  I'm going to repeat my buildroot build but I'll try it with https://github.com/RobertCNelson/linux-stable-rcn-ee/commits/4.1.9-bone16 and see if I get different results with the 1-wire stuff.

RobertCNelson, 2015-12-30:
@cswank i was using 4.1.13-ti-r37 today when i ran the test..

Top of: https://github.com/beagleboard/linux/commits/4.1

Regards,

cswank, 2015-12-30:
@RobertCNelson forgive me but you lost me a bit.  The overlay you gave (https://github.com/RobertCNelson/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts)
has some includes and the paths look like they are from your dtb-rebuilder repo.  How do you build that dtbo file?  Those include statements trigger errors even though I ran the dtc-overlay.sh script which installed dtc: Version: DTC 1.4.1-g1e75ebc9

RobertCNelson, 2015-12-30:
@cswank the repo is self contained..
```
git clone https://github.com/beagleboard/bb.org-overlays
cd bb.org-overlays
make
```

cswank, 2015-12-30:
@RobertCNelson Then I take it you just rename the dtb file that gets created to dtbo.  That's what I just did and the kernel module still doesn't get loaded.  

I used the exact same kernel tag you did so that means the problem is in buildroot somehow.  Could it be the boot process or my uEnv.txt?  There are no errors to be found in dmesg.  I'm stuck for sure.

bootfile=zImage
fdtfile=am335x-boneblack.dtb
loadaddr=0x80007fc0
fdtaddr=0x80F80000
loadfdt=fatload mmc 0:1 ${fdtaddr} ${fdtfile}
loaduimage=fatload mmc 0:1 ${loadaddr} ${bootfile}
console=ttyO0,115200n8
mmcroot=/dev/mmcblk0p2
mmcrootfstype=ext2
uenvcmd=mmc rescan; run loaduimage; run loadramdisk; run loadfdt; run fdtboot
mmc_args=setenv bootargs console=${console} ${optargs} root=${mmcroot}
fdtboot=run mmc_args; bootz ${loadaddr} - ${fdtaddr}
rootfstype=${mmcrootfstype} root wait

RobertCNelson, 2015-12-30:
@cswank rename what, use the default 'dtb'...  then load the modules/dtbo as shown:

https://github.com/RobertCNelson/bb.org-overlays/blob/master/examples/BB-W1-P9.12/example.sh

Regards,

cswank, 2015-12-30:
I see now.  I was confused.

duartecarona, 2016-07-22:
@RobertCNelson we have a cape for GPS and GPRS and it was working perfectly with Kernel 3.8 but now on 4.4 it isn't. Since the location of the bone.capemgr folder has changed is there anything we need to change on our dts file to support the new Kernel? Please find below the dts file we use.

/dts-v1/;
/plugin/;

/ {
    compatible = "ti,beaglebone", "ti,beaglebone-black";

    /* identification */
    part-number = "BB-BONE-GPS-GPRS";
    version = "00A0";
    
    /* state the resources this cape uses */
	exclusive-use =
		/* the pin header P8 uses */
		"P8.39", /* STAT_LED      */
		"P8.40", /* PWRMON        */
		"P8.41", /* GSM_ON_OFF    */
		"P8.30", /* ON_OFF_GPS    */
		"P8.31", /* GSM_CTS       */
		"P8.32", /* GSM_RTS_IN    */
		"P8.37", /* GSM_TXD_IN    */
		"P8.38", /* GSM_RXD       */
		"P8.42", /* GPS_WAKEUP    */
		/* the pin header P9 uses */
		"P9.12", /* PWR_ON_OFF    */
		"P9.24", /* GPS_TX_IN     */
		"P9.26", /* GPS_RX        */
		/* Hardware IP cores in use */
		"uart5", 
		"uart1"; 	

    fragment@0 {
        target = <&am33xx_pinmux>;
        __overlay__ {
            pinctrl_uart6: pinctrl_uart6_pins {
                pinctrl-single,pins = <					 				   
                        0x0C4 0x24 /* P8.38 = UART5_RXD , MODE 4 */
                        0x0C0 0x04 /* P8.37 = UART5_TXD , MODE 4 */
                        0x0D8 0x26 /* P8.31 = UART5_CTS , MODE 6 */
                        0x0DC 0x06 /* P8.32 = UART5_RTS , MODE 6 */
                >;
            };
            pinctrl_uart2: pinctrl_uart2_pins {
                pinctrl-single,pins = <					 				   
                        0x180 0x20 /* P9.26 = UART1_RXD , MODE 0 */
                        0x184 0x00 /* P9.24 = UART1_TXD , MODE 0 */
                >;
            };
            pinctrl_gpio: pinctrl_gpio_pins {
                pinctrl-single,pins = <					 				   
                     	0x0B8 0x27 /* P8.39 = STAT_LED   = GPIO2_12   */
                     	0x0BC 0x27 /* P8.40 = PWRMON     = GPIO2_13   */
                     	0x0B0 0x07 /* P8.41 = GSM_ON_OFF = GPIO2_10  */
                     	0x0EC 0x07 /* P8.30 = ON_OFF_GPS = GPIO2_25  */
                     	0x0B4 0X27 /* P8.42 = GPS_WAKEUP = GPIO2_11  */
                     	0x078 0x07 /* P9.12 = PWR_ON_OFF = GPIO1_28  */	  
                >;
            };
        };
    };
	
	fragment@1{
		target = <&uart6>;
		__overlay__ {
			status 			= "okay";
		};
	};
	
	fragment@2{
		target = <&uart2>;
		__overlay__ {
			status 			= "okay";
		};
	};
			
	fragment@3 {
        target = <&ocp>;
        __overlay__ {
        	#address-cells = <1>;
			#size-cells = <1>;        
        
            uart6 {
                compatible = "bone-pinmux-helper";
                pinctrl-names = "default";
                pinctrl-0 = <&pinctrl_uart6>;
                status = "okay";
            };
            
            uart2 {
                compatible = "bone-pinmux-helper";
                pinctrl-names = "default";
                pinctrl-0 = <&pinctrl_uart2>;
                status = "okay";
            };
            
            gpio {
                compatible = "bone-pinmux-helper";
                pinctrl-names = "default";
                pinctrl-0 = <&pinctrl_gpio>;
                status = "okay";
            };
        };
    };
};