beagleboard/linux #29: am33xx.dtsi in kernels >=3.12 lacks CPU temperature sensor node

puleglot, 2015-06-02:
With kernel 3.8 CPU temperature sensor data is accessible via /sys/class/hwmon/ or using lm_sensors:
```
$ sensors
am335x-bandgap-isa-0000
Adapter: ISA adapter
temp1:        +57.0°C
```
But this sensor is not accessible with kernels >=3.12 (well.. to be honest I tested only 3.14 :)) I guess this is because am33xx.dtsi in kernels >=3.12 lacks the following node:
```
		bandgap@44e10448 {
			compatible = "ti,am335x-bandgap";
			reg = <0x44e10448 0x8>;
		};
```

Could it be added to kernels >=3.12?

RobertCNelson, 2015-06-02:
It turned out the sensor used on the am335x is too inaccurate and variable to be used for any situation. Thus it was decided to not push support mainline for the am335x.

Search linux-omap mailing list for the post.

Regards,

RobertCNelson, 2015-06-02:
Here is the post:
http://permalink.gmane.org/gmane.linux.linaro.devel/16501

puleglot, 2015-06-02:
OK. Thanks!