beagleboard/linux #31: SGX/OpenGL Status

JamesHagerman, 2015-06-04:
Hi there,

I've been trying to get OpenGL running on the Beaglebone Black but can't seem to get all of the pieces into one place.

What is the status of SGX kernel modules in this repo?
Are these modules compiled in already?

1. omaplfb
2. pvrsrvkm
3. bufferclass_ti

They are mentioned on the following TI wiki page: http://processors.wiki.ti.com/index.php/Graphics_SDK_Quick_installation_and_user_guide#Kernel_Modules

Any information would be great!

RobertCNelson, 2015-06-04:
First, there is no "OpenGL" just "OpenGL ES2".

Use either the 3.14.x or 4.1.x branches:

Then for Debian/Ubuntu images used by beagleboard.org users see for more out of the box help..

http://elinux.org/BeagleBoardUbuntu#SGX_BeagleBone.2FBeagleBone_Black
http://elinux.org/BeagleBoardDebian#SGX_BeagleBone.2FBeagleBone_Black

Regards,

JamesHagerman, 2015-06-04:
Okay, I'm trying this out now using linux-image-3.14.43-ti-r66

I see this message in the output of ./sgx_create_package.sh:
cp: cannot stat '/home/jamis/dev/ti-linux-kernel-dev/ignore/ti-sdk-pvr/Graphics_SDK/gfx_rel_es8.x/*.ko': No such file or directory

And later in the output it says:
Please ensure that PSP Linux Kernel is re-built atleast once,
Please refer online guide for further details. (sic)

Is this because the kernel modules are not needed?
Am I missing a step?

Thanks again for the help!

RobertCNelson, 2015-06-04:
There cleaned it up:
https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/7ed87e9b3640033fb3c3294490b919ce037adcaf

It's just the modules have been moved to a deb package.

Yet the graphic's *.bin's are still pretty useless so no one really uses them, thus we never cleaned up the sgx package script til now. ;)

Regards,

JamesHagerman, 2015-06-04:
Thanks!

I'm not sure what you mean by the *.bin's being useless... are they broken? are they missing something?

I noticed that this process doesn't include the demos under GFX_Linux_SDK in the *.bin

Is there a special way to include these?
Are they even worth it? (maybe this is what you mean by useless?)

RobertCNelson, 2015-06-04:
The demo's where dropped from the ti *.bin in 5.01.01.02

Regards,

JamesHagerman, 2015-06-04:
Okay, that's odd. Yesterday I made a TI account, downloaded, and extracted Graphics_SDK_setuplinux_hardfp_5_01_01_02.bin, and see the demos in the ti bin.

Are we looking at different bins?

closed 2020-05-12 by RobertCNelson