beagleboard/linux #35: [Branch 4.1] : tty spawning failed

parthitce, 2015-07-14:
Hello All,

I have tried compiling the BBB Linux version 4.1. I am able to boot the target successfully with the uImage and dtb.

But am not able to login into the console. Am not sure is there anything specific changes needed for tty spawning. The same RFS works fine with older version of the kernel.

I am getting the following message after few minutes ideal boot screen.

HEAD: 3c34b46918cab4eb5dcc7c42173cb743578dbd23

Log:
init Id O0 respawning too fast disabled for 5 minutes

Please help.

Thanks,
Parthiban N

RobertCNelson, 2015-07-14:
Console is back again at ttySx..

Copy this udev rule:

https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/jessie/debian/60-omap-tty.rules

Regards,

parthitce, 2015-07-28:
I have tried the udev rules and still with the same problem.

Output from lsusb:
Bus 001 Device 004: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x Composite Device

I am using tty with ttyUSB0. Is there any specific changes needed for this?

Log:
INIT: Id "O0" respawning too fast: disabled for 5 minutes
INIT: Id "O0" respawning too fast: disabled for 5 minutes

parthitce, 2015-08-23:
After updating the kernel command line arguments, the problem is not seen. Now am able to access the console.

closed 2015-08-23 by parthitce