beagleboard/linux #37: Compile error: multiple definitions of functions in of.h

agherzan, 2015-07-28:
Compiling this linux I get many errors similar to:

 arch/arm/mach-omap2/dss-common.o: In function `of_changeset_add_property_u32':
| /home/andrei/work/resin/resin-yocto/yocto-fido/build/tmp/work-shared/beaglebone/kernel-source/include/linux/of.h:1116: multiple definition of `of_changeset_add_property_u32'
| arch/arm/mach-omap2/id.o:/home/andrei/work/resin/resin-yocto/yocto-fido/build/tmp/work-shared/beaglebone/kernel-source/include/linux/of.h:1116: first defined here
| arch/arm/mach-omap2/dss-common.o: In function `of_changeset_add_property_bool':
| /home/andrei/work/resin/resin-yocto/yocto-fido/build/tmp/work-shared/beaglebone/kernel-source/include/linux/of.h:1122: multiple definition of `of_changeset_add_property_bool'
| arch/arm/mach-omap2/id.o:/home/andrei/work/resin/resin-yocto/yocto-fido/build/tmp/work-shared/beaglebone/kernel-source/include/linux/of.h:1122: first defined here
| /home/andrei/work/resin/resin-yocto/yocto-fido/build/tmp/work-shared/beaglebone/kernel-source/scripts/Makefile.build:337: recipe for target 'arch/arm/mach-omap2/built-in.o' failed

The patch which introduces this is:
commit 8d2e10661f4ee04cc8dfffb15186a4e20bef82d6
Author: Pantelis Antoniou <pantelis.antoniou@konsulko.com>
Date:   Wed Jun 10 18:35:53 2015 +0300

RobertCNelson, 2015-07-29:
Do you have a patch i can append to Pantelis' patch? otherwise use bb.org_defconfig

Regards,

agherzan, 2015-07-29:
I don't think I understood. The patch is already merged. Reverting it fixes these errors.

RobertCNelson, 2015-07-29:
It's not merged in 'mainline', still only an rfc:

https://lkml.org/lkml/2015/6/12/618

Enable CONFIG_OF_DYNAMIC

Regards,

agherzan, 2015-07-29:
The 4.1 branch is not the "mainline" - or the main branch for this repo? I'm really confused.

RobertCNelson, 2015-07-29:
mainline = https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/

The "4.1" branch is the latest version of the "capemgr" patch-set currently being worked on for inclusion into mainline linux..

Regards,

agherzan, 2015-07-29:
That makes sense. So the 4.1 branch of beagleboard is not stable? And shouldn't be used?

agherzan, 2015-07-29:
The idea is that I want to use capemgr on 4.1 with https://github.com/beagleboard/bb.org-overlays . In the instructions there I see references to kernel 4.1 which seems to be broken. That's how I ended up in this situation.

RobertCNelson, 2015-07-29:
It's not broken, if you use the default "bb.org_defconfig"...

We've been shipping it with jessie test images for a few weeks now..

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#2015-07-26

Regards,

agherzan, 2015-07-29:
OK. Do you see any obviously issue if I locally revert this patch and use the ti config fragments?

RobertCNelson, 2015-07-29:
Then you won't be able to use: https://github.com/beagleboard/bb.org-overlays

agherzan, 2015-07-29:
Worked. Thank you Robert.

closed 2019-05-03 by agherzan