beagleboard/linux #38: Random delay between bytes in UART driver

murix, 2015-07-29:
Tested in Kernel  >= 3.8.13-boneXX and all debian 7.4, 7.8, 8.0, 8.1

The ACR122L Contactless Card Reader is crashing due random delay/jitter in synchronous UART writes.

Howto reproduce the bug:
int fd=open("/dev/ttyO2",......);
buffer[100];
memset(buffer,0xff,sizeof(buffer));
write(fd,buffer,sizeof(buffer));

The write must be atomic, but in oscilloscope is clear to see large random delay between bytes.
Large random delay means several milliseconds.

This bug is not present  Angstrom Distribution (BeagleBone Black - 2GB eMMC) 2013-09-04.

RobertCNelson, 2015-07-29:
We have a new "rewritten" usart driver in 4.1.x please retest..

Regards,

murix, 2015-09-24:
@RobertCNelson I'm tested 4.1.3-ti-r6. ACR122L still crashing. The kernel serial driver still inserting very large random delay between writes.

leearmstrong, 2015-11-24:
@murix Did this get fixed at all?

murix, 2015-11-27:
@leearmstrong  not. To workaround partially I'm using "select" and polling. The crashing rate os ACS 122L NFC Cardless Reader decrease from 20/hour to  1/week