beagleboard/linux #39: d0ce07ed09d393f76889e71192d30c185c9530a5 disappeared (rebased) from 4.1 branch?

agherzan, 2015-08-10:
Two weeks ago I managed to build 4.1 branch with the revision: d0ce07ed09d393f76889e71192d30c185c9530a5

Now, this revision disappeared. Are rebases likely to happen on this fork? Or it was just an exception?

Regards,
Andrei

RobertCNelson, 2015-08-10:
master branches always get rebased, use the tags:

https://github.com/beagleboard/linux/tree/4.1.4-ti-r8

Regards,

closed 2015-08-10 by agherzan

sykrim, 2015-08-21:
The .git/packed-refs file contains:
b1b6908a34741b6785005d30b8e76f1305ffdc23 refs/tags/4.1.4-ti-r9 ^b16e5f7b8a29f130ebcac9311988edbcc944b361
But there is not such sha ( b16e5f7b8a29f130ebcac9311988edbcc944b361 ) in your git anymore.

Can you fix this on your side please?

RobertCNelson, 2015-08-21:
@sykrim  use the tags as master gets rebased.. And make sure you didn't do a shallow clone.

https://github.com/beagleboard/linux/commits/4.1.4-ti-r9

sykrim, 2015-08-21:
Did use the tag but when trying to clone it using poky, it says b16e5f7b8a29f130ebcac9311988edbcc944b361 does not exist anymore. Which is true, since I do not see it in the git log, but it does appear in the packed-refs file. So I guess the 4.1.4-ti-r9 tag should not reference a non-existing sha.

RobertCNelson, 2015-08-21:
That commit show up in github's interface.

https://github.com/beagleboard/linux/commit/b16e5f7b8a29f130ebcac9311988edbcc944b361

sykrim, 2015-08-21:
I have a git clone locally on my machine and git log-ing and searching for that sha returns nothing.

sykrim, 2015-08-21:
Need to mention that this is on the 4.1 branch

RobertCNelson, 2015-08-21:
Again.. The branches, 3.8/3.14/4.1 get 'rebased'... Use the tags from master...

sykrim, 2015-08-21:
There is something fishy with the git repo. I do a 

git branch -a --contains b16e5f7b8a29f130ebcac9311988edbcc944b361

and that comes up empty

sykrim, 2015-08-21:
Worked around this "tag in no branch" issue from within bitbake.
Thanks for your support

chelliwell, 2015-08-31:
@skyrim : I've also been having trouble setting yocto/bitbake to pull a specific tag - would be grateful if you'd send me the .bb that you've now got working. Thanks.

sykrim, 2015-09-01:
You can use nobranch=1 in your SRC_URI

chelliwell, 2015-09-01:
Ah ok - hadn't spotted that option. What's your full SRC_URI please?

sykrim, 2015-09-01:
SRC_URI = "git://github.com/beagleboard/linux.git;protocol=git;tag=4.1.4-ti-r9;nobranch=1"

chelliwell, 2015-09-02:
Hmm, tried that. The git fetch doesn't error now, but it seems to pull the wrong version - an expected file (drivers/misc/bone_capemgr.c) isn't there in the downloaded source.

RobertCNelson, 2015-09-02:
Does the tag option work for you guys?
```
tag = "4.1.4-ti-r9"
KERNEL_GIT_URI = "git://github.com/beagleboard/linux.git"
KERNEL_GIT_PROTOCOL = "git"
SRC_URI += "${KERNEL_GIT_URI};protocol=${KERNEL_GIT_PROTOCOL};tag=${tag}"
```

sykrim, 2015-09-02:
Robert, attach nobranch=1 to SRC_URI. See above a sample of the SRC_URI

RobertCNelson, 2015-09-02:
thanks @sykrim 
btw i have a branch setup here:
https://github.com/RobertCNelson/meta-beagleboard-kernel
I think it's still missing a couple pieces to fully intergrate into meta-ti/yocto
But i'm trying to keep it in sync with these tag's..

agherzan, 2015-09-02:
@RobertCNelson we already have it done and working at resin.io. Here is the link: https://github.com/resin-io/meta-resin/tree/master/meta-resin-beaglebone/recipes-kernel/linux

As well it was tested on board.

RobertCNelson, 2015-09-02:
@agherzan very nice! i'll point users to that repo. ;) (my *bb skills are non-existent.)

chelliwell, 2015-09-03:
@agherzan - I'm using that same SRC_URI yet it fetches ('successfully') the wrong version, not 4.1.4-ti-r9. Is there some other bb setting that could be altering/overriding it...?

agherzan, 2015-09-03:
@chelliwell Nothing at all. Those are the only resin specific layers we are using. Just did a build to check this behavior and the kernel got checked out at:

commit b16e5f7b8a29f130ebcac9311988edbcc944b361
Author: Robert Nelson <robertcnelson@gmail.com>
Date:   Mon Aug 10 10:05:09 2015 -0500
    4.1.4-ti-r9 bb.org_defconfig
    4.1 TI Delta: https://github.com/RobertCNelson/ti-linux-kernel/compare/2803cc758722bcfe8fdd2c3ee813bd5e390354b2...1f0ed8c5eb8c05d613458b8cc3c35ab28485a611    
    Signed-off-by: Robert Nelson <robertcnelson@gmail.com>

... which seems valid, right?

chelliwell, 2015-09-04:
I'll have to dig more into what git's doing then.
One thing that may be relevant: I'm using a 'linux-yocto-custom' recipe, in my own layer. Bitbake though warns:
"NOTE: preferred version 3.19% of linux-yocto not available" (three times) - probably from poky/meta-yocto/conf/distro/poky.conf?
(Also, right or wrong, this recipe inherits 'kernel' and 'recipes-kernel/linux/linux-yocto.inc')

chelliwell, 2015-09-07:
@agherzan: what did you run to get the summary above ("commit....." etc") ?

agherzan, 2015-09-07:
@chelliwell did you try to use the recipe in the layer I pointed above? As it is I mean.

chelliwell, 2015-09-07:
I copied what I thought would be the relevant settings, but I'll try that next. (Though I will have to comment out some of your 'resin' specific lines)

agherzan, 2015-09-07:
What did you find resin specific there?

chelliwell, 2015-09-07:
Mainly:

inherit kernel kernel-resin

require recipes-kernel/linux/linux-dtb.inc
require recipes-kernel/linux/setup-defconfig.inc
require recipes-kernel/linux/multi-kernel.inc

chelliwell, 2015-09-07:
I've commented out those lines, and also
  RDEPENDS_kernel-base_append = " am33x-cm3"

bitbake linux-yocto-custom reports:
NOTE: Preparing RunQueue
ERROR: Multiple .bb files are due to be built which each provide virtual/kernel (/home/colin/yocto/trunk/meta-bbb-ln/recipes-kernel/linux/linux-yocto-custom.bb /home/colin/yocto/trunk/poky/meta/recipes-kernel/linux/linux-yocto_3.14.bb).
 This usually means one provides something the other doesn't and should.
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: Failed to fetch URL http://www.openssl.org/source/openssl-1.0.2a.tar.gz, attempting MIRRORS if available
WARNING: Failed to fetch URL http://kernel.org/pub/linux/kernel/v3.0/linux-3.19.tar.xz, attempting MIRRORS if available
Currently 5 running tasks (370 of 451):
...
2: linux-yocto-3.14.36+gitAUTOINC+162dfe3bb0_dbe5b52e93-r0 do_fetch (pid 28183)
...
4: linux-yocto-custom-4.1.4+gitAUTOINC+b16e5f7b8a-b do_fetch (pid 10508)

Both fetches 'succeed', but then it seems to the 3.14 that it attempts to build

chelliwell, 2015-09-11:
Been experimenting and comparing with https://github.com/RobertCNelson/meta-beagleboard-kernel/blob/master/recipes-kernel/linux/linux-beagleboard.org_4.1.bb ;
it seems the key thing I need to change in my recipe is
require recipes-kernel/linux/linux-yocto.inc
   [as described in Yocto Linux Kernel Dev Manual (para 2.4)]
to
require recipes-kernel/linux/linux-dtb.inc

agherzan, 2015-09-23:
And it's still not working?

chelliwell, 2015-09-24:
No, after that change I now have it working.

agherzan, 2015-09-24:
Glad you sorted out. Sorry for not being able to give you more guidance but as we cannot replicate it was hard to.

Cheers!