beagleboard/linux #40: nand/omap2 module is not available if onenand/omap2 is

npf, 2015-08-17:
nand/omap2.ko is not shipped (make modules_install) if onenand/omap2 is built.
(if I'm correct, onenand/omap2 is not supposed to supersede nand/omap2 ?)

After build I have:
~~~
# /beagleboard/linux (master %=)$ find -name omap2.ko
./drivers/mtd/nand/omap2.ko
./drivers/mtd/onenand/omap2.ko
~~~
But only the onenand/omap2 module get shipped upon make modules_install
~~~
# /beagleboard/linux (master %=)$ find /lib/modules/4.1.5/ -name omap2.ko
/lib/modules/4.1.5/kernel/drivers/mtd/onenand/omap2.ko
~~~
Also, on the running system (BeagleBoard C3) the nand is not found.

@rogerq , I don't think it's fixed with your last commits. Could you confirm ?

For ref, my initial discussion with @RobertCNelson : https://github.com/RobertCNelson/linux-dev/issues/40

Thanks

rogerq, 2015-08-21:
Hi Pierre,

On 18/08/15 01:25, Pierre Neyron wrote:
> nand/omap2.ko is not shipped (make modules_install) if onenand/omap2 is built.
> (if I'm correct, onenand/omap2 is not supposed to supersede nand/omap2 ?)

No, they should be independent of each other.

> After build I have:
>
> |# /beagleboard/linux (master %=)$ find -name omap2.ko
> ./drivers/mtd/nand/omap2.ko
> ./drivers/mtd/onenand/omap2.ko
> |
>
> But only the onenand/omap2 module get shipped upon make [...]

We'll have to change the name of at least one of them. I'll send out a patch today.

> 
> Also, on the running system (BeagleBoard C3) the nand is not found.

Without nand/omap2.ko we can't expect nand to be found. Or did you mean
that with nand/omap2.ko the nand is not found?

> 
> @rogerq <https://github.com/rogerq> , I don't think it's fixed with your last commits. Could you confirm ?
> 
> For ref, my initial discussion with @RobertCNelson <https://github.com/RobertCNelson> : RobertCNelson/linux-dev#40 <https://github.com/RobertCNelson/linux-dev/issues/40>

cheers,
-roger

rogerq, 2015-08-21:
Patch posted for review
https://patchwork.kernel.org/patch/7051321/

npf, 2015-08-21:
Hi Roger,
With nand/omap2, the nand of my beagleboard c3 is found, no problem.
Thanks!

closed 2018-09-25 by RobertCNelson