beagleboard/linux #42: Kernel 4.1 compilation error

parthitce, 2015-08-23:
Hello All,

I have tried compiling the Kernel 4.1 in my Ubuntu 14.04. I have installed the cross compiler "gcc-arm-linux-gnueabihf".

**Branch:** 4.1
**Tag:** 4.1.6-ti-r11

**Config file used:**
bb.org_defconfig

**Command used:**
```bash
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- uImage dtbs LOADADDR=0x80008000 -j8
```

**Error:**
```bash
arch/arm/kernel/asm-offsets.c:54:2: error: #error Your compiler is too buggy; it is known to miscompile kernels
 #error Your compiler is too buggy; it is known to miscompile kernels
  ^
arch/arm/kernel/asm-offsets.c:55:2: error: #error and result in filesystem corruption and oopses.
 #error and result in filesystem corruption and oopses.
  ^
make[1]: *** [arch/arm/kernel/asm-offsets.s] Error 1
make: *** [prepare0] Error 2
make: *** Waiting for unfinished jobs....
```

Thanks,
Parthiban N

parthitce, 2015-08-23:
Issue seems to be related to gcc verion.

```bash
parthiban@parthiban:~$ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (Ubuntu/Linaro 4.8.2-16ubuntu4) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

parthitce, 2015-08-23:
Problem is not seen with the gcc version 4.9 from Linaro.

Linaro GCC 4.9-2015.01-3

```bash
parthiban@parthiban:/opt/tooling/gcc-linaro-4.9-2015.02-3-x86_64_arm-linux-gnueabihf/bin$ ./arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (Linaro GCC 4.9-2015.01-3) 4.9.3 20150113 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

RobertCNelson, 2015-08-23:
So... What's the issue? there's a bug in gcc 4.8.x that was just fixed in 4.8.3...

Closing...

closed 2015-08-23 by RobertCNelson