beagleboard/linux #43: exec bit should not be set on am335x-pm-firmware.bin

Enoch247, 2015-09-10:
For what its worth, firmware/am335x-pm-firmware.bin should probably not be set to executable. It is a binary blob and not something the the BBB can run. firmware/am335x-pm-firmware.elf is debatable as well. I have seen elf/so files both with and without the execute bit set, so I'm not sure what the proper convention should be.

RobertCNelson, 2015-09-10:
I'm just copying them directly from this tree:
http://git.ti.com/gitweb/?p=ti-cm3-pm-firmware/amx3-cm3.git;a=tree;f=bin;h=15fb22abcb25cf81c2d9dfcd05cad889e8f1e376;hb=refs/heads/ti-v4.1.y
So the executable setting comes from that.

Enoch247, 2015-09-10:
Ok, I understand. It may be worth notifying TI, if you think they will care. It seems like a silly thing to care about, but then again if its easy to fix...