beagleboard/linux #44: Pin 104 claimed by MMC1

traversjames, 2015-09-15:
In am335x-bone-common.dtsi pin 104 is being claimed by the MMC1 device even though it is not associated with the SDcard connector on the board schematic:
			0x1A0 (PIN_INPUT_PULLUP | MUX_MODE7)	/* mcasp0_aclkr.gpio3_18 */

This is introduced in a8d5991f00ce111e24d74abe802eccb0638deb94 where all mmc pins are initially assigned.

A guess as to why this happened is maybe confusion of the CD pin (gpio0_6) with gpio0_7. gpio0_7 is tied to this gpio3_18 pin on the board on P9_42 so maybe both pins were claimed to avoid conflict.  It doesn't seem necessary and claims a useful PRU/Mcasp resource.

jadonk, 2015-09-23:
Since that patch seems to be signed-off by several TI folks, can you send a note to the Beagle mailing list (http://beagleboard.org/discuss) and CC them? I wouldn't want to fix it without knowing why they put it in and then not having it fixed everywhere.

traversjames, 2015-09-24:
Done:
https://groups.google.com/forum/embed/?place=forum/beagleboard&showsearch=true&showpopout=true&showtabs=true&hideforumtitle=true&parenturl=http%3A%2F%2Fbeagleboard.org%2FCommunity%2FForums#!category-topic/beagleboard/EECWGnvEjQo

traversjames, 2015-09-29:
Looks like this has been approved for removal

RobertCNelson, 2015-09-29:
@traversjames i'll patch it before thursday's snapshot, unless ti's beats me to it..
Regards,

RobertCNelson, 2015-09-29:
now fixed as part of:
https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/553822023931fe2cb02168d6edb547ac811c983a

this will be part of r19 on thursday..

RobertCNelson, 2015-09-29:
nm pushing right now due to this commit in the ti diff:
https://github.com/RobertCNelson/ti-linux-kernel/commit/5bf93b966b88ff5aa668c767102d13a1ea1b6757

RobertCNelson, 2015-09-29:
all pushed out, will be updating dtb-rebuilder/bb.org-overlays in a bit, but safe to close this..

closed 2015-09-29 by RobertCNelson