beagleboard/linux #48: Current G_MULTI_* config not working for windows

skladd, 2015-10-26:
As currently both RNDIS and CDC Ethernet USB gadget are activated, Windows is not able to load drivers for BeagleBone Black with G_MULTI connected by USB:

    CONFIG_USB_G_MULTI_RNDIS=y
    CONFIG_USB_G_MULTI_CDC=y

from [Kernel Documentation](https://www.kernel.org/doc/Documentation/usb/gadget_multi.txt):
> ** Windows host drivers
>
>For the gadget two work under Windows two conditions have to be met:
>
>*** Detecting as composite gadget
>[...]
> The only thing to worry is that the gadget has to have a single configuration so a dual RNDIS and CDC ECM gadget won't work [...]

Could CONFIG_USB_G_MULTI_CDC just be disabled per default?

RobertCNelson, 2015-10-26:
Sure we can do that, in essence we'd be reverting to how we did it in 3.8, which we know works..

```
3.8:
CONFIG_USB_G_MULTI=m
CONFIG_USB_G_MULTI_RNDIS=y
# CONFIG_USB_G_MULTI_CDC is not set
```

```
4.1:
CONFIG_USB_G_MULTI=m
CONFIG_USB_G_MULTI_RNDIS=y
CONFIG_USB_G_MULTI_CDC=y
```

skladd, 2015-10-26:
Is there a reason why this was changed? I like the way how it used to work: Connect the beaglebone the first time, install the drivers from the USB mass storage and then open 192.168.7.2 in the browser...

RobertCNelson, 2015-10-26:
Well, it looks like it was changed back in the v3.14.x-ti era and nobody complained.  Even thou we'd been shipping a v3.14.x testing snapshot for a good year.  So yeah, we are getting a much larger group of testers on v4.1.x then we ever did on v3.14.x..
Regards,

closed 2015-10-26 by RobertCNelson