beagleboard/linux #49: Unable to boot linux 4.1 on beagle bone black Rev C.

goudapatil, 2015-10-30:
Hi,
I have compiled the Linux 4.1 from this repo. When I run it over beagle bone black (using tftp and nfs), Kernel stops @ "Starting Kernel..,".
And no further logs are seen.

Can you please help me.

Thanks,

RobertCNelson, 2015-10-30:
@goudapatil without a real u-boot serial log, it's hard to pin point the exact issue to 100%..

So here's my wild guess, your u-boot environment variables are for loading the kernel/dtb are wrong and thus overlapping, when loading the file into ram:
```
loadaddr=0x82000000
fdtaddr=0x88000000
rdaddr=0x88080000
initrd_high=0xffffffff
fdt_high=0xffffffff
```
Regards.

goudapatil, 2015-11-03:
@RobertCNelson ,
Following are the commands used to compile the Linux.
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabi- bb.org_defconfig 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabi- uImage dtbs LOADADDR=0x80008000

U-Boot args:
tftpboot 0x80F80000 am335x-boneblack_4.1.dtb
tftpboot 0x80007FC0 uImage_4.1
bootm 0x80007FC0 - 0x80F80000

On Uboot Promt I'm getting following log:
U-Boot# bootm 0x80007FC0 - 0x80F80000
## Booting kernel from Legacy Image at 80007fc0 ...
   Image Name:   Linux-4.1.10
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    8204536 Bytes = 7.8 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 80f80000
   Booting using the fdt blob at 0x80f80000
   XIP Kernel Image ... OK
   Using Device Tree in place at 80f80000, end 80f9159c

Starting kernel ...

Can you please let me know whether i have followed the correct procedure or not.

Thanks,

RobertCNelson, 2015-11-03:
@goudapatil 

```
tftpboot 0x80F80000 am335x-boneblack_4.1.dtb
tftpboot 0x80007FC0 uImage_4.1
bootm 0x80007FC0 - 0x80F80000
```
Yeap, they overlap... That's not going to work.. use:

```
tftpboot 0x88000000 am335x-boneblack_4.1.dtb
tftpboot 0x82000000 uImage_4.1
bootm 0x82000000 - 0x88000000
```

Regards,

goudapatil, 2015-11-04:
@RobertCNelson  It works fine now. Thank you. 

Regards,

closed 2015-11-04 by goudapatil