beagleboard/linux #56: uio_pruss

bwilcutt, 2015-12-17:
I have both the uio_pruss and pruss_remoteproc modules installed, no errors reported in the Kernel logs and yet I am not getting any entries created in /sys/class/uio.  It is supposed to show entries such as:

/sys/class/uio/uio0/maps/map0/addr
/sys/class/uio/uio0/maps/map0/size

Has something changed from 3.14 to 4.1 that would cause this?

RobertCNelson, 2015-12-17:
If you want "uio_pruss" use "4.1.x-bone", while this "4.1.x-ti" branch is developing "pruss_remoteproc" which will be replacing uio_pruss...

https://github.com/RobertCNelson/linux-stable-rcn-ee/tree/4.1.15-bone17

Regards,

c1arocque, 2016-02-12:
Hello, I got uio_pruss working by patching am33xx.dtsi. If you look at pruss: for the 4.1 ti kernel the definition is different. Simply exchange the old definition of the pruss (3.8 kernel)  for the new one and everything works.

Regards

Chris

bwilcutt, 2016-02-12:
Thanks for the reply, c1arocque.  Right now, the headache-de-jour (as is with Linux) is the pruss_remoteproc driver is complaining that mailbox entries are of the wrong version-- even though I've built the entire kernel together, pruss driver and all.

Oiy how I long for VxWorks... (with emphasis on WORKS).

RobertCNelson, 2016-02-12:
If you had it working with 3.8 and using uio_pruss, then use the 4.1-bone branch

https://github.com/RobertCNelson/linux-stable-rcn-ee/tree/4.1.17-bone19

or
https://github.com/RobertCNelson/linux-stable-rcn-ee/tree/4.1.17-bone-rt-r19

uio_pruss and pruss_remoteproc are not compatible in the same kernel build, as they each use the pruss dt node, but with different parameters..

Regards,

bwilcutt, 2016-02-12:
"uio_pruss and pruss_remoteproc are not compatible in the same kernel build, as they each use the pruss dt node, but with different parameters.."
Whoa... okay, I wouldn't have thought that... I removed uio_pruss.  At this point, no difference...
When I try inserting pruss_remoteproc.ko, I get:
[   66.747344] pruss_remoteproc: disagrees about version of symbol mbox_send_message
[   66.747384] pruss_remoteproc: Unknown symbol mbox_send_message (err -22)
[   66.747474] pruss_remoteproc: disagrees about version of symbol mbox_free_channel
[   66.747486] pruss_remoteproc: Unknown symbol mbox_free_channel (err -22)
[   66.747720] pruss_remoteproc: disagrees about version of symbol mbox_request_channel
[   66.747737] pruss_remoteproc: Unknown symbol mbox_request_channel (err -22)
That's a dumb from dmesg.  ALL of the code was built together so I don't understand how version numbers could possible differ.  This is the only module that complains like this, too.  I did a make clean, then I did a recursive rm -rf on all *.o's, etc., by hand just to be on the safe side.
mbox exists in the system, it's built, it's included in the kernel build ("*").  I've deleted the Module.symvers before building, too.
Bryan

bwilcutt, 2016-02-12:
"uio_pruss and pruss_remoteproc are not compatible in the same kernel build, as they each use the pruss dt node, but with different parameters.."
Whoa... okay, I wouldn't have thought that... I removed uio_pruss.  At this point, no difference...
When I try inserting pruss_remoteproc.ko, I get:
[   66.747344] pruss_remoteproc: disagrees about version of symbol mbox_send_message
[   66.747384] pruss_remoteproc: Unknown symbol mbox_send_message (err -22)
[   66.747474] pruss_remoteproc: disagrees about version of symbol mbox_free_channel
[   66.747486] pruss_remoteproc: Unknown symbol mbox_free_channel (err -22)
[   66.747720] pruss_remoteproc: disagrees about version of symbol mbox_request_channel
[   66.747737] pruss_remoteproc: Unknown symbol mbox_request_channel (err -22)
That's a dump from dmesg.  ALL of the code was built together so I don't understand how version numbers could possibly differ.  This is the only module that complains like this, too.  I did a make clean, then I did a recursive rm -rf on all *.o's, etc., by hand just to be on the safe side.
mbox exists in the system, it's built, it's included in the kernel build ("*").  I've deleted the Module.symvers before building, too.
Bryan

bwilcutt, 2016-02-12:
I should also mention if I force the module to be loaded (-f) it DOES load. 

    On Friday, February 12, 2016 2:29 PM, Bryan Wilcutt <bwilcutt@yahoo.com> wrote:
 

 "uio_pruss and pruss_remoteproc are not compatible in the same kernel build, as they each use the pruss dt node, but with different parameters.."
Whoa... okay, I wouldn't have thought that... I removed uio_pruss.  At this point, no difference...
When I try inserting pruss_remoteproc.ko, I get:
[   66.747344] pruss_remoteproc: disagrees about version of symbol mbox_send_message
[   66.747384] pruss_remoteproc: Unknown symbol mbox_send_message (err -22)
[   66.747474] pruss_remoteproc: disagrees about version of symbol mbox_free_channel
[   66.747486] pruss_remoteproc: Unknown symbol mbox_free_channel (err -22)
[   66.747720] pruss_remoteproc: disagrees about version of symbol mbox_request_channel
[   66.747737] pruss_remoteproc: Unknown symbol mbox_request_channel (err -22)
That's a dump from dmesg.  ALL of the code was built together so I don't understand how version numbers could possibly differ.  This is the only module that complains like this, too.  I did a make clean, then I did a recursive rm -rf on all *.o's, etc., by hand just to be on the safe side.
mbox exists in the system, it's built, it's included in the kernel build ("*").  I've deleted the Module.symvers before building, too.
Bryan

closed 2016-02-15 by bwilcutt