beagleboard/linux #64: androidboot.console=ttyO0 not working linux kernel 3.14

asebak, 2016-02-01:
this is my uEnv.txt file.  If I remove the androidboot.console=ttyO0 I see the serial output but if I leave it in it just says Starting Kernel... and nothing after that.  Do I need to configure in linux kernel or dts file  to allow the androidboot.console variable to work.  Thanks!

```
kernel_file=zImage

console=ttyO0,115200n8
#video_res=video=720x480-16@60
mmcroot=/dev/mmcblk0p2 rw
mmcrootfstype=ext4 rootwait fixrtc
 
loadkernel=load mmc ${mmcdev}:${mmcpart} ${loadaddr} ${kernel_file}
loadfdt=load mmc ${mmcdev}:${mmcpart} ${fdtaddr} /dtbs/${fdtfile}
 
boot_ftd=run loadkernel; run loadfdt

mmcargs=setenv bootargs consoleblank=0 androidboot.console=ttyO0 mem=512M acpi=noirq vt.global_cursor_default=0 mpurate=1000 ${video_res} qemu=1 console=${console} root=${mmcroot} rootfstype=${mmcrootfstype} init=/init ip=off g_ffs.idVendor=0x18d1 g_ffs.idProduct=0x4e23 g_ffs.iSerialNumber=BBBANDROID

uenvcmd=run boot_ftd; run mmcargs; bootz ${loadaddr} - ${fdtaddr}
```

RobertCNelson, 2016-02-01:
No idea, we don't have a resident android guy...

G-Ray, 2016-07-08:
Try by replacing ttyO0 by ttyS0