beagleboard/linux #66: Startup anomalies

bwilcutt, 2016-02-15:
Just migrated from 4.1.13 to 4.1.17 and during boot, experiencing:
 
 [   **] (1 of 2) A start job is running for LSB: Raise netwo...33s / no limit)[   38.430333] random: nonblocking pool is initialid
[ TIME ] Timed out waiting for device dev-ttyGS0.device.
[DEPEND] Dependency failed for Serial Getty on ttyGS0.
[     *] A start job is running for LSB: Raise network interf... 4s / no limit)[  189.953893] net eth0: initializing cpsw version )

For some reason it is trying and failing to bring up ttyGS0.  

Also, unable to set static IP even after setting the etc/network/interface file... keeps going to DHCP.  This could be connected to the above problem.

Any ideas?

RobertCNelson, 2016-02-15:
ttyGS0 is the g_serial part of g_multi, i suppose we could spawn the systemd process after g_multi loads..

if you setup a static ip, make sure you have connman removed. (it get's in the way when you aren't using dhcp)..

bwilcutt, 2016-02-15:
Robert,

I noticed g_multi and so forth seem to be in the USB sector of the OS, specifically the "legacy" directory.  I didn't see a menu option on menuconfig [I might have missed it??] so I went into the .config file and removed all legacy based USB options.

As an experiment, I used the same .config from 4.1.13, where everything worked fine, and the problem still occurred.  

On a positive note, I can say the pruss_remoteproc, rpmsg and virtio system seem to be playing nicer under 4.1.17... so far.

Rebuilding now... we'll let you know how it turns out here shortly.  Thanks!

bwilcutt, 2016-02-15:
Removing USB legacy bricked my BB so I reverted.

bwilcutt, 2016-02-15:
Robert,

I reverted all the way back to the original bb.org_defconfig and I still get the same issue...

Thanks,
Bryan

bwilcutt, 2016-02-16:
Removed connmand using:

 update-rc.d -f connmand remove

1. My static IP is still being over written by DHCP.  I can't just rm -rf DHCP off the system because some want to use it-- I don't.

2. Still having 4-minute boots with the BB:

  17.461928] pruss-rproc 4a300000.pruss: configured system_events = 0x0800000000000000 intr_channels = 0x00000002 host_intr = 0x00000002
[   17.474276]  remoteproc2: remote processor 4a338000.pru1 is now up
[   17.514979] virtio_rpmsg_bus virtio0: rpmsg host is online
[   17.520774] virtio_rpmsg_bus virtio0: creating channel rpmsg-pru addr 0x1f
[   17.553203] rpmsg_pru rpmsg0: new rpmsg_pru device: /dev/rpmsg_pru31
[**    **] (2 of 2) A start job is running for dev-ttyGS0.device (28s / 1min 30s)[   42.856060] random: nonblocking pool is initialized
[ TIME ] Timed out waiting for device dev-ttyGS0.device.
[DEPEND] Dependency failed for Serial Getty on ttyGS0.

**
Any ideas?

bwilcutt, 2016-02-16:
Update... deleted connmand... IP address is fine now!

bwilcutt, 2016-02-16:
Robert,

Removing connman did resolve the IP/DHCP issue.

Btw, my interfaces file seemed to be what was causing the long delay.  Even though I'm not using a static IP address something else is interfering with bringing up the network interface and, thus, causing the "A start job is running for LSB:" messages.

I changed my interfaces to remove "auto eth0" and that resolved the problem, however I have to bring the interface up manually [ifup] after bootup.  But when I do, it uses my static address [as it should].

Btw, using "allow-hotplug" didn't resolve the issue.  I think there's a bug somewhere because with a static IP address, there should be no waiting on a network service, especially for 4 mins!  This tells me it is still doing some kind of dhcp behind the scenes.  My ps -a does not show dhcpd, however.  Perhaps it is the order in which things are brought up as you mentioned before?  Shouldn't bringing up the network not be one of the first jobs of the OS?

bwilcutt, 2016-02-17:
Robert... more isolation.  Turns out I have an empty "stub" uio driver, it's the only UIO driver in the system.  Bringing this uio driver up causes the system to hang during boot []waiting for ttyGS0].  If I remove the driver, all's good.

Any issues going on with the UIO system?

Thanks,
Bryan

bwilcutt, 2016-02-18:
FINALLY FIXED!

Wasn't trivial to find this, lots of stuff to look at but here's the jest--

My UIO driver connects to the device tree through a standard definition.  The "compatible" in the device tree was the culprit.

Naughty:
compatible = "ti,uio_mydriver", "ti,pruss";

Nice:
compatible = "ti,uio_mydriver";

I am not using the prussdrv code ("ti,pruss"), mind you.  If I remove the "ti,pruss" from the list, wa-la, everything magically works.  The driver is happy, the systemd (worst set of code ever added to Linux IMHO) is happy, everyone is happy.  Before, it tried to add "ttyGS0" to the system before the network interface was up and there was a long waiting period for ttyGS0's driver to time out.  Why? Who knows such things?

This issue never occurred on earlier versions of BB (even as late as 4.1.13].  Don't know how people are going to use the prussdrv.ko with this issue.  Frankly, don't care since they should be using remoteproc anyway.

Bryan
"Suggested but failed alternative name for Linux: Hypophora"
"Linux is free if you don't value your time."

closed 2016-02-18 by bwilcutt