beagleboard/linux #67: G_multi

bwilcutt, 2016-02-16:
How do you  spawn the systemd process after g_multi loads?

bwilcutt, 2016-02-16:
Saw a similar issue ([DEPEND] Dependency failed for Serial Getty on ttyGS0) reported here about this already:

https://groups.google.com/forum/#!topic/beagleboard/zWZzv9ZA3Q4

However the issue was never resolved there [about 4 months ago].

RobertCNelson, 2016-02-16:
1: here's one way: https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L180

2: https://github.com/beagleboard/linux/issues/66#issuecomment-184401533

RobertCNelson, 2016-02-16:
This should take care of it..

https://github.com/RobertCNelson/boot-scripts/commit/5aefdeb6757503584dc6d31821d4178083641fdb

RobertCNelson, 2016-02-18:
sorry, actually, had to revert it...

https://github.com/RobertCNelson/omap-image-builder/commit/29287613adf86e58fca2a3b473e05ba588689b68

getty wasn't coming up when i needed it..  so more testing..

RobertCNelson, 2016-02-19:
Blah, it works in windows, so what broke in linux. ;)

bwilcutt, 2016-02-23:
Okay, so some more experimenting.  I've reverted my kernel build back to 4.1.13 but my target still has 4.1.17's boot core.  The issue is still occurring.  I was hoping it wouldn't so I didn't have to revert the core back to 4.1.13.

Looking around, I don't even know if I could revert back-- I don't see a 4.1.13 label available on GIT for making a target image.  Robert, you know where/how I can get a 4.1.13 image?  I hate going back; the reason I went forward was because I couldn't get the rpmsg, virtio and pru remoteproc to all play nice together but I'm willing to retry that effort.

Thanks,
Bryan

bwilcutt, 2016-04-27:
Closing this issue---

closed 2016-04-27 by bwilcutt