beagleboard/linux #68: Branch 4.1 does not boot on BeagleBone Black

jcdevel, 2016-02-19:
Last working tag is 4.1.17-ti-rt-r46. Tags 4.1.17-ti-rt-r47 and 4.1.17-ti-rt-r48 do not boot. The boot process freezes right after jumping to Linux, then resets (presumably a watchdog reset). This is the serial output:

	U-Boot SPL 2015.01-00001-gb2412df (Jan 29 2015 - 15:01:06)

	U-Boot 2015.01-00001-gb2412df (Jan 29 2015 - 15:01:06), Build: jenkins-github_Bootloader-Builder-105

	       Watchdog enabled
	I2C:   ready
	DRAM:  512 MiB
	MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
	Using default environment

	Net:   <ethaddr> not set. Validating first E-fuse MAC
	cpsw
	Hit any key to stop autoboot:  0 
	gpio: pin 53 (gpio 53) value is 1
	Card did not respond to voltage select!
	Card did not respond to voltage select!
	gpio: pin 56 (gpio 56) value is 0
	gpio: pin 55 (gpio 55) value is 0
	gpio: pin 54 (gpio 54) value is 0
	switch to partitions #0, OK
	mmc1(part 0) is current device
	gpio: pin 54 (gpio 54) value is 1
	Checking for: /uEnv.txt ...
	reading uEnv.txt
	1067 bytes read in 6 ms (172.9 KiB/s)
	gpio: pin 55 (gpio 55) value is 1
	Loaded environment from uEnv.txt
	Importing environment from mmc ...
	Checking if uenvcmd is set ...
	gpio: pin 56 (gpio 56) value is 1
	Running uenvcmd ...
	cpsw Waiting for PHY auto negotiation to complete. done
	link up on port 0, speed 100, full duplex
	BOOTP broadcast 1
	BOOTP broadcast 2
	BOOTP broadcast 3
	DHCP client bound to address 192.168.46.120 (763 ms)
	Checking if client_ip is set ...
	using am335x-boneblack.dtb ...
	gpio: pin 56 (gpio 56) value is 1
	Running nfsboot ...
	Booting from 192.168.46.2 ...
	link up on port 0, speed 100, full duplex
	Using cpsw device
	TFTP from server 192.168.46.2; our IP address is 192.168.46.120
	Filename 'zImage'.
	Load address: 0x82000000
	Loading: #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 #################################################################
		 ###############################
		 384.8 KiB/s
	done
	Bytes transferred = 7811736 (773298 hex)
	link up on port 0, speed 100, full duplex
	Using cpsw device
	TFTP from server 192.168.46.2; our IP address is 192.168.46.120
	Filename 'dtbs/am335x-boneblack.dtb'.
	Load address: 0x88000000
	Loading: ############
		 160.2 KiB/s
	done
	Bytes transferred = 60532 (ec74 hex)
	Kernel image @ 0x82000000 [ 0x000000 - 0x773298 ]
	## Flattened Device Tree blob at 88000000
	   Booting using the fdt blob at 0x88000000
	   Loading Device Tree to 8ffee000, end 8ffffc73 ... OK

	Starting kernel ...

	U-Boot SPL 2015.01-00001-gb2412df (Jan 29 2015 - 15:01:06)

Has something changed after r46? I am using TFTP to load the kernel and the DT.

Thanks in advance.

RobertCNelson, 2016-02-19:
So r46 -> r47:

https://github.com/RobertCNelson/ti-linux-kernel/compare/e9ab7826bd1663682a77861632b4f2685858dc31...5160d38b9145030283afb8c954235bc12ab3b2f1

https://github.com/RobertCNelson/ti-linux-kernel/compare/5160d38b9145030283afb8c954235bc12ab3b2f1...ee09b0f7977d99e84b896d6973ea70706586f147

https://github.com/RobertCNelson/ti-linux-kernel/compare/ee09b0f7977d99e84b896d6973ea70706586f147...f0420b4ab150a06db44c182a2ffb37806dc46d26

RobertCNelson, 2016-02-19:
r47 -> r48:

https://github.com/RobertCNelson/ti-linux-kernel/compare/f0420b4ab150a06db44c182a2ffb37806dc46d26...828f540311fe43d7331d54b81e1a77b53bcc8e84

RobertCNelson, 2016-02-19:
are you doing anything different in the config?

jcdevel, 2016-02-19:
Hi RobertCNelson, I have not changed the configuration. Basically, I have followed the same procedure I always have to update the kernel (replace zImage, dtb and modules). Actually, just by restoring the r46 zImage the board boots again. I have tried different DTs but I think it is something related to the kernel itself, since kernel r46 with DT from r47 also boots, but kernels r47 and r48 freeze. Image sizes are nearly the same (just a few bytes of difference).

It would be helpful if someone else with a BBB could confirm the bug or if it is something I am missing.

Thanks.

jcdevel, 2016-02-23:
Could anyone confirm the problem? Has anyone got r48 working on BBB? Thanks.

RobertCNelson, 2016-02-23:
Sorry unable to confirm the problem:

[test-bbb-3: 4.1.17-ti-rt-r48 (up 6 days, 12 hours, 42 minutes)]

Regards,

jcdevel, 2016-02-27:
Hi RobertCNelson. I have updated U-Boot to 2016.01, just in case, but still not working. I have tried also booting from uSD instead of the onboard eMMC. Nothing. These are the steps I use to build the kernel:

    $ git checkout 4.1.17-ti-rt-r48
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm clean
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm mrproper
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm bb.org_defconfig
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm -j4 zImage
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm dtbs
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm -j4 modules
    $ make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm INSTALL_MOD_PATH=$INSTALL_MOD_PATH modules_install

$INSTALL_MOD_PATH points to the root of my NFS exported filesystem. Then I use TFTP to load zImage and am335x-boneblack.dtb (as I always have). I have tried latest tag 4.1.18-ti-rt-r49, but no luck either. Any kernel newer than r46 fails.

If your board is working fine, it must be something related to my set up, I just do not know what it may be. Could it be related to the board revision? I have a BBB rev B.

Any help will be appreciated.

kamtar, 2016-02-27:
Same problem here with BBB rev C.

jcdevel, 2016-02-28:
I tried the resultant .config from [bb-kernel](https://github.com/RobertCNelson/bb-kernel/tree/am33x-rt-v4.1) and 4.1.18-ti-rt-r49 boots fine. It seems the bb.org_defconfig provided in this repo is no longer valid for the BBB. Is this planned or is it a bug? Or was bb.org_defconfig never intended to work on BBB and it has been working until r46 just by chance?

RobertCNelson, 2016-02-28:
@jcdevel bb.org_defconfig is for the bbb..

btw, do you have earlyprink enabled, i just noticed:

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/7f0a67c7c96edb2ff29b352b6366f3ddb21318d4

^^^ that was me debugging the x15, that should not be enabled..

if you have earlyprink as part of your bootargs, that might cause something..

REgards,

jcdevel, 2016-02-28:
Hi RobertCNelson. Yes, I had earlyprink enabled. After removing it from bootargs, kernel boots fine.

printesoi, 2016-03-04:
@jcdevel, if you want the kernel to work with the CONFIG_EARLY_PRINTK you shoud enable CONFIG_DEBUG_AM33XXUART1.

RobertCNelson, 2016-03-04:
@printesoi it didn't help that i had it enabled for the x15's usart's by default ;)

nicholas-zww, 2017-02-25:
hi

I have exactly same issue here. Anyone has any solution for this case?

RobertCNelson, 2017-02-25:
@nicholas-zww please include more details:

git tag
defconfig used + changes
serial boot log

Regards,

nicholas-zww, 2017-02-26:
I used release "linux-4.4.49-ti-rt-r89", and without any change, just use bb.org_defconfig, and make.
Then i get below output from serial port. It stops at "starting kernel.....", then after a while, it looks like a reset happens, and reboot again.

U-Boot SPL 2014.04-00014-g47880f5 (Apr 22 2014 - 13:23:54)
reading args
spl_load_image_fat_os: error reading image args, err - -1
reading u-boot.img
reading u-boot.img

U-Boot 2014.04-00014-g47880f5 (Apr 22 2014 - 13:23:54)

I2C:   ready
DRAM:  512 MiB
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
*** Warning - readenv() failed, using default environment

Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw, usb_ether
Hit any key to stop autoboot:  0
gpio: pin 53 (gpio 53) value is 1
Card did not respond to voltage select!
mmc0(part 0) is current device
Card did not respond to voltage select!
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
SD/MMC found on device 1
reading uEnv.txt
829 bytes read in 3 ms (269.5 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
gpio: pin 56 (gpio 56) value is 1
Running uenvcmd ...
cpsw Waiting for PHY auto negotiation to complete. done
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.3.8; our IP address is 192.168.3.123
Filename 'zImage'.
Load address: 0x82000000
Loading: T #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         #################################################################
         ###############
         314.5 KiB/s
done
Bytes transferred = 8728752 (8530b0 hex)
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.3.8; our IP address is 192.168.3.123
Filename 'am335x-boneblack.dtb'.
Load address: 0x88000000
Loading: ###########
         404.3 KiB/s
done
Bytes transferred = 55541 (d8f5 hex)
Kernel image @ 0x82000000 [ 0x000000 - 0x8530b0 ]
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Using Device Tree in place at 88000000, end 880108f4

Starting kernel ...