beagleboard/linux #72: EDMA API is broken

ViktorButskikh, 2016-03-08:
Hello.

After migration from 3.8.13 kernel to 4.1.18 on my BBB I have found that compilation of my custom kernel driver which using EDMA API is impossible.
My driver based on this classical tutorial http://processors.wiki.ti.com/index.php/EDMA_sample_test_application. But now in file linux/platform_data/edma.h absent basic function for work with EDMA like as edma_start, edma_stop, edma_alloc_channel ...
 it's happened after 0f30e8c4ca31a7131906444ee92b65c1bd4f92fd commit.

Which guide should I use for porting my application from old kenel to new ?

Thanks

jcdevel, 2016-03-09:
The DMA driver in 4.1 uses the DMA Engine API. The DMA Engine is partially documented in the kernel, but if you are looking for a porting example, you can take a look at [logi-kernel](https://github.com/fpga-logi/logi-kernel/commit/42066f774425afb196dc0f8f1ad40f450da34115).

QuickJack, 2017-04-03:
As a matter of fact, the given example from TI does not work in Kernel 4.4+. The Linux DMA Engine API does work for 4.1-ti kernels but the performance is worse compared to any other solution. However, the same code does not work for the default 4.4-bone kernels.

Does anybody know of an example how to access the EDMA driver directly and not via the Linux DMA Engine API in kernel 4.4+?

pdp7, 2020-06-09:
@QuickJack @jcdevel @ViktorButskikh Have you tried the 4.19 kernel on our current image?
https://github.com/beagleboard/Latest-Images

QuickJack, 2020-06-09:
You can find an overview of the kernels that I have tested at https://github.com/fpga-logi/logi-kernel. I guess that upgrading the kernel to a newer version might require an upgrade of the driver code as well.

pdp7, 2020-06-09:
> You can find an overview of the kernels that I have tested at https://github.com/fpga-logi/logi-kernel. I guess that upgrading the kernel to a newer version might require an upgrade of the driver code as well.

Is the driver that uses EDMA inside here?

https://github.com/fpga-logi/logi-kernel/tree/master/beaglebone-black

QuickJack, 2020-06-23:
The code contains support for different hardware platforms (Logibone R1, Logibone RA2). The main part is in https://github.com/fpga-logi/logi-kernel/blob/master/beaglebone-black/common/main_dm.c and https://github.com/fpga-logi/logi-kernel/blob/master/beaglebone-black/common/main_dma.c. The latter file contains the DMA support which is currently used by default. The actual DMA functionality is implemented in https://github.com/fpga-logi/logi-kernel/blob/master/beaglebone-black/common/logi_dma.c.

pdp7, 2020-07-08:
@ViktorButskikh are you still looking for a solution to this issue?