beagleboard/linux #73: Kernel 3.8.13 - TSC_ADC

Rado9, 2016-03-10:
Hi,
is there any chance that kernel 3.8.13 will be updated from version bone79(13th Oct 2015)? I have problem with touchscreen driver (TSC_ADC "freezing") which was updated on 6th Jan 2016 for kernel 4.1.X: https://github.com/torvalds/linux/blob/master/drivers/input/touchscreen/ti_am335x_tsc.c
Then I would like to update kernel via script "/opt/scripts/tools/update_kernel.sh"

Thanks.

RobertCNelson, 2016-03-10:
Hi @Rado9 ,

update_kernel.sh has multiple kernel branch options, based on command line options used:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Kernel_Options

Regards,

Rado9, 2016-03-10:
Hi @RobertCNelson 
I really appreciate your help and fast reaction.
I have tried these commands:
sudo apt-get update
sudo apt-cache search linux-image | grep **4.1**
sudo apt-get install linux-image-**4.1.18**

E: Unable to locate package linux-image-4.1.18
E: Couldn't find any package by regex 'linux-image-4.1.18'

Can you write me please example commands to update kernel from 3.8.13-bone79 to some kernel 4.1.X.
Thanks.

RobertCNelson, 2016-03-10:
4.1.x-ti
```
4.1.x-ti BeagleBone/BeagleBone Black/BeagleBoard-X15 + RT
--ti-rt-channel --lts-4_1
```

So:

```
cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh --ti-rt-channel --lts-4_1
```

Rado9, 2016-03-10:
What a support from you. Thanks. I am updating kernel right now.

RobertCNelson, 2016-03-10:
```
EDIT:
root@beaglebone:~# sudo apt-cache search linux-image | grep 4.x
linux-headers-3.2.0-4-ixp4xx - Header files for Linux 3.2.0-4-ixp4xx
linux-image-3.2.0-4-ixp4xx - Linux 3.2 for IXP4xx
linux-image-2.6-ixp4xx - Linux for IXP4xx (dummy package)
linux-image-ixp4xx - Linux for IXP4xx (meta-package)
```

Yuck, except your on wheezy....

You might want to also test:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_lxqt

on another sd card..

4.1.x isn't exactly a 100% drop in replacement for 3.8.13.

We have tweaks to make it compatible on a cape by cape basis..

Regards,

Rado9, 2016-03-10:
Thanks, I've just found out after reboot. :) Black screen.
Will be some update for wheezy? Or I must test Debian Jessie?

RobertCNelson, 2016-03-10:
First you need to figure out what type of black screen..

Do you have a usb serial cable?

If this is a factory original image, u-boot can't handle the size of the *.dtb so it'll hard lock on bootup.

Next, depending on u-boot version you can add:

dtb=am335x-boneblack-overlay.dtb

to /boot/uEnv.txt

https://github.com/beagleboard/bb.org-overlays/blob/master/readme.md

But, if your version of u-boot is too old..

If your happy with wheezy and don't want to switch to jessie, grab this 'last" wheezy version: 2015-11-12 

http://beagleboard.org/latest-images

it has all the u-boot changes, to easily swap between 3.8/4.1

Regards,

Rado9, 2016-03-10:
Ok thanks. I will try to recover from blackscreen. I use image from 13th May 2014(3.8.13-bone50) with kernel update to 3.8.13-bone79 with some custom stuff (mysql, java).
So, last option will be to use "last" wheezy version and install custom stuff.

Rado9, 2016-03-11:
I 've connected via usb cable and I can ping ip 192.168.7.2 and connect to BBBrevC via putty.
`uname -a
Linux beaglebone 4.1.18-ti-rt-r52 #1 SMP PREEMPT RT Wed Mar 9 21:56:40 UTC 2016 armv7l GNU/Linux`
I use 4DCAPE-70T from 4D Systems.

Rado9, 2016-03-14:
Debian Wheezy 7.5 (2014-05-14) with my custom stuff
working LCD 4DCAPE-70T
kernel with newest touchscreen driver
Is this combination still possible? 
Thanks.

Rado9, 2016-03-22:
Answer from 4D Systems Team about ADC problems:

> 

If I have understood the problem correctly, this problem is entirely rooted on the beaglebone system. The 4DCape touch connects to the beaglebone ADC directly and relies on the native LCD+touch screen driver (ti-tscadc). Please refer to datasheet for schematics. 
:: Unfortunately, we are not experts on the Beaglebone system. Only thing I could think of is that, this might be part of the limitation for the touch screen driver or incompatibility of the driver with your existing Beaglebone linux system.
:: http://processors.wiki.ti.com/index.php/AM335x_ADC_Driver's_Guide#ADC_Driver_Limitations

> 

Regards,

Rado9, 2016-03-29:
So, there is no chance for touchscreen driver update for my configuration which I mentioned above?

Rado9, 2016-04-01:
We' ve tried screen protector from Trust (originally for iPad) on LCD display, because of "multitouch reduction",  but still once per few days touch control freezes. :( This condition ist not acceptable for our customer, but I really don't know how to fix touchscreen driver for my configuration.