beagleboard/linux #77: Driver crash on boot

bwilcutt, 2016-03-25:
I have a custom driver module (uio).  If I move the module into my home directory and reboot, then load the module (insmod) manually, everything is perfect.   If I let the module exist in the lib/modules system (which is where I originally moved it from) it crashes during boot (with the long 1 1/2 minute wait delay, etc.)

I think my driver may be effecting other drivers to get loaded "before their time".  So, question-- using systemd, how the heck can I assure my driver is the last one loaded in the system?  I used to do this easy-peasy with initd.

Thanks,
Bryan

RobertCNelson, 2016-03-25:
Hi Brian,

I've been fighting something similar, with the wl18xx bluetooth service, it needs to be loaded very late..

In your .service file, use the multi-user.target:

```
After=multi-user.target
```

Regards,

bwilcutt, 2016-03-25:
Aloha Robert!

Well, I'm not using a service file for my uio_mcp3202 driver, that's the rub.  The driver is in the lib/modules directory tree because I built it with the kernel.

Okay... does this mean I should build a service file?   I'll assume so and start on it now.  

Mahalo,
Bryan

bwilcutt, 2016-03-25:
Btw, the driver is built as a M-odule, not a *-built in... why does it load inexplicitly?  Who knows such things but the Oracle?   Otherwise, I have no clue.

RobertCNelson, 2016-03-26:
Ah!  For some odd reason i thought you were already manually loading it..

For right now, a quick workaround, add your module to the blacklist

https://wiki.debian.org/KernelModuleBlacklisting

this will stop them from being automatically loaded on bootup, so you can do it yourself or thru an init script..

But it sounds like your module is missing a dependicy hint for 'depmod'..

bwilcutt, 2016-03-27:
I got that microchip mcp3202 adc UIO driver done along with the microchip encx4j600 ethernet driver.  Both work well under 4.1.15.  How can we work together to integrate?  My company MAY be willing to release into the PD.
I am working on a couple of wireless drivers, a GPRS driver and more ADCs.  I should be able to get thosE into GPL as well...
Bryan

Sent via the Samsung Galaxy Note® 4, an AT&T 4G LTE smartphone

https://wiki.debian.org/KernelModuleBlacklisting

closed 2016-04-06 by bwilcutt