beagleboard/linux #82: CONFIG_HAMRADIO turned off in the latest debian jessie image

mitchmitchell, 2016-04-18:
This removes AX25 support and prevents using the beaglebone with ham radio applications such as Paclink-Unix, Linux RMS Gateway, APRX, and TNC-Black interface to radios.

This is on is X86 versions and was on in the debian wheezy version for the beaglebone so why would it be turned off for the beaglebone now?  It is also on in the latest Raspian image for the Raspberry Pi

RobertCNelson, 2016-04-18:
@mitchmitchell that's for reporting this, i've added it back to the default config's on mainline and adding it to the v4.1.x/v4.4.x branches shortly..

https://github.com/RobertCNelson/linux-dev/commit/ef267ce5f0ea7216636b72734aeff6b5ab7ff284

Regards,

mitchmitchell, 2016-04-19:
great news, thanks for the quick response!

closed 2016-04-19 by mitchmitchell