beagleboard/linux #98: Is the fifo usb mode still patched on 4.1.x builds for the BBB

BrianAdams, 2016-08-16:
This was a fix at one time: https://github.com/Angstrom-distribution/meta-ti/blob/master/recipes-kernel/linux/linux-mainline-3.2/beagle/0011-default-to-fifo-mode-5-for-old-musb-beagles.patch

I am trying to determine if this patch was integrated in the latest ti based builds or if something changed to make it no longer necessary.

RobertCNelson, 2016-08-16:
@BrianAdams that's for the really old beagle's.. omap34xx/omap35xx, it was a silcon bug for the fifo size on the musb. I've moved the patch to a kernel parameter:

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2016.07/0001-omap3_beagle-uEnv.txt-bootz-n-fixes.patch#L59

as it's not needed on the beagle xm = dm3730, or bone/boneblack am335x...

Regards,

BrianAdams, 2016-08-16:
Thanks! Closing.

closed 2016-08-16 by BrianAdams