beagleboard/linux #99: compilation error

goudapatil, 2016-08-23:
Hi,

I am getting following compilation error for gcc (arm-linux-gnueabi-gcc)  version 4.7.3 (Ubuntu/Linaro 4.7.3-12ubuntu1).

  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
arm-linux-gnueabi-gcc: error: unrecognized command line option ‘-fstack-protector-strong’
make[1]: *** [kernel/bounds.s] Error 1
make: *** [prepare0] Error 2

which compiler version to use for compiling the top of the branch (currently i checked out 4.4 branch) code ?
The compilation was working fine for 4.1 branch.
Please help me to address this issue.,

Thanks,
Y M Patil

dlech, 2016-08-23:
The cross compiler that ships with Ubuntu is know to not work well for compiling kernels. You should use the cross compiler listed on this page instead: https://eewiki.net/display/linuxonarm/BeagleBone

RobertCNelson, 2016-08-23:
Hi @goudapatil use gcc-4.9+

https://releases.linaro.org/components/toolchain/binaries/4.9-2016.02/arm-linux-gnueabihf/gcc-linaro-4.9-2016.02-x86_64_arm-linux-gnueabihf.tar.xz

Or disable the stack-protector-strong kernel config.

Regards,

closed 2016-08-23 by goudapatil