beagleboard/linux #101: BeagleBone Black dtb with HDMI Audio disabled

bigguiness, 2016-09-20:
Hello,

I have been trying to get the latest image from BeagleBoard.org (Debian Image 2016-05-13) to work and can't seem to figure out the right combination.

My cape uses pin P5_25. With the older wheezy images I was able to get this pin by loading the overlay with HDMI and EMMC enabled (BB-BONELT-HDMIN) and then use the cape-universal overlay.

Is this possible with the jessie images?

Thanks.

RobertCNelson, 2016-09-20:
@bigguiness in /boot/uEnv.txt set:

```
dtb=am335x-boneblack-nhdmi-overlay.dtb
```

On the next reboot, cape universal will be loaded.

PS, on 2016-05-13 it still might have been too early, so grab the latest from here:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

Regards,

RobertCNelson, 2016-09-20:
@bigguiness 

Opps, i see what you meant...

HDMI no audio + eMMC.. i need to setup that config..

The issue, i couldn't get the mmc working reliably as an overlay, so it's either on or off in v4.1.x+

Regards,

bigguiness, 2016-09-20:
Thanks for looking into this.

So my only option right now is the wheezy images (3.x kernels)?

bigguiness, 2016-11-30:
Has this been fixed in any of the latest Jessie images?

Renha, 2019-01-10:
I have same problem. Any progress with the issue?

RobertCNelson, 2019-01-10:
@Renha , that was 2 years ago, "same problem" isn't valid anymore... Please try again... Please state what issue are "you" having..  Adding the output of:

```code
sudo /opt/scripts/tools/version.sh 
```

Regards,

Renha, 2019-01-11:
Thank you for answer! Ok, my problem of trying to use pins taken by HDMI audio (namely P9_28 P9_29 P9_31) may be "invalid", but "I" still need to solve it. How would I do this in 2k19?

RobertCNelson, 2019-01-11:
@Renha , you can disable hdmi audio and thus use the pis for soemthing else via, addding this line to /boot/uEnv.txt

```
disable_uboot_overlay_audio=1
```
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices

Regards,