beagleboard/linux #103: emmc failures on 4.4.9+ / 4.4.21+

floion, 2016-09-24:
Hi, on the above 2 kernels, and I assume that also on kernel versions in between these 2, we get emmc failures when using docker on a partition with a btrfs filesystem. Here is the relevant dmesg log:

  733.665535] mmcqd/1: page allocation failure: order:2, mode:0x208c020
[  733.680082] CPU: 0 PID: 80 Comm: mmcqd/1 Not tainted 4.4.21+ #1
[  733.680087] Hardware name: Generic AM33XX (Flattened Device Tree)
[  733.680128] [<c0015e09>] (unwind_backtrace) from [<c001259d>] (show_stack+0x11/0x14)
[  733.680148] [<c001259d>] (show_stack) from [<c03dba65>] (dump_stack+0x69/0x78)
[  733.680170] [<c03dba65>] (dump_stack) from [<c0100a4d>] (warn_alloc_failed+0xb5/0xf0)
[  733.680185] [<c0100a4d>] (warn_alloc_failed) from [<c0103393>] (__alloc_pages_nodemask+0x693/0x7d4)
[  733.680196] [<c0103393>] (__alloc_pages_nodemask) from [<c0103629>] (alloc_kmem_pages+0x15/0x78)
[  733.680211] [<c0103629>] (alloc_kmem_pages) from [<c011782f>] (kmalloc_order+0x13/0x20)
[  733.680223] [<c011782f>] (kmalloc_order) from [<c0117855>] (kmalloc_order_trace+0x19/0xbc)
[  733.680235] [<c0117855>] (kmalloc_order_trace) from [<c042e2e5>] (edma_prep_slave_sg+0x79/0x248)
[  733.680256] [<c042e2e5>] (edma_prep_slave_sg) from [<c05c2cf7>] (omap_hsmmc_request+0x353/0x4a4)
[  733.680270] [<c05c2cf7>] (omap_hsmmc_request) from [<c05b131f>] (mmc_start_request+0xcb/0x1ac)
[  733.680280] [<c05b131f>] (mmc_start_request) from [<c05b22f7>] (mmc_start_req+0x1eb/0x2e0)
[  733.680290] [<c05b22f7>] (mmc_start_req) from [<c05bd707>] (mmc_blk_issue_rw_rq+0x97/0x7fc)
[  733.680299] [<c05bd707>] (mmc_blk_issue_rw_rq) from [<c05be017>] (mmc_blk_issue_rq+0x1ab/0x370)
[  733.680309] [<c05be017>] (mmc_blk_issue_rq) from [<c05bf3e1>] (mmc_queue_thread+0x95/0x13c)
[  733.680323] [<c05bf3e1>] (mmc_queue_thread) from [<c00493f7>] (kthread+0xaf/0xc4)
[  733.680341] [<c00493f7>] (kthread) from [<c000ee15>] (ret_from_fork+0x11/0x1c)
[  733.692342] Mem-Info:
[  734.087941] active_anon:45429 inactive_anon:2175 isolated_anon:0
                active_file:32045 inactive_file:31560 isolated_file:0
                unevictable:0 dirty:4275 writeback:1003 unstable:0
                slab_reclaimable:7336 slab_unreclaimable:3555
                mapped:12949 shmem:2204 pagetables:425 bounce:0
                free:1012 free_pcp:139 free_cma:0
[  734.191726] Normal free:3800kB min:2764kB low:3452kB high:4144kB active_anon:181716kB inactive_anon:8700kB active_file:128208kB inactive_file:126324kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:522240kB managed:503092kB mlocked:0kB dirty:17180kB writeback:4012kB mapped:51796kB shmem:8816kB slab_reclaimable:29344kB slab_unreclaimable:14220kB kernel_stack:1568kB pagetables:1700kB unstable:0kB bounce:0kB free_pcp:716kB local_pcp:716kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[  734.348853] lowmem_reserve[]: 0 0 0
[  734.355748] Normal: 1113*4kB (UME) 159*8kB (UME) 33*16kB (UM) 8*32kB (UM) 2*64kB (UM) 2*128kB (U) 1*256kB (U) 0*512kB 0*1024kB 0*2048kB 0*4096kB 0*8192kB = 7148kB
[  734.391678] 65030 total pagecache pages
[  734.411660] 0 pages in swap cache
[  734.423080] Swap cache stats: add 0, delete 0, find 0/0
[  734.444296] Free swap  = 0kB
[  734.447204] Total swap = 0kB
[  734.459255] 130560 pages RAM
[  734.466300] 0 pages HighMem/MovableOnly
[  734.478228] 4787 pages reserved
[  734.489486] 6144 pages cma reserved
[  734.504504] edma 49000000.edma: edma_prep_slave_sg: Failed to allocate a descriptor
[  734.532734] omap_hsmmc 481d8000.mmc: prep_slave_sg() failed
[  734.548775] omap_hsmmc 481d8000.mmc: MMC start dma failure
[  734.568595] mmcblk1: unknown error -1 sending read/write command, card status 0x900
[  734.587773] blk_update_request: I/O error, dev mmcblk1, sector 2243096
[  734.614544] blk_update_request: I/O error, dev mmcblk1, sector 2243104
[  734.633186] blk_update_request: I/O error, dev mmcblk1, sector 2243112
[  734.650927] blk_update_request: I/O error, dev mmcblk1, sector 2243120
[  734.669731] blk_update_request: I/O error, dev mmcblk1, sector 2243128
[  734.685783] blk_update_request: I/O error, dev mmcblk1, sector 2243136
[  734.704844] blk_update_request: I/O error, dev mmcblk1, sector 2243144
[  734.718002] blk_update_request: I/O error, dev mmcblk1, sector 2243152
[  734.736844] blk_update_request: I/O error, dev mmcblk1, sector 2243160
[  734.753321] blk_update_request: I/O error, dev mmcblk1, sector 2243168
[  734.784384] BTRFS error (device mmcblk1p6): bdev /dev/mmcblk1p6 errs: wr 4, rd 0, flush 0, corrupt 0, gen 0

Is this a known issue? I have looked at the existing issues and could not find something related. I see some other people have stumbled upon this before: https://groups.google.com/forum/#!topic/beagleboard/5ZwTUQwGSx4

As in their case, our use case involves a lot of I/O.

Thank you

floion, 2016-09-27:
hi @RobertCNelson is this something that has affected other users too? Are you relying on the upstream mmc driver in this kernel?

floion, 2016-10-27:
ping on this @RobertCNelson

locomundo, 2016-10-31:
We are having this problem too, quite persistently.

pirvanc, 2016-10-31:
We would be very interested in a fix for this issue as well.

RobertCNelson, 2016-10-31:
Any improvement with linux-image-4.8.5-bone2 ?

jaapcrezee, 2016-10-31:
Could it somehow be related to https://cdn.kernel.org/pub/linux/kernel/v4.x/ChangeLog-4.1.29

commit 74225a4cbea38034034add5da67a2f7ee23251c0
Author: Steven Rostedt <rostedt@goodmis.org>
Date:   Thu Jul 14 17:55:21 2016 -0400

I can recall seeing this issue before.

Regards,

Jaap Crezee

adnanoner, 2016-11-01:
Following

floion, 2016-11-02:
@RobertCNelson there are no 4.8 tags. And the 4.9 tags are rc. Do you plan to do a new release?

RobertCNelson, 2016-11-02:
@floion 

the v4.9.x branch will be replacing our v4.4.x in the next quarter or so..

Here's my current progress of getting it ready..
https://groups.google.com/forum/#!category-topic/beagleboard/_TdTH7oPEXE

right now working on eqep..

Regards,

floion, 2017-03-27:
hi @RobertCNelson is the 4.9 branch now stable so that it can be used in a production environment?
Thank you

floion, 2017-03-27:
Or the 4.10 one? Which is the latest stable branch/tag available for use in production?

RobertCNelson, 2017-03-28:
@floion feel free to use the 4.4/4.4-rt/4.4-xenomai or 4.9/4.9-rt branches

willnewton, 2017-06-16:
I saw a similar failure recently with 4.4.21:

```
[ 1325.957185] mmcqd/1: page allocation failure: order:3, mode:0x208c020
[ 1325.964604] CPU: 0 PID: 79 Comm: mmcqd/1 Not tainted 4.4.21+ #1
[ 1325.964609] Hardware name: Generic AM33XX (Flattened Device Tree)
[ 1325.964649] [<c0015e09>] (unwind_backtrace) from [<c001259d>] (show_stack+0x11/0x14)
[ 1325.964667] [<c001259d>] (show_stack) from [<c03dba65>] (dump_stack+0x69/0x78)
[ 1325.964689] [<c03dba65>] (dump_stack) from [<c0100a4d>] (warn_alloc_failed+0xb5/0xf0)
[ 1325.964704] [<c0100a4d>] (warn_alloc_failed) from [<c0103393>] (__alloc_pages_nodemask+0x693/0x7d4)
[ 1325.964714] [<c0103393>] (__alloc_pages_nodemask) from [<c0103629>] (alloc_kmem_pages+0x15/0x78)
[ 1325.964730] [<c0103629>] (alloc_kmem_pages) from [<c011782f>] (kmalloc_order+0x13/0x20)
[ 1325.964741] [<c011782f>] (kmalloc_order) from [<c0117855>] (kmalloc_order_trace+0x19/0xbc)
[ 1325.964752] [<c0117855>] (kmalloc_order_trace) from [<c042e2e5>] (edma_prep_slave_sg+0x79/0x248)
[ 1325.964773] [<c042e2e5>] (edma_prep_slave_sg) from [<c05c2cef>] (omap_hsmmc_request+0x353/0x4a4)
[ 1325.964786] [<c05c2cef>] (omap_hsmmc_request) from [<c05b1317>] (mmc_start_request+0xcb/0x1ac)
[ 1325.964796] [<c05b1317>] (mmc_start_request) from [<c05b22ef>] (mmc_start_req+0x1eb/0x2e0)
[ 1325.964805] [<c05b22ef>] (mmc_start_req) from [<c05bd6ff>] (mmc_blk_issue_rw_rq+0x97/0x7fc)
[ 1325.964815] [<c05bd6ff>] (mmc_blk_issue_rw_rq) from [<c05be00f>] (mmc_blk_issue_rq+0x1ab/0x370)
[ 1325.964824] [<c05be00f>] (mmc_blk_issue_rq) from [<c05bf3d9>] (mmc_queue_thread+0x95/0x13c)
[ 1325.964839] [<c05bf3d9>] (mmc_queue_thread) from [<c00493f7>] (kthread+0xaf/0xc4)
[ 1325.964856] [<c00493f7>] (kthread) from [<c000ee15>] (ret_from_fork+0x11/0x1c)
[ 1325.964860] Mem-Info:
[ 1326.145282] active_anon:12699 inactive_anon:1067 isolated_anon:0
                active_file:44015 inactive_file:54031 isolated_file:0
                unevictable:0 dirty:2392 writeback:3374 unstable:0
                slab_reclaimable:6953 slab_unreclaimable:1743
                mapped:1795 shmem:1103 pagetables:227 bounce:0
                free:3166 free_pcp:183 free_cma:0
[ 1326.187989] Normal free:12652kB min:8192kB low:10240kB high:12288kB active_anon:50796kB inactive_anon:4268kB active_file:176060kB inactive_file:216124kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:522240kB managed:503092kB mlocked:0kB dirty:9568kB writeback:13496kB mapped:7180kB shmem:4412kB slab_reclaimable:27812kB slab_unreclaimable:6972kB kernel_stack:1192kB pagetables:908kB unstable:0kB bounce:0kB free_pcp:732kB local_pcp:732kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[ 1326.244948] lowmem_reserve[]: 0 0 0
[ 1326.248511] Normal: 2986*4kB (UME) 84*8kB (UME) 3*16kB (UE) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB 0*8192kB = 12664kB
[ 1326.263731] 99149 total pagecache pages
[ 1326.271061] 0 pages in swap cache
[ 1326.276348] Swap cache stats: add 0, delete 0, find 0/0
[ 1326.283657] Free swap  = 0kB
[ 1326.288427] Total swap = 0kB
[ 1326.293281] 130560 pages RAM
[ 1326.298250] 0 pages HighMem/MovableOnly
[ 1326.303974] 4787 pages reserved
[ 1326.309196] 6144 pages cma reserved
[ 1326.314707] edma 49000000.edma: edma_prep_slave_sg: Failed to allocate a descriptor
```

What strikes me as odd is this is doing an order 3 allocation for DMA descriptors. I estimate DMA descriptors are 40 bytes so in an order 3 allocation we could get over 800 sg entries which seems like a very large (and not very useful) number.

I wonder if possibly the cause of this is the relatively high value of max_segs (1024) in the omaphsmmc driver?

I notice there have been a couple of patches over the years that have attempted to reduce this number:

https://groups.google.com/d/msg/linux.kernel/-JRzNvTytyA/TglXstxMyusJ
https://github.com/beagleboard/meta-beagleboard/blob/master/common-bsp/recipes-kernel/linux/linux-mainline-3.8/mmc/0002-omap-hsmmc-Correct-usage-of-of_find_node_by_name.patch

willnewton, 2017-06-20:
Patch submitted upstream here:

http://marc.info/?l=linux-omap&m=149786501429968

RobertCNelson, 2017-06-26:
@willnewton 's patch is now part of 4.4.x-ti-r111 and 4.9.x-ti-r43* (r43 is still build testing, till i push out to github.com/beagleboard/linux)