beagleboard/linux #107: Internet over USB not working Beaglebone Black Rev C

sinqbad, 2016-11-05:
I'm running Windows 10 and sharing the internet over USB isn't working. I've shared the connection and set the IP and DNS to obtain automatically so that's and the drivers installed just fine. 
What works is manually `route add default gw 192.168.7.1` and `echo "nameserver 8.8.8.8" >> /etc/resolv.conf` but it doesn't save on restart. I have updated to the latest image and kernel too, but the problem persists.

RobertCNelson, 2016-11-06:
@sinqbad what image?
```
cat /etc/dogtag
```

Support for this out of the box was added on: (8/27/2016) via:

https://github.com/RobertCNelson/boot-scripts/commit/9e7e397a60a0589c5bc013102e03b61aacb9f87f

you can update to that and more fixes via:
```
cd /opt/scripts/ ; git pull
```

Regards,

sinqbad, 2016-11-06:
BeagleBoard.org Debian Image 2016-05-13

This fixed the issue. Thank you so much.

closed 2016-11-06 by sinqbad

RenathoAzevedo, 2018-01-26:
Could someone helpe me on this? I'm not getting it to work.

on Ubuntu 17.10 - artful   --> I do:

enp3s0 - my ethernet
enx6cecebab40c8 - my beaglebone black
```

iptables --table nat --append POSTROUTING --out-interface enp3s0 -j MASQUERADE
iptables --append FORWARD --in-interface enx6cecebab40c8 -j ACCEPT
echo 1 > /proc/sys/net/ipv4/ip_forward
```

on Beaglebone debian stretch: I added on interfaces:

iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.252
    network 192.168.7.0
    gateway 192.168.7.1
    dns-nameservers 8.8.8.8 8.8.4.4
    post-up /sbin/route add default gw 192.168.7.1

and internet doesn't work on beaglebone, what I'm doing wrong ?