beagleboard/linux #109: Unable to boot linux 4.9 on beaglebone green

Pillar1989, 2016-12-12:
Hi @RobertCNelson 
I use your project https://github.com/RobertCNelson/ti-linux-kernel-dev build kernel 4.9. 
But I boot from tftp server. It stop at "Starting kernel .." 
Do you know the reason ?

```
U-Boot SPL 2017.01-rc1-00003-gc480b77 (Dec 11 2016 - 22:22:42)
Trying to boot from MMC2

U-Boot 2017.01-rc1-00003-gc480b77 (Dec 11 2016 - 22:22:42 -0800)

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
Reset Source: Global external warm reset has occurred.
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
=> setenv ipaddr 192.168.199.214
=> setenv serverip 192.168.199.156
=> setenv fdtaddr 0x88000000 
=> setenv loadaddr 0x82000000 
=> setenv loadfdt 'tftpboot ${fdtaddr} am335x-bonegreen.dtb'
=> setenv loadimage 'tftpboot ${loadaddr} zImage'
=> setenv bootcmd 'run loadimage; run loadfdt; bootz ${loadaddr} - ${fdtaddr}'
=> boot
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.199.156; our IP address is 192.168.199.214
Filename 'zImage'.
Load address: 0x82000000
Loading: #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ############################################
	 376 KiB/s
done
Bytes transferred = 9219880 (8caf28 hex)
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.199.156; our IP address is 192.168.199.214
Filename 'am335x-bonegreen.dtb'.
Load address: 0x88000000
Loading: #####
	 382.8 KiB/s
done
Bytes transferred = 58947 (e643 hex)
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8ffee000, end 8ffff642 ... OK

Starting kernel ...
```

RobertCNelson, 2016-12-12:
Works here

```
U-Boot SPL 2016.11-00002-gab8be1c (Dec 07 2016 - 12:54:09)
Trying to boot from MMC1

U-Boot 2016.11-00002-gab8be1c (Dec 07 2016 - 12:54:09 -0600), Build: jenkins-github_Bootloader-Builder-490

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
Reset Source: Global external warm reset has occurred.
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
board_name=[BBG1] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
1119 bytes read in 8 ms (135.7 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
Checking if client_ip is set ...
gpio: pin 56 (gpio 56) value is 1
Running nfsboot_uname_r ...
Booting from 192.168.0.10 ...
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.0.10; our IP address is 192.168.0.157
Filename 'vmlinuz-4.9.0-rc8-ti-r9'.
Load address: 0x82000000
Loading: *#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ###############################################
	 3 MiB/s
done
Bytes transferred = 9270816 (8d7620 hex)
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.0.10; our IP address is 192.168.0.157
Filename 'dtbs/4.9.0-rc8-ti-r9/am335x-bonegreen.dtb'.
[... 423 lines cut ...]

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

beaglebone login: debian

Password: [   36.786480] omap-sham 53100000.sham: hw accel on OMAP rev 4.3
[   37.044570] omap-aes 53500000.aes: OMAP AES hw accel rev: 3.2
[   37.166304] omap-aes 53500000.aes: will run requests pump with realtime priority

Last login: Mon Dec 12 15:53:18 UTC 2016 on ttyS0
Linux beaglebone 4.9.0-rc8-ti-r9 #1 SMP Sat Dec 10 12:47:44 UTC 2016 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
debian@beaglebone:~$ uname -r
4.9.0-rc8-ti-r9
debian@beaglebone:~$ cat /proc/device-tree/model
TI AM335x BeagleBone Green
```

RobertCNelson, 2016-12-12:
Double checked v2017.01-rc1

```
U-Boot SPL 2017.01-rc1-00002-g736b695 (Dec 07 2016 - 12:54:32)
Trying to boot from MMC1

U-Boot 2017.01-rc1-00002-g736b695 (Dec 07 2016 - 12:54:32 -0600), Build: jenkins-github_Bootloader-Builder-490

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
Reset Source: Global external warm reset has occurred.
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
board_name=[BBG1] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
1119 bytes read in 8 ms (135.7 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
Checking if client_ip is set ...
gpio: pin 56 (gpio 56) value is 1
Running nfsboot_uname_r ...
Booting from 192.168.0.10 ...
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.0.10; our IP address is 192.168.0.157
Filename 'vmlinuz-4.9.0-rc8-ti-r9'.
Load address: 0x82000000
Loading: *#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ###############################################
	 3 MiB/s
done
Bytes transferred = 9270816 (8d7620 hex)
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.0.10; our IP address is 192.168.0.157
Filename 'dtbs/4.9.0-rc8-ti-r9/am335x-bonegreen.dtb'.
Load address: 0x88000000
Loading: *#####
	 2.8 MiB/s
done
Bytes transferred = 58947 (e643 hex)
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8ffee000, end 8ffff642 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-rc8-ti-r9 (root@b3-jetson-tk1-2gb) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP Sat Dec 10 12:47:44 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: TI AM335x BeagleBone Green
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
```

Pillar1989, 2016-12-12:
What's your u-boot setting ? It seems autoboot with tftpboot. But I cant do "saveenv" on  v2017.01-rc1 . or should I edit /boot/uEnv.txt ? can you share it for us ?

Pillar1989, 2016-12-13:
Wow, I found it. Just copy /nfs-uEnv.txt as uEnv.txt. set my own ip. It works.
nfs-uEnv.txt like a Easter.

Pillar1989, 2016-12-13:
Sorry,I need reopen this issue. 
It seems tftp doesnt load  initrd file. 
I use your rootfs package https://rcn-ee.com/rootfs/bb.org/testing/2016-11-27/lxqt-4gb/debian-8.6-lxqt-4gb-armhf-2016-11-27.tar.xz
I see your boot is fine.  but how should I do ? I search the Google ,I cannt find any usefull info.

```
U-Boot SPL 2017.01-rc1-00003-gc480b77 (Dec 11 2016 - 22:22:42)
Trying to boot from MMC2

U-Boot 2017.01-rc1-00003-gc480b77 (Dec 11 2016 - 22:22:42 -0800)

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
Reset Source: Global external warm reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
board_name=[BBG1] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
1025 bytes read in 7 ms (142.6 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
Checking if client_ip is set ...
gpio: pin 56 (gpio 56) value is 1
Running nfsboot_uname_r ...
Booting from 192.168.199.134 ...
link up on port 0, speed 100, full duplex
Using cpsw device
TFTP from server 192.168.199.134; our IP address is 192.168.199.133
Filename 'vmlinuz-4.9.0-rc8-ti-r9'.
Load address: 0x82000000
Loading: #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
[... 276 lines cut ...]
[    5.570139] IP-Config: Complete:
[    5.573401]      device=eth0, hwaddr=b0:d5:cc:47:00:8a, ipaddr=192.168.199.133, mask=255.255.255.0, gw=192.168.199.1
[    5.584003]      host=pillar, domain=, nis-domain=(none)
[    5.589355]      bootserver=192.168.199.134, rootserver=192.168.199.134, rootpath=
[    5.944809] VFS: Mounted root (nfs filesystem) on device 0:15.
[    5.958316] devtmpfs: mounted
[    5.962765] Freeing unused kernel memory: 1024K (c1200000 - c1300000)
[    6.040122] Starting init: /etc/init exists but couldn't execute it (error -13)
/bin/sh: 0: Can't open fixrtc
[    6.746131] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
[    6.746131] 
[    6.755334] CPU: 0 PID: 1 Comm: sh Not tainted 4.9.0-rc8-ti-r9 #15
[    6.761539] Hardware name: Generic AM33XX (Flattened Device Tree)
[    6.767703] [<c0112ea8>] (unwind_backtrace) from [<c010db58>] (show_stack+0x20/0x24)
[    6.775496] [<c010db58>] (show_stack) from [<c0721a44>] (dump_stack+0x8c/0xa0)
[    6.782759] [<c0721a44>] (dump_stack) from [<c0268724>] (panic+0x100/0x294)
[    6.789758] [<c0268724>] (panic) from [<c0140234>] (complete_and_exit+0x0/0x2c)
[    6.797102] [<c0140234>] (complete_and_exit) from [<c01402d0>] (do_group_exit+0x50/0xe0)
[    6.805231] [<c01402d0>] (do_group_exit) from [<c0140380>] (__wake_up_parent+0x0/0x30)
[    6.813193] [<c0140380>] (__wake_up_parent) from [<c0108ea0>] (ret_fast_syscall+0x0/0x3c)
[    6.821422] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
[    6.821422] 
[   48.634006] random: crng init done

```

Pillar1989, 2016-12-14:
Good news, I solved this issue.
The key is" init=/lib/systemd/systemd"
```
##Rename as: uEnv.txt to boot via nfs

##https://www.kernel.org/doc/Documentation/filesystems/nfs/nfsroot.txt

##SERVER: sudo apt-get install tftpd-hpa
##SERVER:
##SERVER: zImage boot:
##SERVER: TFTP_DIRECTORY defined in /etc/default/tftpd-hpa
##SERVER: zImage/*.dtb need to be located here:
##SERVER: TFTP_DIRECTORY/zImage
##SERVER: TFTP_DIRECTORY/dtbs/*.dtb
##SERVER:
##SERVER: uname_r boot:
##SERVER: TFTP_DIRECTORY defined in /etc/default/tftpd-hpa
##SERVER: Change TFTP_DIRECTORY to /NFSEXPORT/boot
##SERVER: TFTP_DIRECTORY/vmlinuz-${uname_r}
##SERVER: TFTP_DIRECTORY/dtbs/${uname_r}/*.dtb

##client_ip needs to be set for u-boot to try booting via nfs

client_ip=192.168.199.133

#u-boot defaults: uncomment and override where needed

server_ip=192.168.199.134
gw_ip=192.168.199.1
netmask=255.255.255.0
hostname=pillar
device=eth0
autoconf=off
root_dir=/home/pillar/ti/rootfs
nfs_options=,vers=3
nfsrootfstype=ext4 rootwait fixrtc
init=/lib/systemd/systemd
##use uname_r= only if TFTP SERVER is setup for uname_r boot:
uname_r=4.9.0-ti-r10

```
and connmen ignore eth0
```

root@beaglebone:/mnt# cd 
root@beaglebone:~# cat /etc/connman/main.conf 
[General]
PreferredTechnologies=ethernet,wifi
SingleConnectedTechnology=false
AllowHostnameUpdates=false
PersistentTetheringMode=true
NetworkInterfaceBlacklist=SoftAp0,usb0,eth0
```

closed 2016-12-14 by Pillar1989

disa6302, 2017-10-19:
I realize the image file downloaded from the Beagle bone website is of type .img.xz.

How can I convert to zImage or uImage format without using an SD Card even once?