beagleboard/linux #110: Linux kernel 4.9 wl18xx wlan0 interface isn't created. Debian stretch.

kenaaker, 2017-01-03:
I've gotten the Debian Stretch system installed and booting on two Beaglebone Green Wireless boards, but when I switch the kernel from 4.4.26-ti-r59 to 4.9.0-ti-rt-r13 the wl18xx module doesn't materialize any interface.

The modules seem to get loaded, but I don't see any messages from the wl modules in dmesg.  I've looked at the device tree a bit, but I don't know exactly what I'm looking for and there are many differences.  Here's the uname-a and dmesg output.

By the way, I had some trouble getting Stretch to work until I changed the install root file ownership back to root (it was something like 110, if I remember correctly).

`
Linux beagle-green-2 4.9.0-ti-rt-r13 #1 SMP PREEMPT RT Thu Dec 29 05:59:47 UTC 2016 armv7l GNU/Linux
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-ti-rt-r13 (root@b2-omap5-uevm-2gb) (gcc version 6.2.1 20161124 (Debian 6.2.1-5) ) #1 SMP PREEMPT RT Thu Dec 29 05:59:47 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: TI AM335x BeagleBone Green Wireless
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c13e07c0, node_mem_map df961000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 16 pages/cpu @df92b000 s33376 r8192 d23968 u65536
[    0.000000] pcpu-alloc: s33376 r8192 d23968 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=UUID=d2f545ad-6a18-4716-a6d0-e5551bbf07a2 ro rootfstype=ext4 rootwait coherent_pool=1M quiet
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 441592K/522240K available (12288K kernel code, 1009K rwdata, 4116K rodata, 1024K init, 791K bss, 31496K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0d00000   (13280 kB)
                     .init : 0xc1200000 - 0xc1300000   (1024 kB)
                     .data : 0xc1300000 - 0xc13fc6c0   (1010 kB)
                      .bss : 0xc13fe000 - 0xc14c3f48   ( 792 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000013] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000028] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000035] OMAP clocksource: timer1 at 24000000 Hz
[    0.001184] clocksource_probe: no matching clocksources found
[    0.001400] Console: colour dummy device 80x30
[    0.001449] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001452] This ensures that you still see kernel messages. Please
[    0.001455] update your kernel commandline.
[    0.046727] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.046748] pid_max: default: 32768 minimum: 301
[    0.047021] Security Framework initialized
[    0.047029] Yama: becoming mindful.
[    0.047063] AppArmor: AppArmor disabled by boot time parameter
[    0.047259] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.047267] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.048602] CPU: Testing write buffer coherency: ok
[    0.048613] ftrace: allocating 36693 entries in 108 pages
[    0.160773] CPU0: thread -1, cpu 0, socket -1, mpidr 0
[    0.160850] Setting up static identity map for 0x80100000 - 0x80100060
[    0.222788] Brought up 1 CPUs
[... 200 lines cut ...]
[    8.105570] systemd-journald[184]: Received request to flush runtime journal from PID 1
[   12.774164] using random self ethernet address
[   12.774187] using random host ethernet address
[   12.774202] using host ethernet address: 88:C2:55:56:04:BB
[   12.774210] using self ethernet address: 88:C2:55:56:04:BC
[   12.778741] usb0: HOST MAC 88:c2:55:56:04:bb
[   12.779051] usb0: MAC 88:c2:55:56:04:bc
[   12.779105] using random self ethernet address
[   12.779113] using random host ethernet address
[   12.781876] g_ether gadget: Ethernet Gadget, version: Memorial Day 2008
[   12.781894] g_ether gadget: g_ether ready
[   13.209259] g_ether gadget: high-speed config #1: CDC Ethernet (ECM)
[   15.638177] omap_rng 48310000.rng: OMAP Random Number Generator ver. 20
[   16.482007] omap-aes 53500000.aes: OMAP AES hw accel rev: 3.2
[   16.561499] omap-aes 53500000.aes: will run requests pump with realtime priority
[   16.619646] omap-sham 53100000.sham: hw accel on OMAP rev 4.3
[   18.467689] random: crng init done
[   22.124823] openafs: loading out-of-tree module taints kernel.
[   22.124850] openafs: module license 'http://www.openafs.org/dl/license10.html' taints kernel.
[   22.124856] Disabling lock debugging due to kernel taint
[   22.164677] Key type afs_pag registered
[   22.288840] enabling dynamically allocated vcaches
[   22.288859] Starting AFS cache scan...
[   25.190781] found 23 non-empty cache files (0%).
`

RobertCNelson, 2017-01-03:
@kenaaker 

That's odd, double check that you have "bb-wl18xx-firmware" installed (since v4.4.x worked, it probally was).. BTW, what change did you do to stretch to make it work?

Well, it's working on jessie, wonder if it's a gcc bug in stretch or something else..

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-ti-r13 (root@a1-imx6q-wandboard-2gb) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP Thu Dec 29 06:39:50 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: TI AM335x BeagleBone Green Wireless
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c13db600, node_mem_map df961000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 15 pages/cpu @df92c000 s31680 r8192 d21568 u61440
[    0.000000] pcpu-alloc: s31680 r8192 d21568 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=UUID=145e17fc-7427-4ceb-bc73-30e2b3066572 ro rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 442256K/522240K available (12288K kernel code, 1079K rwdata, 4080K rodata, 1024K init, 735K bss, 30832K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0d00000   (13280 kB)
      .init : 0xc1200000 - 0xc1300000   (1024 kB)
      .data : 0xc1300000 - 0xc140dd64   (1080 kB)
       .bss : 0xc140f000 - 0xc14c6db4   ( 736 kB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000015] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000031] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000039] OMAP clocksource: timer1 at 24000000 Hz
[    0.001263] clocksource_probe: no matching clocksources found
[    0.001719] Console: colour dummy device 80x30
[    0.001743] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001748] This ensures that you still see kernel messages. Please
[    0.001751] update your kernel commandline.
[    0.001771] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.046773] pid_max: default: 32768 minimum: 301
[    0.047055] Security Framework initialized
[    0.047065] Yama: becoming mindful.
[    0.047102] AppArmor: AppArmor disabled by boot time parameter
[    0.047307] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.047315] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.048462] CPU: Testing write buffer coherency: ok
[    0.048529] ftrace: allocating 36867 entries in 109 pages
[    0.151738] CPU0: thread -1, cpu 0, socket -1, mpidr 0
[    0.151802] Setting up static identity map for 0x80100000 - 0x80100060
[    0.156293] Brought up 1 CPUs
[    0.156311] SMP: Total of 1 processors activated (995.32 BogoMIPS).
[    0.156316] CPU: All CPU(s) started in SVC mode.
[... 224 lines cut ...]
[   40.245102] wlan0: send auth to 14:cc:20:1d:b6:f6 (try 1/3)
[   40.274318] wlan0: authenticated
[   40.279513] wlan0: associate with 14:cc:20:1d:b6:f6 (try 1/3)
[   40.284125] wlan0: RX AssocResp from 14:cc:20:1d:b6:f6 (capab=0x411 status=0 aid=1)
[   40.299883] wlan0: associated
[   40.300146] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   40.509637] wlcore: Association completed.
[   40.986378] Bluetooth: Core ver 2.22
[   40.986626] NET: Registered protocol family 31
[   40.986633] Bluetooth: HCI device and connection manager initialized
[   40.986659] Bluetooth: HCI socket layer initialized
[   40.986672] Bluetooth: L2CAP socket layer initialized
[   40.986719] Bluetooth: SCO socket layer initialized
[   41.008295] Bluetooth: HCI UART driver ver 2.3
[   41.008314] Bluetooth: HCI UART protocol H4 registered
[   41.008319] Bluetooth: HCI UART protocol BCSP registered
[   41.008323] Bluetooth: HCI UART protocol LL registered
[   41.008327] Bluetooth: HCI UART protocol ATH3K registered
[   41.008332] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   41.008619] Bluetooth: HCI UART protocol Broadcom registered
[   41.008625] Bluetooth: HCI UART protocol QCA registered
[   41.506210] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   41.506223] Bluetooth: BNEP filters: protocol multicast
[   41.506259] Bluetooth: BNEP socket layer initialized
```

kenaaker, 2017-01-03:
Well, you got one problem on the first go.  I didn't have bb-wl18xx-firmware installed.  So, I fixed that. The behavior changed a bit, the Bluetooth: facility showed up.  But, still no wlan, and no messages from wlcore.

As far as making Stretch work, I don't remember exactly. These days, I can try so many things so quickly, they all blur together and I have trouble remembering what the last thing I did was.

But, with blurry memory and all, I think that I was getting all sorts of odd behavior (that I don't remember the details of either) from Stretch until I looked at the ownership and permissions of the root directory "/" and found that on my sd card it was owned by 110 and not 0.  Once I changed the ownership of "/" to 0, the system started to behave.

I want to suspect something in the wl* module initial probe, but I haven't been able to get any messages of any sort to show up from the wl* modules.

Here's the pertinent output from lsmod | grep -i wl. And also the new dmesg output
```
wl18xx                100443  0
wlcore                228577  1 wl18xx
mac80211              695363  2 wl18xx,wlcore
cfg80211              590551  3 wl18xx,wlcore,mac80211
wlcore_sdio             8971  0

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-ti-rt-r13 (root@b2-omap5-uevm-2gb) (gcc version 6.2.1 20161124 (Debian 6.2.1-5) ) #1 SMP PREEMPT RT Thu Dec 29 05:59:47 UTC 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: TI AM335x BeagleBone Green Wireless
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 130560
[    0.000000] free_area_init_node: node 0, pgdat c13e07c0, node_mem_map df961000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] percpu: Embedded 16 pages/cpu @df92b000 s33376 r8192 d23968 u65536
[    0.000000] pcpu-alloc: s33376 r8192 d23968 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: console=ttyO0,115200n8 root=UUID=d2f545ad-6a18-4716-a6d0-e5551bbf07a2 ro rootfstype=ext4 rootwait coherent_pool=1M quiet
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 441592K/522240K available (12288K kernel code, 1009K rwdata, 4116K rodata, 1024K init, 791K bss, 31496K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0d00000   (13280 kB)
                     .init : 0xc1200000 - 0xc1300000   (1024 kB)
                     .data : 0xc1300000 - 0xc13fc6c0   (1010 kB)
                      .bss : 0xc13fe000 - 0xc14c3f48   ( 792 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=2 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000015] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000029] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000038] OMAP clocksource: timer1 at 24000000 Hz
[    0.001186] clocksource_probe: no matching clocksources found
[    0.001400] Console: colour dummy device 80x30
[    0.001448] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001451] This ensures that you still see kernel messages. Please
[    0.001454] update your kernel commandline.
[    0.046720] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.046741] pid_max: default: 32768 minimum: 301
[    0.047012] Security Framework initialized
[    0.047019] Yama: becoming mindful.
[... 228 lines cut ...]
[   22.879989] openafs: module license 'http://www.openafs.org/dl/license10.html' taints kernel.
[   22.879995] Disabling lock debugging due to kernel taint
[   22.944675] Key type afs_pag registered
[   23.136349] enabling dynamically allocated vcaches
[   23.136368] Starting AFS cache scan...
[   27.047465] found 23 non-empty cache files (0%).
[   31.403953] Bluetooth: Core ver 2.22
[   31.404166] NET: Registered protocol family 31
[   31.404177] Bluetooth: HCI device and connection manager initialized
[... cut ...]
[   31.428112] Bluetooth: HCI UART protocol BCSP registered
[   31.428118] Bluetooth: HCI UART protocol LL registered
[   31.428124] Bluetooth: HCI UART protocol ATH3K registered
[   31.428130] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   31.428453] Bluetooth: HCI UART protocol Broadcom registered
[   31.428462] Bluetooth: HCI UART protocol QCA registered
[   31.775615] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   31.775633] Bluetooth: BNEP filters: protocol multicast
[   31.775670] Bluetooth: BNEP socket layer initialized

```

kenaaker, 2017-01-03:
Just a bit additional information.  I'm switching between kernels by just editing the uname_r value in /boot/uEnv.txt.  Everything else is untouched. Same stretch image. Could be problematical, but it seemed to minimize the failure possibilities.

kenaaker, 2017-01-04:
For variety, I decided to try the Jessie image again with the 4.9.0-ti-rt-r13 version.

I started with the bone-debian-8.6-iot-armhf-2016-12-18-4gb.img.xz image file, put it onto a fresh sd card, and booted it.

Once the fresh image was running, I ran through the connmanctl commands to configure the wlan0, so I could get the 4.9.0 image installed. Then did an apt-get update; and an
```
apt-get install linux-firmware-image-4.9.0-ti-rt-r13 linux-headers-4.9.0-ti-rt-r13 linux-image-4.9.0-ti-rt-r13
```
Once that was finished, I rebooted the system and got the same failure that I'm seeing on stretch.  The wlan0 interface isn't created, no wl18xx or wlcore messages show up in dmesg, even though the modules are loaded.

Ken

kenaaker, 2017-01-06:
Now, just to confuse things further. This morning I powered up the BBGW with stretch and the 4.9 kernel on it (after having been off all night) and the wl18xx wlan0 came up.  Then, after I verified that it was running 4.9.0-ti-rt-r13, I powered it down and powered it back up again (pressed the power button on the board and waited for all the leds to go out).  After the power cycle, the wl18xx wlan0 doesn't work again.

While I was thinking about that I installed 4.9.0-ti-rt-r13 on the other BBGW I have (stretch too) and power cycled that board. That one works through multiple power cycles.

One of the biggest disappointments in my life was when I figured out that computers aren't deterministic. 

Ken

kenaaker, 2017-01-15:
Well, I've tracked the failure fairly deep into the bowels of the system, but I still haven't found the fundamental cause.  wlcore is where the failure shows up, seemingly a timeout in a call to wl12xx_sdio_power_restore() which calls mmc_sdio_init_card() which calls mmc_select_card().
The mmc_select_card() call seems to be the operation that times out, but I don't know why yet.

And the second BBGW still is flawless. It comes up every time.

kenaaker, 2017-04-16:
I have a bit of an update.  No resolution through...  On the BBGW that generally works, I've captured a failure where the entire adapter goes down.  So far, I haven't researched recovery options.  If I reboot the BBGW, the adapter comes up again and works for a while.  The failure generally occurs after some hours or days.  There isn't a lot of traffic on the interface. (In the latest manifestation, 10Mb had been received and 1.3Mb had been transmitted). The failing messages from the system log are
```
[128218.147550] wl1271_sdio mmc2:0001:2: sdio write failed (-110)
[128218.394977] wlcore: ERROR firmware boot failed despite 3 retries

Here is an additional set of messages from dmesg output after the failure.
[128218.146498] ------------[ cut here ]------------
[128218.146557] WARNING: CPU: 0 PID: 620 at drivers/net/wireless/ti/wlcore/sdio.c:145 wl12xx_sdio_raw_write+0xf4/0x1c8 [wlcore_sdio]
[128218.146564] Modules linked in: joydev sg st sr_mod cdrom cmac bnep hci_uart btbcm btqca bluetooth arc4 wl18xx wlcore mac80211 pru_rproc pruss_intc cfg80211 pruss rfkill omap_aes_driver crypto_engine omap_sham pruss_soc_bus openafs(PO) omap_rng rng_core cdc_acm wlcore_sdio evdev tps65217_charger usb_f_ecm g_ether usb_f_rndis u_ether libcomposite uio_pdrv_genirq uio ip_tables x_tables
[128218.146666] CPU: 0 PID: 620 Comm: irq/54-s-wl18xx Tainted: P           O    4.9.13-ti-rt-r26 #1
[128218.146670] Hardware name: Generic AM33XX (Flattened Device Tree)
[128218.146710] [<c0113280>] (unwind_backtrace) from [<c010dda0>] (show_stack+0x20/0x24)
[128218.146729] [<c010dda0>] (show_stack) from [<c0751c30>] (dump_stack+0x80/0x94)
[128218.146743] [<c0751c30>] (dump_stack) from [<c013d6f0>] (__warn+0xf8/0x110)
[128218.146753] [<c013d6f0>] (__warn) from [<c013d820>] (warn_slowpath_null+0x30/0x38)
[128218.146766] [<c013d820>] (warn_slowpath_null) from [<bf06e2b8>] (wl12xx_sdio_raw_write+0xf4/0x1c8 [wlcore_sdio])
[128218.147031] [<bf06e2b8>] (wl12xx_sdio_raw_write [wlcore_sdio]) from [<bf39969c>] (wl1271_ps_elp_wakeup+0x160/0x250 [wlcore])
[128218.147186] [<bf39969c>] (wl1271_ps_elp_wakeup [wlcore]) from [<bf38a9ac>] (wlcore_irq_locked+0x74/0x7b0 [wlcore])
[128218.147323] [<bf38a9ac>] (wlcore_irq_locked [wlcore]) from [<bf38b3b0>] (wlcore_irq+0xd0/0x184 [wlcore])
[128218.147401] [<bf38b3b0>] (wlcore_irq [wlcore]) from [<c01adef0>] (irq_thread_fn+0x2c/0x64)
[128218.147412] [<c01adef0>] (irq_thread_fn) from [<c01ae248>] (irq_thread+0x154/0x22c)
[128218.147429] [<c01ae248>] (irq_thread) from [<c01629bc>] (kthread+0xfc/0x114)
[128218.147447] [<c01629bc>] (kthread) from [<c0109080>] (ret_from_fork+0x14/0x34)
[128218.147530] ---[ end trace 0000000000000002 ]---
[128218.147550] wl1271_sdio mmc2:0001:2: sdio write failed (-110)
```

And the output from uname -a
root@beagle-green-1:~# uname -a
Linux beagle-green-1 4.9.13-ti-rt-r26 #1 SMP PREEMPT RT Wed Mar 29 23:16:43 UTC 2017 armv7l GNU/Linux

Ken Aaker