beagleboard/linux #113: libnl build fails because of DIV_ROUND_UP

BrianSidebotham, 2017-02-02:
When using buildroot and the 4.4 branch kernel, the build fails with some errors along the lines of:

    In file included from ../include/netlink-private/netlink.h:48:0,
                 from handlers.c:28:
    /home/brian/kantar/5135-os/build/next/buildroot-2016.11.2/output/host/usr/arm-buildroot-linux-gnueabihf/sysroot/usr/include/linux/ethtool.h:1186:8: error: variably modified 'queue_mask' at file scope
      __u32 queue_mask[DIV_ROUND_UP(MAX_NUM_QUEUE, 32)];
            ^~~~~~~~~~
This is caused when any userspace application includes `linux/ethtool.h`

I'll create a pull request for a fix and let me know if it's appropriate.

BrianSidebotham, 2017-02-09:
Ping, hello?

RobertCNelson, 2017-02-10:
@BrianSidebotham sorry for the delay, this will be part of r88

BrianSidebotham, 2017-02-10:
@RobertCNelson @jadonk Thanks guys!

BrianSidebotham, 2017-02-10:
@RobertCNelson r88 builds fine now, thanks!

closed 2017-02-10 by BrianSidebotham