beagleboard/linux #115: Building a kernel module for BBB having 3.8.13-bone70

MutturajSK, 2017-02-05:
Hi team,

Recently i got my BBB. I am trying to build a simple Hello world kernel module. How can i set up build environment either on BBB or my host machine (ubuntu 14.04) for **3.8.13-bone70**. Linux headers or source code are unavailable for **3.8.13-bone70**.

RobertCNelson, 2017-02-05:
@MutturajSK those headers are available in our jessie/wheezy image.  I've added them back to the trusty repo:

```
sudo apt-get update
sudo apt-get install linux-headers-3.8.13-bone70
```

Use yakbuild: https://github.com/RobertCNelson/yakbuild

```
git clone https://github.com/RobertCNelson/yakbuild
cd yakbuild
cp recipe.sh.sample recipe.sh
```
Then in recipe.sh set:
```
kernel_tag="3.8.13-bone70"
toolchain="gcc_linaro_gnueabihf_4_7"
```
Then just:
```
./build_kernel.sh
```

Regards,

MutturajSK, 2017-02-05:
@RobertCNelson Thanks! I am able to compile LKM on BBB :-) .

 Can i use **yakbuild** for cross compilation on my host machine??