beagleboard/linux #118: BeagleBone Blue BT_EN is not an LED

dlech, 2017-03-01:
I was checking out the LEDs and noticed that there is an LED named `wl18xx_bt_en` that is not actually an LED. It is defined in [am335x-boneblue-wl1835.dtsi](https://github.com/beagleboard/linux/blob/4.4/arch/arm/boot/dts/am335x-boneblue-wl1835.dtsi#L17). Was this an LED in previous revisions of the board? I have rev A2.

RobertCNelson, 2017-03-01:
@dlech it's not a true led, we just utilize the led subsystem for a quick method to toggle a pin.

since there currently isn't a useful ti wlink8 bluetooth binding (yet)..

Take a look the bb-wl18xx-bluetooth startup script:

https://github.com/rcn-ee/repos/blob/master/bb-wl18xx-firmware/suite/jessie/debian/bb-wl18xx-bluetooth#L40-L47

Regards,

jadonk, 2017-03-02:
On BeagleBone Blue A1 it actually was an LED, but using another GPIO pin for a software configurable WIFI LED seemed more functional for end users. For BeagleBone Black Wireless, an LED for BT_EN continues to exist.

I'm not aware of any equivalent to phy0assoc for Bluetooth. The Bluetooth status is not currently reflected in an LED on BeagleBone Blue.