beagleboard/linux #119: BeagleBone Blue doesn't power off

jadonk, 2017-03-02:
I'll find a serial cable around here at some point to catch the log, but this should be pretty easy to reproduce.

jadonk, 2017-06-07:
@RobertCNelson will this finally get us a proper shutdown on Blue?

SandPox, 2018-02-17:
Kernel panic on shutdown, afaik this happen on kernel 4.9-rt, 4.14-rt, don't know if other version is also affected, this is the log:

`[ 4560.445248] rtc_power_off failed, bailing out.
[ 4560.486394] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[ 4560.486394]
[ 4560.486422] CPU: 0 PID: 1 Comm: systemd-shutdow Not tainted 4.14.19-ti-rt-r35 #1
[ 4560.486431] Hardware name: Generic AM33XX (Flattened Device Tree)
[ 4560.486521] [<c0113d0c>] (unwind_backtrace) from [<c010deb4>] (show_stack+0x20/0x24)
[ 4560.486566] [<c010deb4>] (show_stack) from [<c0f6bd40>] (dump_stack+0x80/0x94)
[ 4560.486602] [<c0f6bd40>] (dump_stack) from [<c01415e0>] (panic+0x100/0x28c)
[ 4560.486640] [<c01415e0>] (panic) from [<c0147788>] (complete_and_exit+0x0/0x2c)
[ 4560.486681] [<c0147788>] (complete_and_exit) from [<c016a8e8>] (SyS_reboot+0x1c0/0x240)
[ 4560.486723] [<c016a8e8>] (SyS_reboot) from [<c0108ea0>] (ret_fast_syscall+0x0/0x5c)`