beagleboard/linux #126: git tag 4.9.27-ti-r33 missing

floion, 2017-06-09:
Hi,
I noticed tag 4.9.27-ti-r33 is missing. Was there an issue with that particular tag or are these tags moving targets and we should not rely on using them?

RobertCNelson, 2017-06-09:
@floion not sure what happened on that one.. As May 8th, 2017 was a monday.. So i can't blame it on forgetting to push on a weekend.  Looks like just got distracted at day job.

For backup, you can always grab the same tag from this repo:

https://github.com/RobertCNelson/linux-stable-rcn-ee

Which actually pre-dates the beagleboard repo, and has almost every tag i've ever released:

https://github.com/RobertCNelson/linux-stable-rcn-ee/tree/4.9.27-ti-r33

^note, since there are over 4K tag's github's web gui just can't handle the list, so you need to manually put the tag after "./tree/"

the other difference between the beagleboard repo, the reference config is named differently:

rcn-ee_defconfig instead of bb.org_defconfig

This is because my kernel *.deb build farm, uses https://github.com/RobertCNelson/linux-stable-rcn-ee/ directly when building the final *.deb's..

Regards,

RobertCNelson, 2017-06-09:
@floion just another quick note for reference. ;)

Pushing to https://github.com/beagleboard/linux/ is manual, (so if i forget or get distracted, it'll happen)

Where as pushing to https://github.com/RobertCNelson/linux-stable-rcn-ee/ is automated and the "only" way my build farm will kick off a kernel *.deb build..

Regards,

floion, 2017-06-09:
Thanks for the explanation.
Let me also clarify: tag 4.9.27-ti-r33 used to exist and now it does not exist anymore.
It's not a big deal that this particular tag is missing now, I was just wondering if these tags are supposed to be volatile or we can safely rely on them in the future.

RobertCNelson, 2017-06-09:
@floion oh, that's not normal..  I don't delete any of the pushed tag's..  I'm cloning the linux repo right now, to see if we lost all, or just some are broken..

Regards,

RobertCNelson, 2017-06-09:
@floion 

weird:

```
voodoo@hades:/mnt/data/data/repo/linux$ cat .git/config 
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/beagleboard/linux
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "4.4"]
	remote = origin
	merge = refs/heads/4.4
voodoo@hades:/mnt/data/data/repo/linux$ git tag --list | grep 4.9.27
4.9.27-ti-r33
4.9.27-ti-rt-r33
```
the weblink works:

https://github.com/beagleboard/linux/tree/4.9.27-ti-r33

must have been a github web-hookup...

Regards,

floion, 2017-06-10:
Yeah, it seems to be okay now.
Thanks for looking at this.

closed 2017-06-27 by floion