beagleboard/linux #131: irq/184-musb-hd High load CPU

kuzinnv, 2017-07-24:
Hi! First, I apologize for my bad English. I use Beaglebone Black with installed Ubuntu 16.04.2 LTS with PREEMPT RT kernel.
`uname -a`
`Linux arm 4.4.68-ti-rt-r111 #1 SMP PREEMPT RT Wed Jun 28 15:12:37 UTC 2017 armv7l armv7l armv7l GNU/Linux`
In USB host connected DAC Schiit Bifrost. As a player I use mpd-0.20.6 with RT-patch. When playback is started, process irq/184-musb-hd causes a high CPU load. Screenshot is attached. Please help me to solve this problem.
P.S. `modprobe musb_hdrc use_dma=n` does not help.
[image]

RobertCNelson, 2017-07-24:
The high load is due to DMA disabled on musb:

CONFIG_MUSB_PIO_ONLY=y

You can rebuild the kernel with MUSB's dma re-enabled, just be aware some usb "devices" don't work well with it.

Regards

kuzinnv, 2017-07-24:
> You can rebuild the kernel with MUSB's dma re-enabled, just be aware some usb "devices" don't work well with it.

The following command shows that DMA is on!
`zgrep ^CONFIG_MUSB_PIO_ONLY /proc/config.gz`
`CONFIG_MUSB_PIO_ONLY=y`

RobertCNelson, 2017-07-24:
Incorrect, CONFIG_MUSB_PIO_ONLY means DMA is disabled. (we have it enabled by default as it makes usb more reliable)

https://cateee.net/lkddb/web-lkddb/MUSB_PIO_ONLY.html

```
All data is copied between memory and FIFO by the CPU. DMA controllers are ignored.

Do not choose this unless DMA support for your SOC or board is unavailable (or unstable). When DMA is enabled at compile time, you can still disable it at run time using the "use_dma=n" module parameter.

```
Regards,

kuzinnv, 2017-07-24:
> you can still disable it at run time using the "use_dma=n" module parameter

I just did it. Nothing changed.

RobertCNelson, 2017-07-24:
Actually you didn't..

```
zgrep ^CONFIG_MUSB_PIO_ONLY /proc/config.gz
CONFIG_MUSB_PIO_ONLY=y
```
+
```
use_dma=n
```

Doesn't do anything, as MUSB_PIO_ONLY "already" disabled dma...

Regards,

kuzinnv, 2017-07-24:
Sorry, I made a mistake.
`use_dma=y`

RobertCNelson, 2017-07-24:
You actually need do:

```
-CONFIG_MUSB_PIO_ONLY=y
+# CONFIG_MUSB_PIO_ONLY is not set
+CONFIG_USB_TI_CPPI41_DMA=y
```
(USB_TI_CPPI41_DMA is removed with MUSB_PIO_ONLY enabled)

Regards,

kuzinnv, 2018-01-19:
> You actually need do:
> `-CONFIG_MUSB_PIO_ONLY=y
> +# CONFIG_MUSB_PIO_ONLY is not set
> +CONFIG_USB_TI_CPPI41_DMA=y`
It did not help me.

machinekoder, 2018-07-14:
Any news on this topic? I have exactly the same problem.