beagleboard/linux #134: 4.9 kernel UART access Unhandled fault: : 1028 [#2] THUMB2

tim-seoss, 2017-07-31:
Consistently when opening ttyO4 (ttyS4) on:

Linux version 4.9.40-bone6 (root@b2-omap5-uevm-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 Fri Jul 28 14:40:08 UTC 2017
~# cat /proc/cmdline 
console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno=BB-SPIDEV0,BB-SPIDEV1,BB-UART1,BB-UART4,BB-UART5 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 cape_universal=enable

Unhandled fault: external abort on non-linefetch (0x1028) at 0xfa1a8008
pgd = dac74000
[fa1a8008] *pgd=48011452(bad)
Internal error: : 1028 [#2] THUMB2
Modules linked in: spidev omap_aes crypto_engine omap_sham omap_rng rng_core spi_omap2_mcspi evdev tps65217_charger usb_f_acm u_serial omap_wdt usb_f_ecm usb_f_rndis u_ether libcomposite leds_gpio uio_pdrv_genirq uio cpufreq_dt ip_tables x_tables
CPU: 0 PID: 794 Comm: lt-ModemManager Tainted: G      D         4.9.40-bone6 #1
Hardware name: Generic AM33XX (Flattened Device Tree)
task: daac7240 task.stack: dc5b2000
PC is at mem_serial_in+0xe/0x18
LR is at omap_8250_pm+0x29/0x90
pc : [<c0564bce>]    lr : [<c05681cd>]    psr: 000f0033
sp : dc5b3d58  ip : 00000000  fp : daa37000
r10: 024000c0  r9 : 00400044  r8 : 00000000
r7 : dad17000  r6 : 00000000  r5 : daa34b20  r4 : c0f916a8
r3 : 00000002  r2 : fa1a8000  r1 : 00000008  r0 : c0f916a8
Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment none
Control: 50c5387d  Table: 9ac74019  DAC: 00000051
Process lt-ModemManager (pid: 794, stack limit = 0xdc5b2210)
Stack: (0xdc5b3d58 to 0xdc5b4000)
3d40:                                                       c0f916a8 daa34b20
3d60: daa34b9c c056362b daa34b20 dad17000 daa34b9c dc490e40 dc5b2000 c0553371
3d80: daa34800 000000c8 c086feac c0f79fc8 dc5b2000 c05629a5 c0562981 dc490e40
3da0: dad17000 c054e01d c0213709 00000982 daa3aa00 c0e0384c dad17000 c059f889
3dc0: 00000004 bc6dc714 00000000 c0e0384c dabd18c0 00000000 daa3aa00 dc490e40
3de0: c086ff28 dc490e40 dc5b3f68 c0213cf7 00000000 bc6dc714 00000000 dc490e40
3e00: daa3aa00 dc490e48 dadf1c80 c0213c89 00000000 c020edc1 00000000 dc5b3e7c
3e20: 00260800 00000000 00000982 dc5b3eb8 dc490e40 c021a4d5 00000000 7fffffff
3e40: 00000006 db802dd0 00000000 00000000 c0e0384c 00000041 dc5b2000 00000000
3e60: 00000000 00000000 ffffffff ffffffff 00000000 00000002 daa3aa00 daab9810
3e80: dbe5f000 00004000 d99cedc0 bc6dc714 dc5b2000 c0e0384c daac7240 dc5b3f68
3ea0: 00000001 fffff000 dc5b2000 00000000 0067c1f0 c021bc33 daab9810 dbe5f000
3ec0: dc812dde 00000005 db3f5015 00000000 00000000 dbf1dee0 daa3aa00 00000101
3ee0: 00000002 000005a6 00000000 00000000 00000000 dc5b3ef8 c0e17a38 c01448dd
3f00: daac75a0 c022525d 40002000 c02047b3 00000000 c02047b3 ffffffff bc6dc714
3f20: db3f5000 00000009 db3f5000 00000000 dc3eaf40 00000002 ffffff9c c022525d
3f40: 00000982 bc6dc714 ffffff9c 00000009 c0e0384c ffffff9c db3f5000 c020f9bb
3f60: 00000000 00000051 00000982 be970000 00000006 00000100 00000001 bc6dc714
3f80: c0106964 006960d8 00000000 004c85c8 00000005 c0106964 dc5b2000 00000000
3fa0: 0067c1f0 c01067a1 006960d8 00000000 0067c1f0 00000982 a3597c00 00696058
3fc0: 006960d8 00000000 004c85c8 00000005 be9729cc be972a54 b681c79c 0067c1f0
3fe0: 00000000 be9729a0 00000000 b6c01c10 000f0030 0067c1f0 00000000 00000000
[<c0564bce>] (mem_serial_in) from [<c05681cd>] (omap_8250_pm+0x29/0x90)
[<c05681cd>] (omap_8250_pm) from [<c056362b>] (uart_startup.part.3+0x33/0x100)
[<c056362b>] (uart_startup.part.3) from [<c0553371>] (tty_port_open+0x51/0x88)
[<c0553371>] (tty_port_open) from [<c05629a5>] (uart_open+0x25/0x2c)
[<c05629a5>] (uart_open) from [<c054e01d>] (tty_open+0xbd/0x484)
[<c054e01d>] (tty_open) from [<c0213cf7>] (chrdev_open+0x6f/0xe8)
[<c0213cf7>] (chrdev_open) from [<c020edc1>] (do_dentry_open+0x14d/0x230)
[<c020edc1>] (do_dentry_open) from [<c021a4d5>] (path_openat+0x2c5/0xc48)
[<c021a4d5>] (path_openat) from [<c021bc33>] (do_filp_open+0x4f/0x94)
[<c021bc33>] (do_filp_open) from [<c020f9bb>] (do_sys_open+0xf3/0x174)
[<c020f9bb>] (do_sys_open) from [<c01067a1>] (ret_fast_syscall+0x1/0x4c)
Code: f890 304d 6842 4099 (5c50) b2c0
---[ end trace 0367fb15b78ea764 ]---

Similar backtrace occurs with:

screen  /dev/ttyS4 115200,cs8,-parenb,cstopb,-hupcl,clocal

Can't reproduce the issue with...
Linux version 4.4.79-bone18 (root@a5-imx6q-wandboard-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 Sat Jul 29 06:12:55 UTC 2017

:~# dpkg -S /boot/vmlinuz-4.9.40-bone6 /boot/vmlinuz-4.4.79-bone18 
linux-image-4.9.40-bone6: /boot/vmlinuz-4.9.40-bone6
linux-image-4.4.79-bone18: /boot/vmlinuz-4.4.79-bone18

RobertCNelson, 2017-07-31:
@tim-seoss 

Can you verify with u-boot overlays?

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
```
root@beaglebone:~# uname -r
4.9.40-bone7
```
```
enable_uboot_overlays=1

uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-UART1-00A0.dtbo 
uboot_overlay_addr3=/lib/firmware/BB-UART4-00A0.dtbo 
uboot_overlay_addr4=/lib/firmware/BB-UART5-00A0.dtbo 

disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
```

i tried your screen command, no opps, but nothing was connected to usart 4..
```
screen /dev/ttyS4 115200,cs8,-parenb,cstopb,-hupcl,clocal
```

```
U-Boot SPL 2017.07-00002-g61c3ee0fb5 (Jul 19 2017 - 17:02:08)
Trying to boot from MMC1

U-Boot 2017.07-00002-g61c3ee0fb5 (Jul 19 2017 - 17:02:08 -0500), Build: jenkins-github_Bootloader-Builder-591

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global external warm reset has occurred.
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw
Press SPACE to abort autoboot in 2 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
2074 bytes read in 34 ms (59.6 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.9.40-bone7 ...
8013360 bytes read in 542 ms (14.1 MiB/s)
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
loading /boot/dtbs/4.9.40-bone7/am335x-boneblack-uboot.dtb ...
52731 bytes read in 63 ms (817.4 KiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/AM335X-20-00A0.dtbo ...
378 bytes read in 80 ms (3.9 KiB/s)
uboot_overlays: loading /lib/firmware/BB-SPIDEV0-00A0.dtbo ...
1235 bytes read in 92 ms (12.7 KiB/s)
uboot_overlays: loading /lib/firmware/BB-SPIDEV1-00A0.dtbo ...
1235 bytes read in 221 ms (4.9 KiB/s)
uboot_overlays: loading /lib/firmware/BB-UART1-00A0.dtbo ...
883 bytes read in 251 ms (2.9 KiB/s)
uboot_overlays: loading /lib/firmware/BB-UART4-00A0.dtbo ...
883 bytes read in 353 ms (2 KiB/s)
uboot_overlays: loading /lib/firmware/BB-UART5-00A0.dtbo ...
883 bytes read in 85 ms (9.8 KiB/s)
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1105 bytes read in 196 ms (4.9 KiB/s)
uboot_overlays: uboot loading of [/lib/firmware/BB-HDMI-TDA998x-00A0.dtbo] disabled by /boot/uEnv.txt [disable_uboot_overlay_video=1]...
uboot_overlays: loading /lib/firmware/BB-ADC-00A0.dtbo ...
695 bytes read in 315 ms (2 KiB/s)
uboot_overlays: cape universal disabled, external cape enabled or detected...
loading /boot/initrd.img-4.9.40-bone7 ...
5988918 bytes read in 415 ms (13.8 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet] ...
debug: [bootz 0x82000000 0x88080000:5b6236 88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8fa49000, end 8ffff236 ... OK
   reserving fdt memory region: addr=88000000 size=6e000
   Loading Device Tree to 8f9d8000, end 8fa48fff ... OK

Starting kernel ...
```

Regards,

tim-seoss, 2017-08-23:
Seems to work when using u-boot overlays instead, thanks.

closed 2017-08-23 by tim-seoss