beagleboard/linux #136: can bus

zdcgit, 2017-08-10:
I use linux-4.4, and generate kernel module, then copy to BBB.
load can module:
sudo modprobe can
sudo modprobe can-dev
sudo modprobe can-raw
then ifconfig -a
there is no can0/can1.
Please give me some advice，thanks!

battlesnake, 2017-08-29:
Did you load the device tree overlays BB-DCANx?

zdcgit, 2017-08-29:
I have loaded it, but can not find can0/can1.

battlesnake, 2017-08-29:
If you read the `slots` file, the overlays are definitely listed there, but there are no CAN interfaces listed when you then do `ip link show`?

dmitry7887, 2017-09-12:
I have the same issue, but on 4.9.49.Used this instructions http://www.embedded-things.com/bbb/enable-canbus-on-the-beaglebone-black/.  But I have not  /sys/devices/bone_capemgr.  How can I add can0 support?

battlesnake, 2017-09-12:
If you're using a mainline kernel, you'll need to add CAN to the platform devicetree which is passed to the kernel via u-boot.

dmitry7887, 2017-09-12:
I used beaglebone/linux@4.9 kernel. Could you please describe what I need to add to uEnv.txt to boot dcan1 ?

RobertCNelson, 2017-09-12:
@dmitry7887 for v4.1.x+ it's been: /sys/devices/platform/bone_capemgr/slots

dmitry7887, 2017-09-12:
@RobertCNelson thank you!
I noticed this dir. But I have an issue:
`root@beaglebone:~# echo BB-DCAN1 > /sys/devices/platform/bone_capemgr/slots
-sh: can't create /sys/devices/platform/bone_capemgr/slots: Permission denied`

RobertCNelson, 2017-09-12:
@dmitry7887 that means u-boot is in control:

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes

Regards,

dmitry7887, 2017-09-14:
@RobertCNelson thank you!
Fixed by adding in defconfig
`CONFIG_BONE_CAPEMGR=y
CONFIG_BEAGLELOGIC=m
CONFIG_TIEQEP=m
CONFIG_EEPROM_AT24=y
CONFIG_EEPROM_AT25=y
CONFIG_EEPROM_LEGACY=m
CONFIG_EEPROM_MAX6875=m
CONFIG_EEPROM_93XX46=m
CONFIG_CAPE_BONE_ARGUS=y
CONFIG_BEAGLEBONE_PINMUX_HELPER=y`
And now I can use 
`echo 'BB-CAN1' > /sys/devices/platform/bone_capemgr/slots` and see can0 interface