beagleboard/linux #139: unable to find slots on debian stretch on beaglebone black

troywweber7, 2017-09-26:
I'm unable to find the capemgr slots file under debian stretch image on the beaglebone black. Usually it is under `/sys/devices/platform/bone_capemgr` but all I see is

```bash
$ ls -la /sys/devices/platform/bone_capemgr/
total 0
drwxr-xr-x  3 root root    0 Sep 26 22:10 .
drwxr-xr-x 21 root root    0 Sep 26 22:10 ..
lrwxrwxrwx  1 root root    0 Sep 26 22:10 driver -> ../../../bus/platform/drivers/bone_capemgr
-rw-r--r--  1 root root 4096 Sep 26 22:10 driver_override
-r--r--r--  1 root root 4096 Sep 26 22:10 modalias
lrwxrwxrwx  1 root root    0 Sep 26 22:10 of_node -> ../../../firmware/devicetree/base/bone_capemgr
drwxr-xr-x  2 root root    0 Sep 26 22:10 power
lrwxrwxrwx  1 root root    0 Sep 26 22:10 subsystem -> ../../../bus/platform
-rw-r--r--  1 root root 4096 Sep 26 22:10 uevent
```

RobertCNelson, 2017-09-26:
@troywweber7 this is a recent change, now when U-Boot has taken care of the overlays, the "slots" file is fully disabled.

Take a look at the options available to you in /boot/uEnv.txt via:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

Regards,

troywweber7, 2018-02-22:
thanks @RobertCNelson. So does that mean that `cape_enable=bone_capemgr.enable_partno=univ-all,BB-ADC,BB-PWM0,BB-PWM1,BB-PWM2` still works as intended? Is that line no longer necessary?

RobertCNelson, 2018-02-22:
@troywweber7 

All 5 of those are enabled out of the box with the images now shipping under:

https://beagleboard.org/latest-images

For the pwm's make sure to reference https://github.com/beagleboard/bb.org-overlays/blob/master/examples/cape-unversal-pwm.txt and the config-pin pwm <pin#>..

If things aren't working on first bootup, run:
```
sudo /opt/scripts/tools/version.sh
```
and we will debug it..

Regards,

troywweber7, 2018-02-22:
Well, a program I wrote that uses bonescript that was previously working on the following pins is now producing these error messages:

```
info: No pinmux for P8_33
info: No pinmux for P8_46
info: No pinmux for P8_36
info: No pinmux for P8_34
info: No pinmux for P8_29
info: No pinmux for P8_27
```

I'm assuming I need to enable some feature on the pin at boot before the program can start up...

troywweber7, 2018-02-22:
@RobertCNelson The output of `sudo /opt/scripts/tools/version.sh`:

```
git:/opt/scripts/:[89f2e9309c2322c15e2bb6b55afd214313d94842]
eeprom:[A335BNLT00C04417BBBK1CB0]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.416355] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.423576] gpio-of-helper ocp:cape-universal: ready
END
```

RobertCNelson, 2018-02-22:
```
info: No pinmux for P8_33 -> hdmi
info: No pinmux for P8_46 -> hdmi
info: No pinmux for P8_36 -> hdmi
info: No pinmux for P8_34 -> hdmi
info: No pinmux for P8_29 -> hdmi
info: No pinmux for P8_27 -> hdmi
```
Open /boot/uEnv.txt add uncomment:

```
disable_uboot_overlay_video=1
```
Regards,

troywweber7, 2018-02-22:
@RobertCNelson I was just about to draw that conclusion, thank you! It does seem like this `config-pin` utility is a lot simpler to work with. Cheers!

closed 2018-02-22 by troywweber7

grossman83, 2018-05-05:
I'm well aware that I'm asking a stupid question, but I'm afraid I'm at the beginning of the learning curve regarding the BBB and there is so much old and outdated information on the BBB it's hard for me to find the correct, and most importantly current, way of enabling certain functionality on boot, or even once running.

My situation is that I would like to enable a device tree overlay that was written by alexanderhaim that enables RTS and CTS on uart2. I think the overlay came with my beaglebone image (BeagleBoard.org Debian Image 2018-01-28)  or it's possible I downloaded it but it and many others are already compiled into .dtbo files found in /lib/firmware. I would like to enable BB-UART2-RTSCTS-00A0.dtbo. I've tried to edit the /boot/uEnv.txt file to enable this cape but it doesn't seem to be working. Any help / pointers would be greatly appreciated.

Here is what my uEnv.txt looks like currently. How should it be modified to enable the overlay mentioned above?

Note that I'm enabling cape_universal. Is that overwriting my uart overlay?

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.78-ti-r94
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
uboot_overlay_addr4=/lib/firmware/BB-UART2-RTSCTS-00A00.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/BB-UART2-RTSCTS-00A00.dtbo
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=
#MG 17APR2018 commented out next line and added rtscts
#cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4
#cape_enable=bone.capemgr.enable_partno=BB-UART2-RTSCTS-00A0.dtbo
#MG 17APR2018 hopefully enabled RTSCTS

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

mostafa222, 2018-08-30:
hi. i have a question. i want to enable uart1 and use uart2 pins(txd & rxd) as gpio. I enable uart1 by overlay BB-UART1-00A0.dtbo , and i can send data successfully but I can't use uart2 pint(txd & rxd) as gpio. if i don't use BB-UART1-00A0.dtbo, uart2 pin works as gpio. when i use BB-UART1-00A0.dtbo, i can't use uart2 pin as gpio . i can't find relationship between BB-UART1-00A0.dtbo and uart2 pin. please help me. thanks