beagleboard/linux #140: gpio group for gpio and pwm pins within sysfs

troywweber7, 2017-09-27:
Browsing the sysfs I find that you must be root to play with any of the pins set up in sysfs because all the ownership is set as `root:root`. It would be ideal if, like the Raspberry Pi, the sysfs gpio and pwm representations all had the ownership of `root:gpio` and then anyone added to the gpio group could work more easily with the gpio and pwm pins. I've found some hacky solutions (which are more difficult to implement for pwm), but it seems that this behavior should built into the kernel.

Below is a script I have that helps set the permissions, but I need to work on it for the pwm pins because it seems you have to enable capes and export values for certain folders to start showing up...

```bash
#!/bin/bash
# 
# Copy or link this file in /bin/
# 
# This script modifies files/folders in the sysfs to provide access to non-root
# users to the gpio and other pins on the beaglebone. The user must simply
# be added to the 'gpio' group to access.

GROUP=root:gpio

# provide user with access to the cape manager
echo chown ${GROUP} /sys/devices/platform/bone_capemgr/slots
chown ${GROUP} /sys/devices/platform/bone_capemgr/slots
echo chmod ug+wr /sys/devices/platform/bone_capemgr/slots
chmod ug+wr /sys/devices/platform/bone_capemgr/slots

# allow user to write to the usr# leds on the board
echo chown -R ${GROUP} /sys/class/leds/beaglebone:green:usr*/
chown -R ${GROUP} /sys/class/leds/beaglebone:green:usr*/
echo chmod -R ug+wr /sys/class/leds/beaglebone:green:usr*/
chmod -R ug+wr /sys/class/leds/beaglebone:green:usr*/

# below is taken and adapted from the original script:
# https://github.com/cnobile2012/RobotControl/blob/master/contrib/fix_udev_gpio.sh

GPIO_LOC=/sys/class/gpio

echo chown -R ${GROUP} ${GPIO_LOC}
chown -R ${GROUP} ${GPIO_LOC}

echo
echo chmod 2775
find ${GPIO_LOC} -type d -exec echo {} \; -exec chmod 2775 {} \;

echo
echo chmod ug+wr
find ${GPIO_LOC} -name "direction" -exec echo {} \; -exec chmod ug+wr {} \;
find ${GPIO_LOC} -name "edge" -exec echo {} \; -exec chmod ug+wr {} \;
find ${GPIO_LOC} -name "value" -exec echo {} \; -exec chmod ug+wr {} \;
find ${GPIO_LOC} -name "active_low" -exec echo {} \; -exec chmod ug+wr {} \;

echo
echo chmod ug+w
echo ${GPIO_LOC}/export
chmod ug+w ${GPIO_LOC}/export
echo ${GPIO_LOC}/unexport
chmod ug+w ${GPIO_LOC}/unexport

echo
echo chmod ug+wr
find ${GPIO_LOC} -name "uevent" -exec echo {} \; -exec chmod ug+wr {} \;
find ${GPIO_LOC} -name "autosuspend_delay_ms" -exec echo {} \; -exec chmod ug+wr {} \;
find ${GPIO_LOC} -name "control" -exec echo {} \; -exec chmod ug+wr {} \;

# experimental permissions to allow PWM control

#PWM_LOC=/sys/devices/platform/ocp/*pwm*
#
#echo
#echo chown -R ${GROUP} ${PWM_LOC}
#chown -R ${GROUP} ${PWM_LOC}
#
#echo
#echo chmod ug+wr
#find ${PWM_LOC} -name "export" -exec echo {} \; -exec chmod ug+wr {} \;
#find ${PWM_LOC} -name "enable" -exec echo {} \; -exec chmod ug+wr {} \;
#find ${PWM_LOC} -name "duty_cycle" -exec echo {} \; -exec chmod ug+wr {} \;
#find ${PWM_LOC} -name "period" -exec echo {} \; -exec chmod ug+wr {} \;
#find ${PWM_LOC} -name "polarity" -exec echo {} \; -exec chmod ug+wr {} \;
#find ${PWM_LOC} -name "uevent" -exec echo {} \; -exec chmod ug+wr {} \;
```

If this cannot be baked into the kernel, maybe someone can improve this script to properly provide gpio group access to the pwm pins...

RobertCNelson, 2017-09-27:
@troywweber7

What image are you using?

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[fac16ce5790a9b98fe69edd11c1e41f31685990c]
eeprom:[A335PBGL00A21736GPB30111]
dogtag:[BeagleBoard.org Debian Image 2017-09-26]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00003-g11d92ba68a]
kernel:[4.4.88-ti-r128.2]
nodejs:[v6.11.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/PB-I2C2-RTC-6-CLICK.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/PB-SPI0-ETH-CLICK.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170927.0-0rcnee1~stretch+20170927]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```
gpio: root:gpio

```
debian@beaglebone:~$ ls -lha /sys/class/gpio/
total 0
drwxrwxr-x  2 root gpio    0 Sep 27 18:24 .
drwxr-xr-x 55 root root    0 Sep 27 18:24 ..
-rw-rw----  1 root gpio 4.0K Sep 27 18:24 export
lrwxrwxrwx  1 root gpio    0 Sep 27 18:24 gpio110 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio110
lrwxrwxrwx  1 root gpio    0 Sep 27 18:24 gpio111 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio111
lrwxrwxrwx  1 root gpio    0 Sep 27 18:24 gpio112 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio112
lrwxrwxrwx  1 root gpio    0 Sep 27 18:24 gpio113 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio113
lrwxrwxrwx  1 root gpio    0 Sep 27 18:24 gpio114 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio114
<>
```

RobertCNelson, 2017-09-27:
pwm is another issue, it's fixed in v4.11.x+ kernels...

RobertCNelson, 2017-09-27:
led:

```
debian@beaglebone:~$ ls -lha /sys/class/leds/
total 0
drwxrwxr-x  2 root gpio 0 Sep 27 18:24 .
drwxr-xr-x 55 root root 0 Sep 27 19:32 ..
lrwxrwxrwx  1 root gpio 0 Sep 27 18:24 beaglebone:green:usr0 -> ../../devices/platform/leds/leds/beaglebone:green:usr0
lrwxrwxrwx  1 root gpio 0 Sep 27 18:24 beaglebone:green:usr1 -> ../../devices/platform/leds/leds/beaglebone:green:usr1
lrwxrwxrwx  1 root gpio 0 Sep 27 18:24 beaglebone:green:usr2 -> ../../devices/platform/leds/leds/beaglebone:green:usr2
lrwxrwxrwx  1 root gpio 0 Sep 27 18:24 beaglebone:green:usr3 -> ../../devices/platform/leds/leds/beaglebone:green:usr3
```

RobertCNelson, 2017-09-27:
Here's teh pwm patch:
https://github.com/RobertCNelson/bb-kernel/blob/am33x-v4.11/patches/drivers/pwm/0001-pwm-Create-device-class-for-pwm-channels.patch

currently i haven't back-ported it to v4.9.x or v4.4.x, as i don't want to break bonescript/adafruit libs'..

RobertCNelson, 2017-09-27:
heres the udev rules:

https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/stretch/debian/80-gpio-noroot.rules
https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/stretch/debian/81-pwm-noroot.rules
https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/stretch/debian/82-gpio-config-pin.rules
Regards,

bigguiness, 2018-01-20:
Is it possible to also add changing the /dev/gpiochip* group to root:gpio so that the libgpiod tools would work without sudo?

RobertCNelson, 2018-01-20:
Sure I'll give it a try today.

RobertCNelson, 2018-01-20:
@bigguiness 

```
debian@bborg-stretch-v8-18:~$ gpioinfo
gpiochip3 - 32 lines:
	line   0:  "GMII1_COL"       unused   input  active-high 
	line   1:  "GMII1_CRS"       unused   input  active-high 
	line   2: "GMII1_RXER"       unused   input  active-high 
	line   3: "GMII1_TXEN"       unused   input  active-high 
	line   4: "GMII1_RXDV"       unused   input  active-high 
	line   5:   "I2C0_SDA"       unused   input  active-high 
	line   6:   "I2C0_SCL"       unused   input  active-high 
	line   7:       "EMU0"       unused   input  active-high 
	line   8:       "EMU1"       unused   input  active-high 
	line   9: "GMII1_TXCLK" unused input active-high 
	line  10: "GMII1_RXCLK" unused input active-high 
	line  11:         "NC"       unused   input  active-high 
	line  12:         "NC"       unused   input  active-high 
	line  13: "USB1_DRVVBUS" unused input active-high 
	line  14: "MCASP0_ACLKX" unused input active-high 
	line  15: "MCASP0_FSX"       unused   input  active-high 
	line  16: "MCASP0_AXR0" "P9_30" input active-high [kernel]
	line  17: "MCASP0_AHCLKR" unused input active-high 
	line  18: "MCASP0_ACLKR" "P9_92" input active-high [kernel]
	line  19: "MCASP0_FSR"      "P9_27"   input  active-high [kernel]
	line  20: "MCASP0_AXR1" "P9_91" input active-high [kernel]
	line  21: "MCASP0_AHCLKX" unused input active-high 
	line  22:         "NC"       unused   input  active-high 
	line  23:         "NC"       unused   input  active-high 
	line  24:         "NC"       unused   input  active-high 
	line  25:         "NC"       unused   input  active-high 
	line  26:         "NC"       unused   input  active-high 
	line  27:         "NC"       unused   input  active-high 
	line  28:         "NC"       unused   input  active-high 
	line  29:         "NC"       unused   input  active-high 
	line  30:         "NC"       unused   input  active-high 
	line  31:         "NC"       unused   input  active-high 
gpiochip2 - 32 lines:
	line   0:  "GPMC_CSN3"       unused   input  active-high 
	line   1:   "GPMC_CLK"      "P8_18"   input  active-high [kernel]
	line   2: "GPMC_ADVN_ALE" "P8_07" input active-high [kernel]
	line   3: "GPMC_OEN_REN" "P8_08" input active-high [kernel]
	line   4:   "GPMC_WEN"      "P8_10"   input  active-high [kernel]
	line   5: "GPMC_BE0N_CLE" "P8_09" input active-high [kernel]
	line   6:  "LCD_DATA0"       unused   input  active-high 
	line   7:  "LCD_DATA1"       unused   input  active-high 
	line   8:  "LCD_DATA2"       unused   input  active-high 
	line   9:  "LCD_DATA3"       unused   input  active-high 
	line  10:  "LCD_DATA4"       unused   input  active-high 
	line  11:  "LCD_DATA5"       unused   input  active-high 
	line  12:  "LCD_DATA6"       unused   input  active-high 
	line  13:  "LCD_DATA7"       unused   input  active-high 
	line  14:  "LCD_DATA8"       unused   input  active-high 
	line  15:  "LCD_DATA9"       unused   input  active-high 
	line  16: "LCD_DATA10"       unused   input  active-high 
	line  17: "LCD_DATA11"       unused   input  active-high 
	line  18: "GMII1_RXD3"       unused   input  active-high 
	line  19: "GMII1_RXD2"       unused   input  active-high 
	line  20: "GMII1_RXD1"       unused   input  active-high 
	line  21: "GMII1_RXD0"       unused   input  active-high 
	line  22:  "LCD_VSYNC"       unused   input  active-high 
	line  23:  "LCD_HSYNC"       unused   input  active-high 
	line  24:   "LCD_PCLK"       unused   input  active-high 
	line  25: "LCD_AC_BIAS_EN" unused input active-high 
	line  26:  "MMC0_DAT3"       unused   input  active-high 
	line  27:  "MMC0_DAT2"       unused   input  active-high 
	line  28:  "MMC0_DAT1"       unused   input  active-high 
	line  29:  "MMC0_DAT0"       unused   input  active-high 
	line  30:   "MMC0_CLK"       unused   input  active-high 
	line  31:   "MMC0_CMD"       unused   input  active-high 
[... 60 lines cut ...]
	line  26:  "GPMC_AD10"      "P8_14"   input  active-high [kernel]
	line  27:  "GPMC_AD11"      "P8_17"   input  active-high [kernel]
	line  28: "GMII1_TXD0"       unused   input  active-high 
	line  29: "RMII1_REFCLK" unused input active-high 
	line  30: "GPMC_WAIT0"      "P9_11"   input  active-high [kernel]
	line  31:   "GPMC_WPN"      "P9_13"   input  active-high [kernel]
debian@bborg-stretch-v8-18:~$ ls -lha /dev/gpiochip*
crw-rw---- 1 root gpio 254, 0 Jan 20 22:06 /dev/gpiochip0
crw-rw---- 1 root gpio 254, 1 Jan 20 22:06 /dev/gpiochip1
crw-rw---- 1 root gpio 254, 2 Jan 20 22:06 /dev/gpiochip2
crw-rw---- 1 root gpio 254, 3 Jan 20 22:06 /dev/gpiochip3
```
```
debian@bborg-stretch-v8-18:~$ cat /etc/udev/rules.d/85-gpio-noroot.rules 
# /etc/udev/rules.d/85-gpio-noroot.rules
#
# Corrects sys GPIO permissions on the board so non-root users in the gpio group can manipulate bits
#
# Change group to gpio
SUBSYSTEM=="gpio", PROGRAM="/bin/sh -c '/bin/chown -R root:gpio /dev/gpiochip*'"
# Change user permissions to ensure user and group have read/write permissions
SUBSYSTEM=="gpio", PROGRAM="/bin/sh -c '/bin/chmod -R ug+rw /dev/gpiochip*'"
```

adding to bb-customization packages right now.. ;)

bigguiness, 2018-01-21:
Strange. With the bone-debian-9.3-lxqt-2gb-armhf-2018-01-14-2gb.img.xz image, after the apt upgrade and a reboot, the dev nodes look like this:

debian@beaglebone:~$ ls /dev/gpio* -la
crw-rw---- 1 root gpio 254, 0 Jan 20 23:47 /dev/gpiochip0
crw-rw---- 1 root gpio 254, 1 Jan 20 23:47 /dev/gpiochip1
crw-rw---- 1 root root 254, 2 Jan 20 23:47 /dev/gpiochip2
crw------- 1 root root 254, 3 Jan 20 23:47 /dev/gpiochip3
debian@beaglebone:~$ gpioinfo
gpioinfo: error accessing gpiochip gpiochip3: Permission denied

Hartley

RobertCNelson, 2018-01-22:
Hartley,

Are you using U-boot Or Kernel overlays:

```
sudo /opt/scripts/tools/version.sh
```
What you see could make sense with kernel overlays..

Regards,

bigguiness, 2018-01-22:
Following is the serial console output from U-boot:

Regards,
Hartley

U-Boot SPL 2018.01-00002-gcc9c2d1992 (Jan 09 2018 - 14:46:27)
Trying to boot from MMC2

U-Boot 2018.01-00002-gcc9c2d1992 (Jan 09 2018 - 14:46:27 -0600), Build: jenkins-github_Bootloader-Builder-26

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54: /lib/firmware/BB-BONE-NH7C-01-A0.dtbo [0xe3f55df]
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 2 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
Card did not respond to voltage select!
mmc_init: -95, time 13
Card did not respond to voltage select!
mmc_init: -95, time 12
Card did not respond to voltage select!
mmc_init: -95, time 13
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
Card did not respond to voltage select!
mmc_init: -95, time 12
Card did not respond to voltage select!
mmc_init: -95, time 12
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
2061 bytes read in 27 ms (74.2 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.9.76-ti-r91 ...
9961280 bytes read in 694 ms (13.7 MiB/s)
uboot_overlays: [uboot_base_dtb=am335x-boneblack-uboot.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblack-uboot.dtb ...
loading /boot/dtbs/4.9.76-ti-r91/am335x-boneblack-uboot.dtb ...
61174 bytes read in 73 ms (818.4 KiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/BB-BONE-NH7C-01-A0.dtbo ...
4292 bytes read in 151 ms (27.3 KiB/s)
uboot_overlays: loading /lib/firmware/BB-BONE-eMMC1-01-00A0.dtbo ...
1105 bytes read in 255 ms (3.9 KiB/s)
uboot_overlays: loading /lib/firmware/BB-ADC-00A0.dtbo ...
695 bytes read in 40 ms (16.6 KiB/s)
uboot_overlays: [uboot_detected_capes=BB-BONE-NH7C-01,] ...
uboot_overlays: cape universal disabled, external cape enabled or detected...
loading /boot/initrd.img-4.9.76-ti-r91 ...
6176864 bytes read in 420 ms (14 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-NH7C-01, coherent_pool=1M net.ifnames=0 quiet] ...
debug: [bootz 0x82000000 0x88080000:5e4060 88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8fa1b000, end 8ffff060 ... OK
   reserving fdt memory region: addr=88000000 size=70000
   Loading Device Tree to 8f9a8000, end 8fa1afff ... OK

Starting kernel ...

bigguiness, 2018-01-22:
Robert,

Must be a race condition somewhere.

I moved the chunk in generic-startup.sh that fixes the gpiochip* to the end of the script and they now are correct.

BTW, what is triggering this script? A systemd service? Also, is the udev rule still needed if this script is doing the changes?

Hartley

RobertCNelson, 2018-01-22:
There's a systemd service that calls that script:
```
generic-board-startup.service
```
We probally don't need both, but till we figure out a better udev event to trigger on for /dev/gpiochip* it'll work...

Regards,

RobertCNelson, 2018-01-22:
We trigger off SUBSYSTEM, 
```
debian@bborg-buster-v8-19:~$ udevadm info /dev/gpiochip3
P: /devices/platform/ocp/481ae000.gpio/gpiochip3
N: gpiochip3
E: DEVNAME=/dev/gpiochip3
E: DEVPATH=/devices/platform/ocp/481ae000.gpio/gpiochip3
E: MAJOR=254
E: MINOR=3
E: OF_COMPATIBLE_0=ti,omap4-gpio
E: OF_COMPATIBLE_N=1
E: OF_FULLNAME=/ocp/gpio@481ae000
E: OF_NAME=gpio
E: SUBSYSTEM=gpio
E: USEC_INITIALIZED=7631930
E: net.ifnames=0
```

amilcarlucas, 2020-08-26:
I still have this issue with https://rcn-ee.com/rootfs/2020-04-09/elinux/ubuntu-18.04.4-console-armhf-2020-04-09.tar.xz
I get:
```
ll /sys/class/gpio

total 0
drwxr-xr-x  2 root root    0 Aug 25 23:15 ./
drwxr-xr-x 62 root root    0 Aug 25 23:15 ../
--w-------  1 root root 4096 Aug 25 23:15 export
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpio12 -> ../../devices/platform/ocp/44e07000.gpio/gpiochip0/gpio/gpio12/
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpio13 -> ../../devices/platform/ocp/44e07000.gpio/gpiochip0/gpio/gpio13/
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpiochip0 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpiochip0/
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpiochip32 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpiochip32/
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpiochip64 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpiochip64/
lrwxrwxrwx  1 root root    0 Aug 25 23:15 gpiochip96 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpiochip96/
--w-------  1 root root 4096 Aug 25 23:15 unexport
```
As you can see the files still belong to root:root instead of root:gpio
and the chmod of the *export files should be 0220 instead of the current 0200

RobertCNelson, 2020-08-26:
@amilcarlucas , feel free to update "bb-customizations" and report back..

```
sudo apt update
sudo apt install --only-upgrade bb-customizations
sudo reboot
```

and report back..

Regards,

amilcarlucas, 2020-08-26:
Thanks, I opened a PR to fix the export issueshttps://github.com/beagleboard/customizations/pull/11
I hope that is the corerect repo to do it ?