beagleboard/linux #141: Git history incoherent

embexus, 2017-09-30:
I'm using directly this repository to build for with yocto, I think there are problem how git is used because the references are changed from time to time which lead to incoherence to find an old refrence !
For example I used the branch v4.4 with commit 834d157fbba62aecddb1af4b9312b218eb0a2de1 now you cannot it's missing from that branch !
maybe you are using git push --force ?!!!

dlech, 2017-09-30:
This repo contains a stack of patches on top of the mainline linux kernel. It is rebased frequently. If you need something that won't change history, use a tag instead of a branch.

embexus, 2017-10-02:
I see, but it's not the way how git works ! Git is a SCM for tacking and controlling changes and not to losing them !!
You should rethink maybe your branching workflow and use own branches naming for beaglebone v4.x-bbb for example and merge mainline kernel to them.

RobertCNelson, 2017-10-02:
@embexus okay good to know, i guess i'll just stop "pushing" to this repo:

Our main development tree is located here:

https://github.com/RobertCNelson/ti-linux-kernel-dev

Feel free to integrate that into your yocto patchset...

(end sarcasm)

Use the tag's, as the main master branches is un-stable, it's been this way since 2013. We don't have the resources to store our patchset in multiple formats.  I only push it to https://github.com/beagleboard/linux/ to make things easy, so end users don't have to run our patch.sh script..

Regards,

closed 2017-10-02 by RobertCNelson