beagleboard/linux #142: Trying to get 1 wire temp. sensor working

cren14, 2017-10-31:
New to beaglebone black but:

Found some code was hoping to use to get my 1 wire digital DS18B20 temperature sensor working but it wouldn't work. Just updated the debian image to the newest one October 10 2017. And now am not even seeing slots. I was using this nano code and instructions http://www.bonebrews.com/temperature-monitoring-with-the-ds18b20-on-a-beaglebone-black/

What is the best way to get this working in the new debian image? Should I just find a 3.8 image that a lot of people talk about online? Or is there a workaround? All the information I am finding is related to older versions so isn't very helpful. Does anyone have a newer version of this code?

Thanks!

RobertCNelson, 2017-10-31:
Hi @cren14 , this should just work.

Plug your sensor into P9.12, then add this option to /boot/uEnv.txt

```
uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo
```

Then just:

```
sudo modprobe wire
sudo modprobe w1-gpio
```
and:

```
ls /sys/bus/w1/devices/28-*
```

If it doesn't work, reply with:

```
sudo /opt/scripts/tools/version.sh
```
and we can debug..

cren14, 2017-11-01:
When I try to save the uEnv.txt file I am getting a "/boot/uEnv.txt" E212: Can't open file for writing". I tried using the :x! command and also :wq!

Also am I changing the exiting uboot overlad addr0=  listed under "Overide capes with eeprom" or do I add it in somewhere else in the txt file?

Results of ...................sudo /opt/scripts/tools/version.sh

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[d36fe9a7be9ebfc872b10a470e904ab4c61c4516]
eeprom:[A335BNLT00C02917BBBK27D1]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]
kernel:[4.4.91-ti-r133]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1~stretch+20171009]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
debian@beaglebone:~$

RobertCNelson, 2017-11-01:
Just use nano, i don't use vim, so i can't help you there..
```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]
```
Your old bootloader in the eMMC is getting in the way..

```
sudo dd if=/dev/zero of=/dev/mmcbkl1 count=1 seek=1 bs=128k
```
Regards,

cren14, 2017-11-03:
@RobertCNelson 

Thank you for your help. I was finally able to edit the file using sudo nano and can now see the temperature sensor reading.

Can you explain what doing the sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k did?

This is just the first step in my goal of what I am trying to do so I am sure there will be more posts in the future.

Thanks!

RobertCNelson, 2017-11-03:
It removed the old conflicting version of u-boot from your eMMC

gkazarian, 2018-02-11:
Hey @RobertCNelson, I'm trying to do the exact same thing as the original poster. I just fresh installed the latest Debian image on my BBB and tried what you mentioned up in the comments and it seems like it just wont work... 
This is the result I get for the version.sh:
```
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C04815BBBK04FA]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.380045] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END
```
when i try to do ls /sys/bus/w1/devices/28-* (even with sudo) i get a No such file or directory response...

Any help or pointers would be great! Thanks!

RobertCNelson, 2018-02-12:
@gkazarian see above:
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
```
That's blocking:

```
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
```

Regards,

closed 2018-03-05 by cren14

RobertCNelson, 2018-10-07:
@davemkirk no it'll need modifications, see the hint for "P2_08"..

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts#L36

besides "W1" isn't tied to a specific peripheral, you can use any gpio..

Regards,

davemkirk, 2018-10-08:
Thanks, makes sense 👍 

However, with my modified overlay I'm now getting a less helpful error...

```
[   53.182739] Driver for 1-wire Dallas network protocol.
[   53.196648] w1-gpio onewire: gpio_request (pin) failed
[   53.286200] w1-gpio: probe of onewire failed with error -16
```

Any idea what `error -16` is about?

Also, here is my modified overlay, any input on where I'm going wrong is greatly appreciated...
```
{
        fragment@0 {
                target = <&ocp>;
                __overlay__ {
                        P2_08_pinmux { status = "disabled"; };
                };
        };

        fragment@1 {
                target = <&am33xx_pinmux>;
                __overlay__ {

                        dallas_w1_pins: pinmux_dallas_w1_pins {
                                pinctrl-single,pins = <
                                        AM33XX_IOPAD(0x0878, 0x37)
                                >;
                        };
                };
        };

        fragment@2 {
                target-path="/";
                __overlay__ {

                        onewire {
                                status = "okay";
                                pinctrl-names = "default";
                                pinctrl-0 = <&dallas_w1_pins>;
                                compatible = "w1-gpio";
                                gpios = <&gpio1 28 GPIO_ACTIVE_HIGH>;
                        };
                };
        };
};

RobertCNelson, 2018-10-08:
Yeah, that should work, can we see your full serial boot log?

Regards,

BobCochran, 2018-10-08:
@davemkirk I also have a Pocket Beagle, so I would be interested in knowing how you have the sensor connected to it, what OS and OS version you are running on it, and also I'd love to look at your code. I'm not at all skilled with this. I've never done deep BeagleBoard stuff. I have the sensor, however, and I do have the interest. So I can learn and benefit from all of you in this exercise. Thanks and of course only if you have the time and inclination -- no pressure of any kind.

davemkirk, 2018-10-08:
@RobertCNelson Here it is...

```
U-Boot SPL 2018.03-00002-gac9cce7c6a (Apr 05 2018 - 13:07:46 -0500)
Trying to boot from MMC1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
Failed (-5)

U-Boot 2018.03-00002-gac9cce7c6a (Apr 05 2018 - 13:07:46 -0500), Build: jenkins-github_Bootloader-Builder-47

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: Internal.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
Failed (-5)
Model: BeagleBoard.org PocketBeagle
<ethaddr> not set. Validating first E-fuse MAC
Net:   usb_ether
Press SPACE to abort autoboot in 2 seconds
board_name=[A335PBGL] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
1881 bytes read in 40 ms (45.9 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.4.113-ti-r149 ...
9396680 bytes read in 637 ms (14.1 MiB/s)
loading /boot/dtbs/4.4.113-ti-r149/am335x-pocketbeagle.dtb ...
130113 bytes read in 110 ms (1.1 MiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/PB-W1-P2.08-00A0.dtbo ...
952 bytes read in 383 ms (2 KiB/s)
uboot_overlays: loading /lib/firmware/AM335X-PRU-UIO-00A0.dtbo ...
883 bytes read in 101 ms (7.8 KiB/s)
loading /boot/initrd.img-4.4.113-ti-r149 ...
6461152 bytes read in 452 ms (13.6 MiB/s)
debug: [console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet] ...
debug: [bootz 0x82000000 0x88080000:6296e0 88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8f9d6000, end 8ffff6e0 ... OK
   reserving fdt memory region: addr=88000000 size=80000
   Loading Device Tree to 8f953000, end 8f9d5fff ... OK

Starting kernel ...

[    0.001944] clocksource_probe: no matching clocksources found
[    1.266313] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
[    1.435038] omap_voltage_late_init: Voltage driver support not added
[    1.443349] PM: Cannot get wkup_m3_ipc handle
rootfs: recovering journal
rootfs: clean, 129095/483840 files, 644385/1947648 blocks
[[0m[0;31m*     [0m] (1 of 4) A start job is running for dev-ttyGS0.device (5s / 1min 30s)
[K[[0;32m  OK  [0m] Found device /dev/ttyS0.
[[0;32m  OK  [0m] Started Serial Getty on ttyS0.

Debian GNU/Linux 9 beaglebone ttyS0

BeagleBoard.org Debian Image 2018-06-17

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

beaglebone login: [   52.573069] w1-gpio onewire: gpio_request (pin) failed
```

davemkirk, 2018-10-08:
@BobCochran I'll gladly share my setup info once I get it working as intended 😀

davemkirk, 2018-10-21:
I'm not exactly sure why but I was able to get the w1 overlay I posted above working on PocketBeagle simply by upgrading to the from kernel 4.4.113-ti-r149 to 4.14.71-ti-r80. 

1-wire DS18B20 temp sensor now works on PocketBeagle exactly as expected for me now 👍

nachbar, 2018-10-26:
Thanks @RobertCNelson ; I was able to get this to work on the Beaglebone Black, following your instructions.

To clarify for those who may not understand exactly what @RobertCNelson meant when he said "this should  just work," the steps to compile a new overlay file are not necessary with the current Debian image available at [https://beagleboard.org/latest-images](https://beagleboard.org/latest-images)

Rather, a One Wire driver (called "W1") is included in the distribution, as is the overlay necessary to run the One Wire bus on P9, Pin 12.  If you can use that pin, all you have to do is follow the steps laid out in the initial response to @cren14 above: copy the line provided at the end of /boot/uEnv.txt , `sudo reboot`, and then issue the two `sudo modprobe` invocations, and your devices should be visible as above.  If these are DS18B20 temperature sensors, read their output with
```
cat /sys/bus/w1/devices/28-01(redacted)77/w1_slave
```

Also, I don't think it is necessary to issue the dd invocations to zero out `/dev/mmcbkl1` any longer, even though that device is still mentioned in the script output, which (for my working system created from the distro Debian 9.5 2018-10-07 4GB SD IoT flashed onto the eMMC) is:

```
$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian: 
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[(redacted)]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.107048] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.118908] gpio-of-helper ocp:cape-universal: ready
END
```
The last two lines in my dmesg output are (for the two DS18B20 devices I have connected to my BBB using that single pin):
```
[   91.829065] w1_master_driver w1_bus_master1: Attaching one wire slave 28.02(redacted)aa crc e1
[   91.950247] w1_master_driver w1_bus_master1: Attaching one wire slave 28.01(redacted)77 crc 68
```

Also, it appears that the manual `modprobe` invocations are not necessary on reboot after the first time they are used.

Again, thanks to @RobertCNelson for providing this information.

shapirc, 2019-03-07:
Any guidance on what changes are needed to use the 1W driver with another pin? For example I am trying to use P8.27, and I've modified the .dts file accordingly and compiled/installed the .dtbo, but can't seem to see my 1Wire sensor in /sys/bus/w1/devices (as opposed to when I am using the P9.12 .dtbo).

RobertCNelson, 2019-03-07:
@shapirc , post your changes...

shapirc, 2019-03-07:
My ultimate goal is to use P8.27 and P8.28. As of right now I am able to use P8.28 with this file (which I'd hoped would work for both), but P8.27 does not function:
`/*
 * Copyright (C) 2015 Robert Nelson <robertcnelson@gmail.com>
 *
 * Virtual cape for onewire on connector pin P9.12
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 */

/* Modified by Connor Shapiro for use with P8.27 & P8.28 */

/dts-v1/;
/plugin/;

#include <dt-bindings/board/am335x-bbw-bbb-base.h>
#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/am33xx.h>

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black", "ti,beaglebone-green";

	/* identification */
	part-number = "BB-W1-P8.27_28";
	version = "00A0";

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P8.27",
		"P8.28";

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@0 {
		target = <&ocp>;
		__overlay__ {
			P8_27_pinmux { status = "disabled"; };
			P8_28_pinmux { status = "disabled"; };
		};
	};

	fragment@1 {
		target = <&am33xx_pinmux>;
		__overlay__ {

			dallas_w1_pins: pinmux_dallas_w1_pins {
				pinctrl-single,pins = < 
					BONE_P8_27 0x37
					BONE_P8_28 0x37
				>;
			};
		};
	};

	fragment@2 {
		target-path="/";
		__overlay__ {

			onewire2 {
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&dallas_w1_pins>;

				compatible = "w1-gpio";
				gpios = <&gpio2 22 GPIO_ACTIVE_HIGH>;
			};
		};
	};

	fragment@3 {
		target-path="/";
		__overlay__ {

			onewire3 {
				status = "okay";
				pinctrl-names = "default";
				pinctrl-0 = <&dallas_w1_pins>;

				compatible = "w1-gpio";
				gpios = <&gpio2 24 GPIO_ACTIVE_HIGH>;
			};
		};
	};
};
`

RobertCNelson, 2019-03-07:
Split this: into two:

```
	fragment@1 {
		target = <&am33xx_pinmux>;
		__overlay__ {

			dallas_w1_pins: pinmux_dallas_w1_pins {
				pinctrl-single,pins = < 
					BONE_P8_27 0x37
					BONE_P8_28 0x37
				>;
			};
		};
	};
```

LIke:

```
	fragment@1 {
		target = <&am33xx_pinmux>;
		__overlay__ {

			dallas_w1_1_pins: pinmux-dallas-w1-1pins {
				pinctrl-single,pins = < 
					BONE_P8_27 0x37
				>;
			};

			dallas_w1_2_pins: pinmux-dallas-w1-2-pins {
				pinctrl-single,pins = < 
					BONE_P8_28 0x37
				>;
			};
		};
	};
``` 

Regards,

WalterCEden, 2019-05-23:
I'm also a relatively newbie to BBB and trying to get a DS18B20 temperature sensor working with 1-wire.   Here is the output from version.sh (this was booted from an SD card).

git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C04417BBBK1847]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.107595] pinctrl-single 44e10800.pinmux: 142pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.119429] gpio-of-helper ocp:cape-universal: ready
END

Here is my uEnv.txt file.   

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.71-ti-r80
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/w1-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

So I booted by holding down the user button while pressing power on.  

I executed the two sudo modprobe commands with no errors report.  

All I see in /sys/bus/w1/devices is w1_bus_master1

Suggestions?

jassjagga, 2020-07-19:
I'm also having the same problem try all above mentions setups.

`git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT000C1825BBBG1315]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   60.386052] remoteproc remoteproc0: wkup_m3 is available
[   60.443726] remoteproc remoteproc0: powering up wkup_m3
[   60.443758] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   60.444045] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   62.943729] remoteproc remoteproc1: 4a334000.pru is available
[   62.957958] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   62.943729] remoteproc remoteproc1: 4a334000.pru is available
[   62.943889] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   62.957958] remoteproc remoteproc2: 4a338000.pru is available
[   62.958137] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.949750] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[   60.219360] pinctrl-single 44e10800.pinmux: pin PIN30 already requested by ocp:P9_12_pinmux; cannot claim for ocp:onewire@0
[   60.418513] pinctrl-single 44e10800.pinmux: pin-30 (ocp:onewire@0) status -22
[   60.495847] pinctrl-single 44e10800.pinmux: could not request pin 30 (PIN30) from group pinmux_dallas_w1_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.962986] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
error 

`'/sys/bus/w1/devices/28-*': No such file or directory`

anyone help, please ??

WalterCEden, 2020-07-20:
It has been a while and I don't have the system in front of me at the moment to confirm but I ended up getting a new version of Debian that fixed a problem that was preventing one-wire from working.   As soon as I switched to that version, my code worked just fine.   Ironically, after getting this to work, we stopped using one-wire in the design though and I don't have it in front of me anymore!   I suspect that if you search the web some more you may find the reference to the version that works that I found.  That's how I found out about it.

RobertCNelson, 2020-07-20:
@jassjagga can you please share your version of BB-W1-P9.12-00A0.dts, as the error you got, should not happen:
```
 | grep pinctrl-single 
[ 0.949750] pinctrl-single 44e10800.pinmux: 142 pins, size 568 
[ 60.219360] pinctrl-single 44e10800.pinmux: pin PIN30 already requested by ocp:P9_12_pinmux; cannot claim for ocp:onewire@0 
[ 60.418513] pinctrl-single 44e10800.pinmux: pin-30 (ocp:onewire@0) status -22 
[ 60.495847] pinctrl-single 44e10800.pinmux: could not request pin 30 (PIN30) from group pinmux_dallas_w1_pins on device pinctrl-single
```

This line takes care of that:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts#L51

Regards,

jassjagga, 2020-07-20:
thank you so much @RobertCNelson for the reply. I did figure out my error and I am good now. I was just doing some little stupid errors. but thanks for your time and Help.