beagleboard/linux #144: Pocketbeagle P2_24 gpio function not working

derekmulcahy, 2017-11-01:
I tried to toggle P2_24 using config-pin but it had no effect.

According to https://github.com/beagleboard/pocketbeagle/issues/4 the silkscreen is wrong for P2_24. It is labelled on the PCB as GPIO 48 but is really GPIO 44 aka GPMC_AD12. This is confirmed in the schematic. This error has been carried over into the device tree. The device tree setting for P2_24 is currently.

```
                        P2_24 {
                                gpio-name = "P2_24";
                                gpio = <0x167 0x10 0x0>;
                                input;
                                dir-changeable;
                        };
```

I think the 0x10 should be 0xc. This matches with the values used for P2_22 and others. Changing this and editing config-pin to use GPIO 44 for P2_24 works!

The pinmux settings in the device tree and config-pin are also incorrect as they are set for GPIO 48.

RobertCNelson, 2017-11-01:
@jadonk can you help clarify this, what "ball" is P2_24 really connected too?

Looking at the "eagle" schematic from git:

https://github.com/beagleboard/pocketbeagle/tree/master/EAGLE

P2.24 is still labeled as GPIO 48, which is ball "P6" in the eagle schematic..  "P6" doesn't exist on the am335x.json file for this package.  This should be "GPIO 44" that's U1 right????

RobertCNelson, 2017-11-01:
U12 is GPIO 0_27...

RobertCNelson, 2017-11-01:
T12 looks correct, @jadonk can you please confirm then i'll push out a new, v4.4.x-t/v4.9.x-ti/v4.13.x-bone/v4.14.x-ti/v4.14.x-bone kernel's..

Regards,

```
diff --git a/tools/pinmux-generator/PocketBeagle.dts b/tools/pinmux-generator/PocketBeagle.dts
index c41804e..7d9b3ec 100644
--- a/tools/pinmux-generator/PocketBeagle.dts
+++ b/tools/pinmux-generator/PocketBeagle.dts
@@ -538,17 +538,19 @@ P2_22_qep_pin: pinmux_P2_22_qep_pin { pinctrl-single,pins = <
 
 /* P2_23                VOUT-3.3V */
 
-/* P2_24 (ZCZ ball R13) */
+/* P2_24 (ZCZ ball T12) */
 P2_24_default_pin: pinmux_P2_24_default_pin { pinctrl-single,pins = <
-	AM33XX_IOPAD(0x0840, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* gpmc_a0.gpio1_16 */
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* gpmc_ad12.gpio1_12 */
 P2_24_gpio_pin: pinmux_P2_24_gpio_pin { pinctrl-single,pins = <
-	AM33XX_IOPAD(0x0840, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };	/* gpmc_a0.gpio1_16 */
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT | INPUT_EN | MUX_MODE7) >; };	/* gpmc_ad12.gpio1_12 */
 P2_24_gpio_pu_pin: pinmux_P2_24_gpio_pu_pin { pinctrl-single,pins = <
-	AM33XX_IOPAD(0x0840, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE7) >; };	/* gpmc_a0.gpio1_16 */
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE7) >; };	/* gpmc_ad12.gpio1_12 */
 P2_24_gpio_pd_pin: pinmux_P2_24_gpio_pd_pin { pinctrl-single,pins = <
-	AM33XX_IOPAD(0x0840, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* gpmc_a0.gpio1_16 */
-P2_24_pwm_pin: pinmux_P2_24_pwm_pin { pinctrl-single,pins = <
-	AM33XX_IOPAD(0x0840, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE6) >; };	/* gpmc_a0.ehrpwm1_tripzone_input */
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* gpmc_ad12.gpio1_12 */
+P2_24_pruout_pin: pinmux_P2_24_pruout_pin { pinctrl-single,pins = <
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE6) >; };	/* gpmc_ad12.pr1_pru0_pru_r30_14 */
+P2_24_qep_pin: pinmux_P2_24_qep_pin { pinctrl-single,pins = <
+	AM33XX_IOPAD(0x0830, PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE4) >; };	/* gpmc_ad12.eqep2a_in */
 
 /* P2_25 (ZCZ ball E17) */
 P2_25_default_pin: pinmux_P2_25_default_pin { pinctrl-single,pins = <
@@ -1213,16 +1215,17 @@ P2_22_pinmux {
 
 /* P2_23                VOUT-3.3V */
 
-/* P2_24 (ZCZ ball R13) */
+/* P2_24 (ZCZ ball T12) */
 P2_24_pinmux {
 	compatible = "bone-pinmux-helper";
 	status = "okay";
-	pinctrl-names = "default", "gpio", "gpio_pu", "gpio_pd", "pwm";
+	pinctrl-names = "default", "gpio", "gpio_pu", "gpio_pd", "qep", "pruout";
 	pinctrl-0 = <&P2_24_default_pin>;
 	pinctrl-1 = <&P2_24_gpio_pin>;
 	pinctrl-2 = <&P2_24_gpio_pu_pin>;
 	pinctrl-3 = <&P2_24_gpio_pd_pin>;
-	pinctrl-4 = <&P2_24_pwm_pin>;
+	pinctrl-4 = <&P2_24_qep_pin>;
+	pinctrl-5 = <&P2_24_pruout_pin>;
 };
 
 /* P2_25 (ZCZ ball E17) */
@@ -1611,7 +1614,7 @@ cape-universal {
 
 	P2_24 {
 		gpio-name = "P2_24";
-		gpio = <&gpio1 16 0>;
+		gpio = <&gpio1 12 0>;
 		input;
 		dir-changeable;
 	};
diff --git a/tools/pinmux-generator/PocketBeagle.sh b/tools/pinmux-generator/PocketBeagle.sh
index 8bac2a6..5bcf67e 100755
--- a/tools/pinmux-generator/PocketBeagle.sh
+++ b/tools/pinmux-generator/PocketBeagle.sh
@@ -241,7 +241,7 @@ echo "P2_23_INFO=\"VOUT-3.3V\"" >>${file}_config-pin.txt
 echo "P2_23_CAPE=\"\"" >>${file}_config-pin.txt
 echo "" >>${file}_config-pin.txt
 
-pcbpin="P2_24" ; ball="R13" ; find_ball
+pcbpin="P2_24" ; ball="T12" ; find_ball
 
 cp_info_default="spi1_cs0" ; cp_default="spi" ; default="SPI" ; pcbpin="P2_25" ; ball="E17" ; find_ball
 
diff --git a/tools/pinmux-generator/PocketBeagle_config-pin.txt b/tools/pinmux-generator/PocketBeagle_config-pin.txt
index ccc21bf..5a1d127 100644
--- a/tools/pinmux-generator/PocketBeagle_config-pin.txt
+++ b/tools/pinmux-generator/PocketBeagle_config-pin.txt
@@ -334,11 +334,11 @@ P2_23_PIN="power"
 P2_23_INFO="VOUT-3.3V"
 P2_23_CAPE=""
 
-P2_24_PRU="80"
-P2_24_GPIO="48"
+P2_24_PRU="76"
+P2_24_GPIO="44"
 P2_24_PIN="gpio"
-P2_24_PINMUX="default gpio gpio_pu gpio_pd pwm"
-P2_24_INFO="gpio1_16 default gpio1_16 gpio1_16 gpio1_16 ehrpwm1_tripzone_input"
+P2_24_PINMUX="default gpio gpio_pu gpio_pd qep pruout"
+P2_24_INFO="gpio1_12 default gpio1_12 gpio1_12 gpio1_12 eqep2a_in pruout"
 P2_24_CAPE=""
 
 P2_25_PRU="73"
```

jadonk, 2018-01-02:
@RobertCNelson sorry to not see this earlier. Confirmed the issue some time back.

jadonk, 2018-01-29:
@RobertCNelson ping.

RobertCNelson, 2018-01-29:
Ah this was fixed around on:

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/671bd40ad3f42c30643ab389b5a41e45697cbdcd

merged to all, safe to close.

Regards,

closed 2018-01-29 by RobertCNelson