beagleboard/linux #145: pru rproc in 4.14

dlech, 2017-11-01:
I am trying out the 4.14 kernel. It seems that the pru_rproc driver is missing. Will it be coming back?

RobertCNelson, 2017-11-01:
Hi @dlech , it hasn't been merged back in by ti, right now's it's 99% mainline..

http://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.14.y

Regards,

RobertCNelson, 2017-11-14:
Just a heads up, the TI merge is now happening:

http://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.14.y

no pru stuff yet, just sleep/cpufreq/rtc etc..

Regards,

RobertCNelson, 2018-01-23:
@dlech okay heads up pru_rproc changes inbound on v4.14.x-ti:

http://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.14.y

http://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=commit;h=2e4d3628430b3e4932e9236d7fae980237557b42

I'll cut a tag shortly to get a better diff, right now most of it is related to your da850. ;)

Regards,

RobertCNelson, 2018-01-23:
@dlech here is the raw commit diff:

https://github.com/RobertCNelson/ti-linux-kernel/compare/ba319d355cc887dcc27a7b005285fd83d720c2b7...4adf1bb08f2c89821c72e3c26a8f37bd963f849f

Regards,

closed 2018-04-13 by dlech