beagleboard/linux #148: kernel 4.14 RNDIS usb on windows 7 doesn't work

HarelAshwal, 2018-01-20:
LK"I

Hi,
I just compiled kernel 4.14 and tried to use it with my Windows 7 OS.
If I understand currently , with 4.14 usb driver changes, it should work with Windows 7 without any drivers.
It doesn't work... device manager still shows undetected RNDIS driver...

any ideas?

Harel

RobertCNelson, 2018-01-20:
this should be fixed via:

https://github.com/RobertCNelson/boot-scripts/commit/714e162ba98cf3d2897e7fc95e951c6df15a7d0a

```
cd /opt/scripts/
git pull
```

HarelAshwal, 2018-01-22:
LK"I

tried it, still doesn't work.
the values seems to be updated on the system, but still no effect on device manager on windows 7

root@arm:/sys/kernel/config/usb_gadget/g_multi/functions/rndis.usb0# cat class
ef
s ot@arm:/sys/kernel/config/usb_gadget/g_multi/functions/rndis.usb0# cat subclass
04
l ot@arm:/sys/kernel/config/usb_gadget/g_multi/functions/rndis.usb0# cat protocol
01
root@arm:/sys/kernel/config/usb_gadget/g_multi/functions/rndis.usb0#

debian@arm:~$ uname -a
Linux arm 4.14.13-bone12 #1 Sun Jan 14 01:00:21 IST 2018 armv7l GNU/Linux

HarelAshwal, 2018-01-23:
LK"I

yeah checked again, it works.
thanks

closed 2018-01-23 by HarelAshwal