beagleboard/linux #150: Kernel 4.9 only have CPU frequency at 800Mhz

SandPox, 2018-02-17:
For some reason, kernel 4.9-rt (I didn't tested the non-rt version) only able to reach 800Mhz max on CPU clock, on `cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq` show `800000`, the kernel 4.14-rt is not affected by this.

RobertCNelson, 2018-02-19:
@SandPox can you please share the output of this script:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

SandPox, 2018-02-20:
**root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[89f2e9309c2322c15e2bb6b55afd214313d94842]
eeprom:[A335BNLTBLA21712EL002139]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]:[location: dd MBR]
kernel:[4.9.81-ti-rt-r99]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.591393] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.638327] pinctrl-single 44e10800.pinmux: pin PIN85 already requested by 48024000.serial; cannot claim for ocp:P9_21_pinmux
[    1.649792] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P9_21_pinmux) status -22
[    1.657235] pinctrl-single 44e10800.pinmux: could not request pin 85 (PIN85) from group pinmux_P9_21_uart_pin  on device pinctrl-single
[    1.678360] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for ocp:P9_22_pinmux
[    1.689727] pinctrl-single 44e10800.pinmux: pin-84 (ocp:P9_22_pinmux) status -22
[    1.697164] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group pinmux_P9_22_uart_pin  on device pinctrl-single
[    1.720374] pinctrl-single 44e10800.pinmux: pin PIN97 already requested by 48022000.serial; cannot claim for ocp:P9_24_pinmux
[    1.731750] pinctrl-single 44e10800.pinmux: pin-97 (ocp:P9_24_pinmux) status -22
[    1.739188] pinctrl-single 44e10800.pinmux: could not request pin 97 (PIN97) from group pinmux_P9_24_uart_pin  on device pinctrl-single
[    1.760397] pinctrl-single 44e10800.pinmux: pin PIN96 already requested by 48022000.serial; cannot claim for ocp:P9_26_pinmux
[    1.771761] pinctrl-single 44e10800.pinmux: pin-96 (ocp:P9_26_pinmux) status -22
[    1.779199] pinctrl-single 44e10800.pinmux: could not request pin 96 (PIN96) from group pinmux_P9_26_uart_pin  on device pinctrl-single
[    1.805958] pinctrl-single 44e10800.pinmux: pin PIN100 already requested by ocp:P9_31_pinmux; cannot claim for 481a0000.spi
[    1.817229] pinctrl-single 44e10800.pinmux: pin-100 (481a0000.spi) status -22
[    1.824410] pinctrl-single 44e10800.pinmux: could not request pin 100 (PIN100) from group pinmux_P9_31_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
root@beaglebone:~#
**

SandPox, 2018-02-20:
Looks like the non-rt version 4.9 have the same problem
```
root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[89f2e9309c2322c15e2bb6b55afd214313d94842]
eeprom:[A335BNLTBLA21712EL002139]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]:[location: dd MBR]
kernel:[4.9.81-ti-r99]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.404716] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.444968] pinctrl-single 44e10800.pinmux: pin PIN85 already requested by 48024000.serial; cannot claim for ocp:P9_21_pinmux
[    1.456517] pinctrl-single 44e10800.pinmux: pin-85 (ocp:P9_21_pinmux) status -22
[    1.463990] pinctrl-single 44e10800.pinmux: could not request pin 85 (PIN85) from group pinmux_P9_21_uart_pin  on device pinctrl-single
[    1.485134] pinctrl-single 44e10800.pinmux: pin PIN84 already requested by 48024000.serial; cannot claim for ocp:P9_22_pinmux
[    1.496541] pinctrl-single 44e10800.pinmux: pin-84 (ocp:P9_22_pinmux) status -22
[    1.503999] pinctrl-single 44e10800.pinmux: could not request pin 84 (PIN84) from group pinmux_P9_22_uart_pin  on device pinctrl-single
[    1.526916] pinctrl-single 44e10800.pinmux: pin PIN97 already requested by 48022000.serial; cannot claim for ocp:P9_24_pinmux
[    1.538330] pinctrl-single 44e10800.pinmux: pin-97 (ocp:P9_24_pinmux) status -22
[    1.545789] pinctrl-single 44e10800.pinmux: could not request pin 97 (PIN97) from group pinmux_P9_24_uart_pin  on device pinctrl-single
[    1.566982] pinctrl-single 44e10800.pinmux: pin PIN96 already requested by 48022000.serial; cannot claim for ocp:P9_26_pinmux
[    1.578381] pinctrl-single 44e10800.pinmux: pin-96 (ocp:P9_26_pinmux) status -22
[    1.585837] pinctrl-single 44e10800.pinmux: could not request pin 96 (PIN96) from group pinmux_P9_26_uart_pin  on device pinctrl-single
[    1.612014] pinctrl-single 44e10800.pinmux: pin PIN100 already requested by ocp:P9_31_pinmux; cannot claim for 481a0000.spi
[    1.623322] pinctrl-single 44e10800.pinmux: pin-100 (481a0000.spi) status -22
[    1.630526] pinctrl-single 44e10800.pinmux: could not request pin 100 (PIN100) from group pinmux_P9_31_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
root@beaglebone:~# cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_max_freq
800000
root@beaglebone:~#

```

RobertCNelson, 2018-02-21:
@SandPox thanks for the report, i got fixes staged, need to fix one more thing, then this will be pushed today:

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/3d260b28db9f09bd99e7f7583fc3d669b2ea71ad

Regards,

richardschuetz, 2018-02-22:
I'm observing a similar issue on the BeagleBone Black Wireless. `OSD3358-00A0.dtbo` isn't loaded for the device by **U-Boot 2018.01** anymore. As a result the default OPP defintion for 1 GHz from `am33xx.dtsi` does not fit the silicon:
```
cpufreq: cpufreq_online: CPU0: Running at unlisted freq: 1000000 KHz
cpu cpu0: dev_pm_opp_set_rate: failed to find current OPP for freq 1000000000 (-34)
cpufreq: cpufreq_online: CPU0: Unlisted initial frequency changed to: 800000 KHz
```

richardschuetz, 2018-02-22:
Nevermind. It has been addressed in 534c8e5b3b0606f2e410903f7ae0f37b618dd09f. Still missing in https://github.com/RobertCNelson/dtb-rebuilder/tree/4.9-ti, though.

RobertCNelson, 2018-02-22:
@richardschuetz yeah i fubard this transition.  Background, mainline changed the opp frequency node name, thus i couldn't use the same  OSD3358-00A0.dtbo overlay hack on v4.4.x/v4.9.x/v4.14.x kernel's..   While i had v4.4.x/v4.14.x and mainline ready for this removal, i fubared v4.9.x-ti...

closed 2018-02-25 by SandPox