beagleboard/linux #151: Enable/Disable PWM in BBB Programmatically (C/C++)

Soheilica, 2018-02-20:
Hi,

Im using BeagleBoard Black and the question is: 

How can I Enable / Disable PWM Pin on BBB programmatically (C/C++)?

For the moment I use system ("Config-pin Px:xx pwm") but it seems that after a reboot the PWM does Not run unless I manually go in the ocp folder and export manually the PWM.

Any thoughts are appreciated.

Thank you

RobertCNelson, 2018-02-20:
@Soheilica correct after a reboot, you'll need to export the pwm.

I didn't answer you question on the forum because, I don't have a good "C/C++" answer for you, and you are really limiting your options.

The best way to use the pwm, is to use the adafruilt python library: https://github.com/adafruit/adafruit-beaglebone-io-python

Regards,

ericrini, 2019-01-18:
I think it's confusing to say that the Python library is "the best" way to do anything. Perhaps it is the "easiest" or "most user friendly". However the is valid reasons to want to understand how to do it in a lower level way?

closed 2019-01-18 by RobertCNelson