beagleboard/linux #154: Enabling SPI in Debian 9

SalahuddinAsh, 2018-03-16:
I'm following [this tutorial](https://elinux.org/BeagleBone_Black_Enable_SPIDEV) to enable SPI on BBB.

When I followed it on Debian 7.5, it worked fine.

But when I followed it on Debian 9.3, it didn't work.
The output of `uname -a` is:
`Linux beaglebone 4.9.82-ti-r102 #1 SMP PREEMPT Thu Feb 22 01:16:12 UTC 2018 armv7l GNU/Linux`

After `echo BB-SPI1-01 > $SLOTS`, a new entry was added to $SLOTS but no new devices appeared in `/dev/`.

When I run `dmesg`, I found those errors:
[image]

Any idea how to fix it?

fivdi, 2018-03-16:
The tutorial is old and doesn't work work with the Kernel/Debian version that you're using. There's some information related to the topic [here](https://github.com/julianduque/beaglebone-io/issues/44#issue-287592899). Use the `config-pin` utility rather than overlays and the slots file to configure SPI.

SalahuddinAsh, 2018-03-16:
@fivdi 
Many thanks for your quick reply. I'll check the link and come back with the results.

SalahuddinAsh, 2018-03-16:
@fivdi 
I tried configuring the two spi. I could configure `SPI0` but when I tried to configure `P9.28`, `P9.29`, `P9.31`, the output was:
```
P9_XX pinmux file not found!
sudo: no askpass program specified, try setting SUDO_ASKPASS
cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_XX_pinmux/state
```
Someone [here](https://groups.google.com/forum/#!topic/beagleboard/EYSwmyxYjdM) suggested to add the following lines in `uEnv.txt` but it didn't work for me.
```
cape_enable=bone_capemgr.enable_partno=cape-universala
cmdline=coherent_pool=1M quiet cape_universal=enable
```
How can I solve this problem?

I have another question: how can I set `spi_cs` for example to be input and not output (i.e., to act as a slave)? When I tried using `in` option, the pin was set as a gpio!
How can I do this using `config-pin`?

Thanks

SalahuddinAsh, 2018-03-16:
I found the answer for the first problem [here](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/171). 
Still the second one.

fivdi, 2018-03-16:
Sorry, I've never used a BeagleBone as an SPI slave and can't provide any information here.

sandebert, 2018-03-21:
I don't know enough about the syntax in uEnv.txt to be sure, but my gut tells me that you made a typo here:
```
cape_enable=bone_capemgr.enable_partno=cape-universala
cmdline=coherent_pool=1M quiet cape_universal=enable
```

In the cmdline you reference `cape_universal`, but in cape_enable you call it `cape-universala`. (An extra _a_ last.)

akashlad007, 2019-03-13:
> @fivdi
> I tried configuring the two spi. I could configure `SPI0` but when I tried to configure `P9.28`, `P9.29`, `P9.31`, the output was:
>
> ```
> P9_XX pinmux file not found!
> sudo: no askpass program [...]

Hello,
         I am also using the same debian release i.e. debian 4.14.71-ti-r80. I am facing the same problem as you are, can you please elaborate how did you enabled the spi0. As whenever i am writing to my uEnv.txt file the entire system gets corrupted and i have to burn the image again. Also i have a spi test code in c it gets compiled using gcc but whenever i run it using ./spidev_test -D /dev/spidev1.0 this command i get output as:     can't open device: No such file or directory     Aborted. could you please help me.

Thanks.